zigpy/zha-device-handlers #234: Add support for Sage Doorbell sensor

hrp, 2020-01-09:
I would like to help add support for the Sage Doorbell sensor. This product is also known as the Echostar Bell sensor.

* Info: https://darwinsden.com/sage-doorbell-sensor-review-smartthings-device-handler/
* Example from Hubitat: https://github.com/trunzoc/Hubitat/blob/master/Drivers/Sage_Doorbell/Sage_Doorbell.groovy

I started working on this here: 4a4a39294.

Now that I have a new class for this device, how do I test this on my HA installation?

hrp, 2020-01-31:
@dmaciaszek Would love to push on this but I actually have no idea how to test this or develop on HA. Any tips?

walthowd, 2020-03-13:
This actually works for me currently, straight out of the box. I get a zha_event (I think an on then an off)

dmulcahey, 2020-03-28:
done by #303

closed 2020-03-28 by dmulcahey

alexdelprete, 2021-02-19:
Battery power of the Sage is not reported correctly. Can it be fixed?

mguaylambert, 2021-02-19:
@alexdelprete the cluster isnthere but not reporting anything. It is a limitation on the device. They probably did not implement it yet and would have done it via a firmware update. Unfortunately those products are abandoned now. I guess you could always try to hack the firmware?

Adminiuga, 2021-02-21:
What he said.

alexdelprete, 2021-02-21:
> @alexdelprete the cluster is there but not reporting anything. It is a limitation on the device. They probably did not implement it yet and would have done it via a firmware update. Unfortunately those products are abandoned now. I guess you could always try to hack the firmware?

I just read in the HE driver changelog that battery capability is not operational. So it's the device. Thanks for answering.

"0.30 (11/20/2016) - Removed non-operational battery capability"