zigpy/zha-device-handlers #235: Error adding LDS ZBT-CCTSwitch-D0001 (ecosmart remote) to Group

tube0013, 2020-01-10:
When I try to add the ecosmart remote to a group I get this error:

[zigpy.endpoint] [0x3f5e:1] Cannot add 0x0002 group, no groups cluster

Adminiuga, 2020-01-10:
Remotes cannot be added to groups, i.e. be a group members. Remotes are sending commands to the groups -> you bind remote to a group
Only devices with on_off, level and color **server** clusters can/should be added to the groups

tube0013, 2020-01-10:
Can you explain a bit more what you meant in this comment then?  

https://github.com/dmulcahey/zha-device-handlers/issues/209#issuecomment-564662104

Does this mean if the remote is bound to a bulb directly with touchlink and that bulb is an HA group with other bulbs?

Adminiuga, 2020-01-10:
In #209 i meant that remote cannot do `IEEE` binds, where it sends commands to a specific destination IEEE address. Beside `IEEE` (which I called direct bind) you can also do a `group bind` --> remote sends commands to a specific group_id (as opposed to a specific IEEE) and I meant that remote should be bound to a group.
Now what happens when it is already bound to a  bulb -- I don't know, haven't played with it yet. 
Can you have that remote in ZHA and also bound to a touchlink bulb? Can you explains the steps how you did it?

Adminiuga, 2020-01-10:
My speculation, there could be a group already on the ZLL cluster and if you add other lights to the same group id, then they would be controlled by the same remote. Questions remains open if the remote can support both/multiple groups at the same time.

tube0013, 2020-01-10:
noI haven't done that. I was just trying to understand what you wrote in that comment. now I think I understand better.  Is there a way to bind the remote to a group in HA with ui or service calls now, or is that still to come..

dmulcahey, 2020-02-02:
Support for group binding is in the current beta.

ajayjohn, 2020-02-20:
@dmulcahey , @Adminiuga ,
Could you please help me understand what i am doing wrong?

I added remote X to my ZHA instance on HA. I noticed that this automatically created a group with the name 'No name group 0x4003'. If I add any ZHA lights to this group, the remote seems to control them perfectly and all buttons work.

Now, I add remote Y to my ZHA instance. I notice that no new groups have been added automatically. The only group that I see is the earlier one ending in 4003. So I go ahead and create a new group with a different set of lights in it. Then I group bind all available entries on remote Y to the newly added group using the 'Group Binding' option (I see no error messages in the log, but I don't see any success messages either). However, on using the remote, I notice that both remotes still control lights on the original group ending in 4003.

In short, how many ever remotes I add, they all seem to control just the one group that was automatically added with the first remote and I am unable to bind any of the remotes to new groups that I created manually.

ajayjohn, 2020-03-22:
@dmulcahey , @Adminiuga ,
Could you please help me out with the issue posted above? I know this is not a support thread, but I have asked the same question in multiple HA community threads too and have never received a response that worked for me :(

Adminiuga, 2020-03-23:
Open a new issue, as this one is different and probably should be closed. Enable debug and provide logs of pairing the second remote and binding it to a group.

Adminiuga, 2020-03-23:
But I can confirm those remotes by default create group 0x4003

Adminiuga, 2020-03-23:
And I can confirm your findings: Even when remote reports a successful bind to a group, it does not appear to issue command to that group. At least the light is not reacting to it and I haven't tried zigbee sniffer to see if anything is being sent out.

ajayjohn, 2020-03-28:
Thanks, @Adminiuga!
Let me try to pull some logs and post an issue. Am glad that you could replicate the problem.

lordkev, 2020-04-16:
@ajayjohn Did you ever figure out binding remotes to groups? Didn't see an issue created, and I'm having the same problem. I've tried with both a Hue Dimmer Switch (RWL020) and one of the newer Lutron Auroras. It seems to bind to my custom light group ok (as fas as I can tell), but doesn't seem to do anything.

It does work for my Lutron Connected Bulb Remote, but it looks like that quirk is using a workaround (GroupBoundCluster). 

Happy to create a new issue but wanted to make sure I'm not overlooking anything.

ajayjohn, 2020-04-20:
@lordkev No, I never really found a solution :(
In fact, I got busy with some other stuff and haven't really been able to focus on even getting the right logs for this issue.

closed 2021-08-20 by github-actions[bot]