zigpy/zha-device-handlers #236: Support for Ikea SYMFONISK Puck?

cwesterfield, 2020-01-12:
Context: https://old.reddit.com/r/homeassistant/comments/dssysn/nr_rabbit_hole_part_10_sonic_attack_using_the/

Device:
https://www.ikea.com/us/en/p/symfonisk-sound-remote-black-10433847/

I'm interested in exactly what the reddit fellow is doing with nodered but I use a HUSBZB-1 with ZHA. 

Is that supported, or could it be in the future?

dmulcahey, 2020-01-31:
added in 0.0.32. fixed by #255

closed 2020-01-31 by dmulcahey

Royal2000H, 2020-02-27:
I've upgraded to 0.0.33 and added a Symfonisk remote. I get multiples for every event. So if I just do a quick rotate of the knob, I get (for example) 4 move events followed by 5 stop events. I assume this is not intentional?

andrewpc, 2020-03-16:
I was about to raise a bug for the same issue. A single click on the controller generates the following:

`Event 4 fired 11:12 AM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:94:5f:03:1:0x0006",
        "device_ieee": "14:b4:57:ff:fe:94:5f:03",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-16T00:12:29.126929+00:00",
    "context": {
        "id": "600ff24d548a4c02a5453fbb96a94435",
        "parent_id": null,
        "user_id": null
    }
}
Event 3 fired 11:12 AM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:94:5f:03:1:0x0006",
        "device_ieee": "14:b4:57:ff:fe:94:5f:03",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-16T00:12:29.118013+00:00",
    "context": {
        "id": "2419d822e7b74c348bd41903625d30d1",
        "parent_id": null,
        "user_id": null
    }
}
Event 2 fired 11:12 AM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:94:5f:03:1:0x0006",
        "device_ieee": "14:b4:57:ff:fe:94:5f:03",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-16T00:12:29.111741+00:00",
    "context": {
        "id": "4b54c6e15833460da6ccb420c83ce9c6",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 11:12 AM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:94:5f:03:1:0x0006",
        "device_ieee": "14:b4:57:ff:fe:94:5f:03",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-16T00:12:29.095338+00:00",
    "context": {
        "id": "26e05ae4bd9e4e898f0a603e0e43ea6c",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 11:12 AM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:94:5f:03:1:0x0006",
        "device_ieee": "14:b4:57:ff:fe:94:5f:03",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-16T00:12:29.072146+00:00",
    "context": {
        "id": "fa8580fded2a4b349827f5a4f8fcb003",
        "parent_id": null,
        "user_id": null
    }
}`

This causes havoc with automations and also affects the double and triple click implementation.

My research seems to indicate zigbee2mqtt solves this by using a debounce [feature](https://www.zigbee2mqtt.io/devices/E1744.html) which ignores followup events with a defined time window, whereas there is specific conversation in the Hubitat forum that indicates that depending on the cluster that is bound, then the behaviour changes. See here: https://community.hubitat.com/t/ikea-symfonisk-remote-on-french-ikea-zigbee/22585/15

dmulcahey, 2020-03-16:
We rely on the device to do sane things. If someone wants to contribute debounce code they are welcome to.

andrewpc, 2020-03-16:
Thanks David. Any thoughts on my second link? Could the device be initialised/configured differently?

dmulcahey, 2020-03-16:
Hrm, that’s really interesting. I’ll try it out tomorrow

andrewpc, 2020-03-19:
Did you get a chance to take a look at this one?