zigpy/zha-device-handlers #237: Ikea Motion Sensor (E1745)

SimplyGardner, 2020-01-12:
HomeAssistant (0.103.6) + ZHA + ConBee II

**Issue:**
I am not sure if this is an issue or even if the IKEA motion sensor is supported. It does however seem to work well although it has sometimes strange behaviour - breaking the connection multiple times throughout the day and HA reports it as 'Unavailable'.

**Supporting Docs:**
Here is what the history graph looks like
[image]

I get multiple reports in the logs when the sensor goes offline
[image]

**I have two theories:** 

1. Either ZHA does not fully support this sensor or;
2. The sensor is turning its self off when the light source is at a certain level. I notice this in the IKEA instructions manual:

[image]

Any insights would be helpful.

SimplyGardner, 2020-01-13:
Further investigation.

I've left it most of the day and it looks like if motion is not detected after sometime - the motion sensor puts its self in 'unavailable' mode. When it is showing in this status, I walked under the sensor and it immediately woke up and reported its new status as 'detected'.

New image:
[image]

Adminiuga, 2020-01-13:
ZHA would mark device unavailable if it doesn't receive any traffic from the device for 2 hours. ZHA would try to "ping" the device and that's what probably is causing those `0xf0 error frames` meaning device is in a deep sleep and is not replying to any requests.

I should have one of those and will try to reproduce.

SimplyGardner, 2020-01-13:
@Adminiuga Appreciate your reply - thank you.

I have a few Xiaomi Aqara sensors connected also and never have any issues with them. Just the ikea motion sensor.

Adminiuga, 2020-01-14:
It is not happening with Xbee
[image]

Going to try with Conbee II too

SimplyGardner, 2020-01-14:
@Adminiuga - Running on a Conbee II. I would see if you can leave it overnight - maybe face it towards to wall so it does not detect any movement. Mine is continuing to report as 'unavailable'.

[image]

Also, I noticed from your screenshot that ZHA brings the IKEA motion sensor in as a 'contact sensor' and reports the state as closed or open? That happened to me also...

SimplyGardner, 2020-01-14:
Just looking at the commits - wondering if #232 would fix this issue?

Adminiuga, 2020-01-15:
I was able to reproduce the issue with Ikea Motion E1525 Sensor
```
2020-01-15 20:00:15 DEBUG (SyncWorker_3) [homeassistant.components.zha.core.device] [0xa98f](TRADFRI motion sensor): Attempting to checkin with device - missed checkins: 1
2020-01-15 20:00:15 DEBUG (MainThread) [zigpy.device] [0xa98f] Extending timeout for 0x1f request
2020-01-15 20:00:15 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 32, 0, <DeconzAddressEndpoint address_mode=2 address=0xa98f endpoint=1>, 260, 0, 1, b'\x00\x1f\x00\x04\x00', 2, 0)
2020-01-15 20:00:22 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x20 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xa98f endpoint=1>, status: 0xf0
2020-01-15 20:00:22 DEBUG (MainThread) [zigpy.device] [0xa98f] Delivery error for seq # 0x1f, on endpoint id 1 cluster 0x0000: message send failure
2020-01-15 20:01:15 DEBUG (SyncWorker_1) [homeassistant.components.zha.core.device] [0xa98f](TRADFRI motion sensor): Attempting to checkin with device - missed checkins: 2
2020-01-15 20:01:15 DEBUG (MainThread) [zigpy.device] [0xa98f] Extending timeout for 0x21 request
2020-01-15 20:01:15 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 34, 0, <DeconzAddressEndpoint address_mode=2 address=0xa98f endpoint=1>, 260, 0, 1, b'\x00!\x00\x04\x00', 2, 0)
2020-01-15 20:01:22 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x22 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xa98f endpoint=1>, status: 0xf0
2020-01-15 20:01:22 DEBUG (MainThread) [zigpy.device] [0xa98f] Delivery error for seq # 0x21, on endpoint id 1 cluster 0x0000: message send failure
```

So there's no traffic from the device and device does not reply to the "ping" request in the form of attribute read. 
However if the sensor is triggered, then it reports immediately.

dmulcahey, 2020-01-15:
What did you have to do to reproduce it? What was different between this attempt and the previous attempt to reproduce it?

Adminiuga, 2020-01-15:
This time it was connected to Conbee II. I've connected it back to XBee and trying again. Maybe I haven't waited long enough last time, cause on Conbee it took about 3 hours to timeout. Probably because there was a battery report which reset the timer.

SimplyGardner, 2020-01-15:
I’m also have the ConBee II and it times out after about 3 hours too

Adminiuga, 2020-01-16:
So I don't think it is related to the radio. Same thing is happening with XBee
[image]

It only recovered after receiving the battery update. The device was not responding to check-in pings.

Kanishkaz, 2020-07-24:
I have this same issue with an IKEA Motion sensor connected to ZHA using a Conbee II. The device would become unavailable after a few hours or when HA is restarted. Triggering the sensor does not restablish a connection.
Only way to fix it is by removing the device and re-adding it.

The device also appear as unk_model by unk_manufacturer. I have plenty of Xiaomi sensors that do not have this issue.
Has there been any progress on this issue?

Cheers.

mguaylambert, 2020-08-27:
I have the same issue with the husbzb-1, if I leave it overnight with no motion, it becomes un-available. Is there better motion detection out there?

mguaylambert, 2020-08-29:
Another issue seems to be that when waking it up from deep sleep, it sends open but in Home Assistant, from unavailable to open does not seem to trigger automations. Not sure who’s fault here.

mguaylambert, 2020-09-02:
Something very interesting happened. One of my motion sensor decided to route thru a IKEA bulb instead of talking directly with the coordinator (HUSBZB-1) ans this one never get’s unavaillable. The other one is directly attached to the coordinator and get unavaillable for a period of time and then the sensor checks-in and shows availlable for 13hours and then goes back to unavaillable (i’m away of the home for the week).

dmulcahey, 2020-12-21:
@Adminiuga thoughts on what comes next here?

Adminiuga, 2020-12-21:
I now believe this was a case of ConBee not keeping its children in persistent memory. So children can talk to coordinator, but coordinator cannot talk to children, and therefore cannot ping device to confirm it is online.

This was fixed in zigpy_deconz and should not be an issue any longer.

dmulcahey, 2020-12-21:
Closing. Please reopen if the issue is not fixed.

closed 2020-12-21 by dmulcahey

MattWestb, 2020-12-22:
With EZSP (Ver 6.7.6.0) both old and new motion sensor is reported online (without motion detection reported) and both direct connected the the coordinator for some weeks (also working then there is connected thru one router).

Rudd-O, 2022-07-27:
The same happens with ZiGate (not ZiGate+) device and ZHA.  No motion events are ever reported.  But if I reset the device, I see "became unavailable" in the logbook.

I'm tossing this garbage sensor in the can.

grepis, 2022-11-03:
If someone find this post, checkin interval set in tradfri motion sensor to 172800 (or 12h). ZHA automatically mark device as unavailable after 6h. You can change checkin interval by writing 13200 (55m) into corresponding attribute:
cluster 0x0020, attribute 0.
Somehow build-in screen to write attributes doesn't work (I suspect type mismatch). Expected uint32. There are tow options:
- use re-configure button on screen. (Note: since no button to wake up device, I pulled battery and put it back after reconfigure startes).
- use zha-toolkit:
`service: zha_toolkit.attr_write
data:
  ieee: your_device ieee
  endpoint: 1
  cluster: 32
  attribute: 0
  attr_type: 35
  attr_val: 13200
`  
Still the same trick with pulled battery out is required.

MattWestb, 2022-11-04:
IKEA controllers normally need need being reconfigured after being paired or they is not reporting OK to the system but it not so easy to do then need waking it up as you have write.
If i remember right is ZHA setting up  `checkin interval` to 55 minutes then (and is not changing the `long pull interval` then we have taking it away for saving the batteries for all IKEA controllers and letting it as default in device).

I have one old (ZLL) and one new (ZB3) im my production system and the new is doing Checkin ever hour (the old ZLL is not supporting it but is also working OK).

```
IKEA motion sensor New Checkin event was fired
07:01:10 - 11 minutes ago
IKEA motion sensor New Checkin event was fired
06:04:25 - 1 hour ago
IKEA motion sensor New Checkin event was fired
05:07:41 - 2 hours ago
```
The motion sensor is pritty long in the system so have not testing how it  working adding them new with current version of ZHA but i think its over 1.5 year now then we was getting EZSP 6.7.8.0 firmware for our coordinators.

Also if the motion sensor is not having one Zigbee 3 router as parent i think the `checkin interval` is being set OK but its parent can doing strange things (it shall only forwarding the configured checkin addressed to the coordinator IEEE but it can being one general routing problem in the no ZB3 parent).