zigpy/zha-device-handlers #240: Question: Lights - Enable Smooth transition?

SimplyGardner, 2020-01-14:
Currently using some Hue bulbs with ZHA. Everything is working brilliantly.

When I manually change the brightness or colour of the bulbs, it instantly flips to the new setting without a period of transition. On Deconz and also the Hue bridge directly, when changing the brightness/colour it transitions from the current to the new setting smoothly.

Does anyone know if this is possible with ZHA?

I know you can do this on scenes etc but we do a lot of manual changes in our house.

Adminiuga, 2020-01-14:
well, in lovelace UI is there a field for "transition" during brighness/color changes?
If you want it for manual contol, you would need to modify card for the light control to include transition attribute. Otherwise it is already available for service calls.

SimplyGardner, 2020-01-14:
Thanks @Adminiuga. I am mainly controlling my lights from the HomeKit platform. HA exposes these lights to HomeKit for me.

SimplyGardner, 2020-01-14:
My lights support the fading. As a test I added another Hue bulb and called the ``light.turn_on`` service - incorporating the ``transition`` item.

How do I get the default behaviour of ZHA to always use a transition when changing the brightness or colour of lights? It uses a transition when turning them on and off - fading it and out smoothly.

[image]

dmulcahey, 2020-02-02:
Are you asking about when controlling from automations? or even when toggling in the UI? If it is the later we would need to change the light class to support this.

SimplyGardner, 2020-02-02:
Hi @dmulcahey - when controlling via the UI.

I have these lights also exposed to HomeKit via Home Assistant and changing the class would probably help to enable to smooth transition here also.

Adminiuga, 2020-02-03:
IMO this is a setting which can't make all users happy.  I think if not specified by the calling service, the light should turn on as fast as possible.

dmulcahey, 2020-02-03:
I agree @Adminiuga but I wanna investigate something. I think this setting can be stored on the bulbs. If so, we can default it to 0 / no transition and allow people to change it. If that’s the case we can change the light class to read this from the lights and use it as a default if it isn’t sent in the call. Does that make sense?

dmulcahey, 2020-02-03:
hrm, looks like the attrs are optional and the singled bulbs I have don't support them.

Adminiuga, 2020-02-03:
Can transition time be specified in lights profile?

Adminiuga, 2020-02-03:
Apparently not

tube0013, 2020-02-03:
FWIW The Osram bulbs I used to use would default to a 5 second transition. I could change it via the level control cluster and did it via an automation when the light came back from unavailable. No bulb since (innr, hue, echo smart or Ledvance) have supported modifying the transition time via cluster

SimplyGardner, 2020-02-03:
> @Adminiuga  IMO this is a setting which can't make all users happy.

Agreed - although integrating a smooth transition with ensure that ZHA is consistent with other platforms such a Deconz and Hue. They all raise the lights smoothly. 

> @dmulcahey  hrm, looks like the attrs are optional and the singled bulbs I have don't support them.

I have a bunch of Hue bulbs I am happy to test with - you may need to show me how haha

> @tube0013 I could change it via the level control cluster 

Are you saying you can edit the light to interstate the transition for the bulb in question? If so how? Haha. I don’t mind going through all my bulbs and manually tweaking

Adminiuga, 2020-02-04:
can https://github.com/home-assistant/home-assistant/pull/31452 help with this?

@SimplyGardner some lights support On/Off transition time attributes.
Go to Config -> Zha -> Pick your light -> Find Level Cluster. Write transition time in 1/10th of a second to:
- off transition time
- on transition time
- on/off transition time
[image]

SimplyGardner, 2020-02-26:
Hi @Adminiuga - thanks so much and apologies for the late reply. Only just getting round to testing this out but very excited to try.

I've added bulb to ZHA and the on transition and off transition is smooth. The issue is when changing the ``brightness`` or ``colour`` while the light is ``on``.

Looking at your screenshot... how do I add the items? I have attached a screenshot of what is in my system... What do I select in the ``Green`` box? 

Also what values do I add in the ``red`` and ``orange`` box?

[image]

miguelcaravantes, 2020-06-27:
the same issue i don't find the setting for change color smoothly as i did it with philips hue bridge

TheJulianJES, 2020-08-11:
It seems like that Home Assistant Core had a DEFAULT_TRANSITION of 0.5 seconds in the past. (https://github.com/home-assistant/core/pull/26654/files)
In the upcoming 0.114 release, the default transition seems to be 1/10th of a second (in order to fix a bug): https://github.com/home-assistant/core/pull/38739

However, I'd also love to see a default transition of one second (or half a second again) in Home Assistant to match other lights with in-built transitions. I think there should be a setting for the default durations somewhere in Hass. I'm not sure if this is relevant to this project though.

Adminiuga, 2020-08-11:
default transition was 0 --max speed for quite a few HA releases now, but apparently it is causing issue with some vendors so to improve the compatibility the default transition was changed to .1s. This is only applicable if you change the brightness or set duration, for plain on/off calls this doesn't do anything.

TheJulianJES, 2020-08-11:
Yeah, okay. I'd love to be able to change the default transition time (when changing colors) from the .1s to 1s. (Every other light I use has a default transition time of 1s (Philips Hue for example (also when used via deCONZ, I think.)))
Not sure how easy something like this could be implemented though (regarding setting the value yourself).

Adminiuga, 2020-08-11:
Do light profiles allow you to change the duration?

dmulcahey, 2020-08-11:
I wanna say that was just worked on recently...

TheJulianJES, 2020-08-11:
Yeah, this is already in the beta: https://rc.home-assistant.io/blog/2020/08/05/release-114/#other-noteworthy-changes
I'm not sure if this could be used to set a default transition for Hass though (?)

TheJulianJES, 2020-08-12:
```
id,x,y,brightness,transition
group.all_lights.default,0,0,255,5
```
Tried this in the latest beta - the transition of 5 seconds works when turning the light on (and the colors fades when changing the color). However, this only happens if the color changes happens in the turn_on initial 5s transition.
So, I guess a "default color fade" is currently not possible with ZHA and Home Assistant.

TheJulianJES, 2020-08-12:
Perhaps a "solution" to this "issue" could be to get the color profile and use that as a default transition duration (if present) here: https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L195
(Not sure about this though.)

TheJulianJES, 2020-08-18:
Today I tried my "custom implementation" of a one second transition between color/brightness changes on an up-to-date LEDVANCE Gardenpole light. It seems like this causes an "issue", when the brightness is changed, as the transition values of the brightness change are being displayed (for a short while) in the Home Assistant user interface.
So although this is not (and probably shouldn't be) a high priority for ZHA -- something like a transition setting on a per light basis might be a good idea for the future (this would also take some time and work to implement).

TheJulianJES, 2020-09-22:
> Today I tried my "custom implementation" of a one second transition between color/brightness changes on an up-to-date LEDVANCE Gardenpole light. It seems like this causes an "issue", when the [...]

On newer ZHA hue bulbs which now send attribute reports (ZLL models of Hue lights don't), the brightness change of (even the 0.1s) transition is displayed in the UI - making it feel "laggy".
The bug report on the Hass repo is here:
https://github.com/home-assistant/core/issues/40300

RefineryX, 2020-12-07:
Is there any update on this? My Phillips Hue bridge has broken and being forced to put all my lights through my Conbee II stick via ZHA.

I forgot about the issue with the changing of transitions via HomeKit. I would love to see the smooth transition with changing colours or brightness. 

On/off transition is absolutely fine.

RefineryX, 2020-12-08:
Just playing around in HA. The service call works with adding ``transition: 1``. Is this something that I can integrate permanently in the code somewhere?

```
entity_id: light.philips_lct010_7bfa4702_level_light_color_on_off
brightness_pct: 20
transition: 1
```

Adminiuga, 2020-12-08:
configure a light profile???

RefineryX, 2020-12-08:
> configure a light profile???

Would this work as a system wide setting for transitioning the brightness and colour changes? I.e, if I decided to change the brightness from Lovelace or HomeKit, the transition would be 1s or whatever I decide?

If so... do you have any guidance on how I would go about enabling that?

Adminiuga, 2020-12-08:
https://www.home-assistant.io/integrations/light/

Adminiuga, 2020-12-08:
but i haven't tried it

RefineryX, 2020-12-08:
Yes, I was able to search google and find the exact same link above. The question I was asking was how do I apply it to this use case? After reading the documentation, I am not sure I am understanding it correctly. It appears that the [light integration](https://www.home-assistant.io/integrations/light/) allows users to to track and control either when the light is switched on or switched off.

But I do not see anything for changing a light that is already on?

**On/Off transition works fine:** Currently, that already works fine and see that the light turns on/off with a smooth transition. No changes are required here.

**Transition when the bulb is already on:** The issue I am facing is when **changing the colour or brightness of a bulb that is already on**. I would like to keep all the settings exactly the same but just add a transition time of ``1s`` each time either the colour(+temp) or brightness is changed to give it a smooth transition. I think currently the default is set to ``0``.

TheJulianJES, 2020-12-08:
Comment in other thread for visibility here: https://github.com/zigpy/zha-device-handlers/issues/297#issuecomment-739910539

RefineryX, 2020-12-08:
Thanks. I am going to assume that the light_profiles does not work as you mentioned in your previous comment.

Trying to piece this together in my mind and thanks for bearing with me haha.

I am looking at the file you linked too (https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L195).

If I managed to find that file on my system, what would I update on there?
What would be the downsides of editing that file? Is there a risk that it would end up being overwritten with every release?

TheJulianJES, 2020-12-08:
It would need to be `else 10` instead of `else 1` at the end of the line. 1 is 1/10th of a second as the default transition time. You could download the Hass source (for the version that you're running) and put the whole zha folder in your custom components directory. Then modify the light.py file. If you update, you need to do this before updating if possible, as things will break sooner or later otherwise. Tbh I wouldn't recommend to do this.

Adminiuga, 2020-12-08:
per https://community.home-assistant.io/t/light-profiles-csv/91407/5 seems like you need to create the `light_profiles.csv` file in your config directory and have a line with `group.all_lights.default` entity ID with
```
group.all_lights.default,x_color_float_number,y_color_float_number,brightness_number,default_transition_number
```

RefineryX, 2020-12-08:
> put the whole zha folder in your custom components directory. Then modify the light.py file

@TheJulianJES thanks for the information and I agree with you, that seems pretty 'hacky' and a long route just to be able to change the transition of the lights. I have a range of Zigbee devices and concerned about keeping my own version in the customs_compotent folder.

> per https://community.home-assistant.io/t/light-profiles-csv/91407/5 seems like you need to create the `light_profiles.csv` file in your config directory and have a line with `group.all_lights.default` entity ID with

I am out of my depth with that as I really don't understand i. @TheJulianJES, you mentioned that this would not work, right?

RefineryX, 2020-12-08:
So I have created a ``light_profiles.csv`` file and added the below. I am going to test using my living room lamp first but I don't see any difference.
```
id,x,y,brightness,transition
light.living_room_lamp,0.5119,0.4147,144,1
```

Its a real shame that I have to specify x,y and brightness settings. Does this mean every time the lamp is turned on, it will default to those settings?

Adminiuga, 2020-12-08:
1. There should not be any headers
2. I did not test with individual lights
3. yes

RefineryX, 2020-12-08:
Hmmm, still doesn't work. I event put the transition to 15 seconds to test.

No worries anyways, appreciate both your both - thank you @Adminiuga and @TheJulianJES. 

I am going to switch everything over to Deconz this weekend.

Adminiuga, 2020-12-08:
I've tested with `group.all_lights.default` and can confirm it is working.

>light.living_room_lamp,0.5119,0.4147,144,15

This is supposed to be `light.living_room_lamp.default` 
[image]

The only downside, it would always do a color transition too. To fix that a PR is required.

Adminiuga, 2020-12-08:
I've tested it with both `group.all_lights.default` and `entity_id.default` and can confirm it is working as expected, but in the end it does depend on the device, whether it supports the color transition or not.

Test setup:
1. `light_profiles.csv` with the following content
```
light.ikea_of_sweden_tradfri_bulb_e26_ws_clear_806lm_f14a95fe_level_light_color_on_off.default,0.0,0.100,100,5
group.all_lights.default,0.343,0.159,250,5
light.ledvance_br30_rgbw_32c50000_level_light_color_on_off.default,0.,0.1,250,5
```
3. Ledvance Color bulb BR30 and GE45857 dimmer
4. Ledvance was turned on with red color using a service call, then turned off
5. Ledvance was turned on from UI here's content of the service call
[image]

and it corresponds to light profile settings for that specific entity. It smoothly increased brightenss and transitioned from red to blue as expected

6. GE45857 was set to on with brightness 7, then turned off using service call
7. GE45857 was turned on using UI and the service was called with
[image]

that corresponds to light profile default setting for all lights
and light correctly transitioned into specified state, but couldn't set the color, because it is just a simple dimmer
[image]

RefineryX, 2020-12-08:
Thanks. Yes, needed the ``.default``. But again, can I clarify... this is just for turning on and off the lights right? And, does not address my issue?

Adminiuga, 2020-12-08:
No, it does not.

RefineryX, 2020-12-08:
Hmm... okay, let's ignore that.

Let's leave this open incase anyone can think of a way to get this to work. Does anyone know how Deconz handle the transitions and how it works?  When requesting a change of brightness or colour when the lights are connected with Deconz, it works.

closed 2021-08-20 by github-actions[bot]