zigpy/zha-device-handlers #248: Can't add Xiaomi Switch

deltabweb, 2020-01-22:
I'm trying to add a Xiaomi switch as a device but can't get it to work

```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0x8302 (00:15:8d:00:02:b7:ca:5c) joined the network
[0x8302] Requesting 'Node Descriptor'
[0x8302] Extending timeout for 0x09 request
Device 0x8302 (00:15:8d:00:02:b7:ca:5c) joined the network
Skip initialization for existing device 00:15:8d:00:02:b7:ca:5c
[0x8302:zdo] ZDO request 0x0013: [0x8302, 00:15:8d:00:02:b7:ca:5c, 128]
Ignoring message (b'18000a050042126c756d692e73656e736f725f737769746368') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18010a0100200a') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'1c3412020a02ff4c0600100121c40b21a8012400000000002130042051') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18030a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18040a00001001') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18050a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18060a00001001') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18070a00001000') on cluster 6: unknown endpoint or cluster id: 1
Ignoring message (b'18080a00001001') on cluster 6: unknown endpoint or cluster id: 1
[0x8302] Delivery error for seq # 0x09, on endpoint id 0 cluster 0x0002: message send failure
[0x8302] Extending timeout for 0x0b request
```

Apparently the device is connected and the different messages shown are triggered by button presses but I am unable to add the button to Home Assistant.

Is anyone able to help / point me in the right direction? :)

Adminiuga, 2020-01-22:
it was not added correctly. Xiaomi are PITA. Remove device from ZHA config panel. Reset the device and re-pair it. Keep pressing the button on the device to keep it awake.

deltabweb, 2020-01-22:
Thanks, I managed to get it working after a few more tries.
I was just trying to reset the device and hope for the best. Turns out after the reset I had to press the link button a few times like you said (3 or 4) and each time new logs were added until it finally got added properly.

closed 2020-01-22 by deltabweb