zigpy/zha-device-handlers #250: Some pins is missed in XBee quirks.

MaximumSU, 2020-01-26:
/zhaquirks/xbee/__init__.py 
ENDPOINT_MAP = {
    0: 0xD0,
    1: 0xD1,
    2: 0xD2,
    3: 0xD3,
    4: 0xD4,
    5: 0xD5,
    7: 0xD7,
    8: 0xD8,
    9: 0xD9,
    10: 0xDA,
    11: 0xDB,
    12: 0xDC,
}
6 pin is missed;

ENDPOINT_TO_AT = {
    0xD0: "D0",
    0xD1: "D1",
    0xD2: "D2",
    0xD3: "D3",
    0xD4: "D4",
    0xD5: "D5",
    0xD8: "D8",
    0xD9: "D9",
    0xDA: "P0",
    0xDB: "P1",
    0xDC: "P2",
}
D6 and D7 pinns is missed. Why? Is it some serious reason to do that?

Adminiuga, 2020-01-26:
Reserved for RTS/CTS?

Adminiuga, 2020-01-26:
tagging @Shulyaka

MaximumSU, 2020-01-26:
> 
> 
> Reserved for RTS/CTS?

You may configure it as Input/Output pins in XCTU, it is not denied. Almost all pins in XBee can be configured for something else except of Input/Output. Why that 2 pins must be excluded?

Shulyaka, 2020-01-26:
Hi @MaximumSU, what XBee device do you have and with which firmware?

MaximumSU, 2020-01-26:
> 
> 
> Hi @MaximumSU, what XBee device do you have and with which firmware?

XBee S2C 
Product Family: XB24C
Function Set: ZIGBEE TH Reg
Firmware version: 4061

MaximumSU, 2020-01-26:
I checked the  zigpy/zigpy-xbee  also - it is even worst :-( 
D6, D7, D9, D10, D11, D12, D13, D14 in api.py not exists.

Shulyaka, 2020-01-26:
I can see no problem adding them, but zigpy-xbee have to do it first

MaximumSU, 2020-01-26:
> I can see no problem adding more pins, but zigpy-xbee have to add the corresponding commands it first

I already opened the issue in zigpy-xbee. Only one quetion for me not cleare - IR IO Sampling Rate. For IC we have bitmask, fut for IR - no. From the Help of XCTU:

> Set the IO sampling rate to enable periodic sampling. If set >0, all enabled digital IO and analog inputs will be sampled and transmitted every IR milliseconds.

How do you think - they mean, what pin must be not disabled (any role from the list), or it assigned to Input/Output role only?

Shulyaka, 2020-01-26:
> How do you think - they mean, what pin must be not disabled (any role from the list), or it assigned to Input/Output role only?

If a pin is configured as digital i/o, its logical value will be sent in the sample report. If its role is not digital i/o (RSSI PWM, analog input, etc) its status will not be reported. That's what they mean.

MaximumSU, 2020-01-26:
> all enabled digital IO and analog inputs will be sampled 
So, analog inputs also sampled...

Shulyaka, 2020-01-26:
Sorry, I was incorrect. Analog inputs are also sampled

MaximumSU, 2020-01-28:
> I checked the [zigpy/zigpy-xbee](https://github.com/zigpy/zigpy-xbee/issues/78) also - it is even worst :-(
> D6, D7, D9, D10, D11, D12, D13, D14 in api.py not exists.

I'm sorry, it is not correct. Only D9 and D14 is missing in api, all another pins available. I expected to see, for example, "D10" but it is named "P0" in AT commands, so i'm not detected them.

MaximumSU, 2020-01-30:
The changes for api.py was [commited](https://github.com/zigpy/zigpy-xbee/commit/09fd091a618d382fc4d05757fa554b93ebbb6996).

PR #256 (2020-01-31): XBee additional io pins
Expose additional IO pins for XBee.
Fixes #250
closed 2020-01-31 by dmulcahey