zigpy/zha-device-handlers #254: AQARA switch  WXKG03LM 2016 no events

albalaing, 2020-01-28:
Existe un problema con el interruptor AQARA WXKG03LM 2016 que se identifica como lumi.sensor_86sw1. Se da de alta sin problemas y utiliza el Quirk zhaquirks.xiaomi.aqara.remote_b186acn01.
Pero cuando lo pulso no consigo que dectecte nada. He observado que este interrupto utiliza el cluster 6 OnOff para notificar que se ha pulsado. Y aprecen los siguientes mensajes:

[zigpy_deconz.uart] Frame received: 0x17a10031002a00220200000104a7823dca8301008d150001040106000b00182f0a000010010000100000afefa3000102c3
[zigpy_deconz.api] APS data indication response: [42, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x82a7>, 1, 260, 6, b'\x18/\n\x00\x00\x10\x01\x00\x00\x10\x00', 0, 175, 239, 163, 0, 1, 2, -61]
[zigpy.device] Ignoring message (b'182f0a0000100100001000') on cluster 6: unknown endpoint or cluster id: 'No cluster ID 0x0006 on (00:15:8d:00:01:83:ca:3d, 1)'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x82a7>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'182f0a0000100100001000'
He intentado añadir el cluster 6 OnOff al Quirk pero no lo he conseguido.

tehspede, 2020-03-01:
Hi! I seem to have the same problem, no events under zha_event.

closed 2020-03-01 by dmulcahey

radu022003, 2023-06-13:
using the Sonoff Zigbee 3.0 USB Dongle Plus V2 I can pair my 2 WXKG03LM devices but when I press the button there is no zha_event

PR #279 (2020-03-01): Add support for older Aqara WXKG03LM and WXKG02LM
Add OnOff to OUTPUT_CLUSTERS to support older devices that send this command on click. Fixes #254