zigpy/zha-device-handlers #259: Ikea motion sensor E1745 - no motion events received in Home Assistant 0.105

mck182, 2020-01-31:
I have RPi4 with HUSBZB-1 stick and I've successfully paired the E1745 motion sensor with Home Assistant in docker, v0.105. However the sensor remains in `closed`, no matter what.

When I wave my hand in front of the sensor, I can see this in the log:
```
homeassistant    | 2020-01-31 21:15:43 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1754b1ed562e14b45995b529ab5592dbc4e17b641231648aedc6ead97e'
homeassistant    | 2020-01-31 21:15:43 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
homeassistant    | 2020-01-31 21:15:43 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401060001ff0c0100009258af5ccfffff03011000'
homeassistant    | 2020-01-31 21:15:43 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=6 sourceEndpoint=1 destinationEndpoint=255 options=268 groupId=0 sequence=146>, 88, -81, 0xcf5c, 255, 255, b'\x01\x10\x00']
homeassistant    | 2020-01-31 21:15:43 DEBUG (MainThread) [zigpy.zcl] [0xcf5c:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=0>
homeassistant    | 2020-01-31 21:15:43 DEBUG (MainThread) [zigpy.zcl] [0xcf5c:1:0x0006] ZCL request 0x0000: []
homeassistant    | 2020-01-31 21:15:43 DEBUG (MainThread) [zigpy.zcl] [0xcf5c:1:0x0006] No handler for cluster command 0
homeassistant    | 2020-01-31 21:15:45 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
homeassistant    | 2020-01-31 21:15:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'725521adcf737e'
homeassistant    | 2020-01-31 21:15:45 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2055a1ad4cb87e'
homeassistant    | 2020-01-31 21:15:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
homeassistant    | 2020-01-31 21:15:45 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
```

But the state doesn't change in Home Assistant, nor does any automation trigger.

What could be the issue here? I'm happy to provide more logs, just tell me what to do.

dmulcahey, 2020-01-31:
Please provide a screenshot of the device card from the ZHA config panel

mck182, 2020-01-31:
[image]

dmulcahey, 2020-02-01:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1676

Looks like deconz got this working. Will need to look through the code and I’ll probably have to get one of these to get it to work correctly. I’ll let you know when I make progress.

Adminiuga, 2020-02-02:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1676#issuecomment-520787141
Maybe it just needs a binding?

Adminiuga, 2020-02-02:
Similar to opple switch?

dmulcahey, 2020-02-02:
something like that. They were able to get it to work on join... I just have to dig into the code a bit.

mck182, 2020-02-06:
So few days later, I've reinstalled Home Assistant in Docker, readded the motion sensor and now it seems to work as expected, HA shows proper state on motion and I can use it for automations. I'll keep a close eye on this and I'll reopen this with more details. Thanks for the support.

closed 2020-02-06 by mck182