zigpy/zha-device-handlers #265: IKEA 2-button switch (square) - shows only as a battery

pleszczynski, 2020-02-05:
I have three IKEA 2-button square switches (2 bought separately, one with a bulb). They all discover without problems, but all show only as battery sensors.

Two of them register events in zha_event correctly, third one does not. 

Sample events:
```
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "ec:1b:bd:ff:fe:a3:9c:13:1:0x0008",
        "device_ieee": "ec:1b:bd:ff:fe:a3:9c:13",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-02-05T11:24:19.201131+00:00",
    "context": {
        "id": "49f098a00ea54cb9a0e4d46c2b5550e1",
        "parent_id": null,
        "user_id": null
    }
}
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "ec:1b:bd:ff:fe:a3:9c:13:1:0x0008",
        "device_ieee": "ec:1b:bd:ff:fe:a3:9c:13",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_with_on_off",
        "args": [
            0,
            83
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-02-05T11:24:16.955182+00:00",
    "context": {
        "id": "7b68c363664242c798687c06f64c0eaa",
        "parent_id": null,
        "user_id": null
    }
}
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "ec:1b:bd:ff:fe:a3:9c:13:1:0x0006",
        "device_ieee": "ec:1b:bd:ff:fe:a3:9c:13",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-02-05T11:24:13.671841+00:00",
    "context": {
        "id": "3aa38a204a444835861b26ea77291115",
        "parent_id": null,
        "user_id": null
    }
}
```

Discovery log
[tradfri13.log](https://github.com/dmulcahey/zha-device-handlers/files/4159203/tradfri13.log)

From what I understand, this should work as of 0.0.31, that's exactly what I'm using.

pleszczynski, 2020-02-05:
I fixed the one that did not fire events by pairing it again. Now all of them emit events, but none show as sensor apart from battery level.

dmulcahey, 2020-02-05:
They are remotes you will not get sensors for them just the events. Use the events or device triggers for your automations.

closed 2020-02-05 by dmulcahey