zigpy/zha-device-handlers #266: WXKG01LM - Xiaomi Long Press

bryantlee, 2020-02-07:
Hi team!

Looking to see if it was possible to support long press options for the button.  It appears the button does not send explicit long press events, but perhaps Aquara hubs actually just calculate the difference between button press and release.

Zigbee2Mqtt appears to also do this as well.
https://www.zigbee2mqtt.io/devices/WXKG01LM.html

Any way to support this within Quirks?

Adminiuga, 2020-02-07:
>## How to use device type specific configuration
>__long_timeout__: The WXKG01LM only reports a button press and release. By default, Zigbee2mqtt publishes a long click when there is at least 1000 ms between both events. It could be that due to delays in the network the release message is received late. This causes a single click to be identified as a long click. If you are experiencing this you can try experimenting with this option (e.g. long_timeout: 2000).

bryantlee, 2020-02-13:
@Adminiuga - just making sure your response wasnt to me.  Do you think this is desirable functionality since it exists in zigbee2mqtt?

Adminiuga, 2020-02-14:
Just confirming that it is not explicit button/zigbee functionality.

peregrinus, 2020-03-17:
Hi @Adminiuga apologies for the ignorance but how do I utilise the 'long_timeout' option in home assistant. I don't see this in the ZHA config options for that device. Thanks

dmulcahey, 2020-03-17:
He was referencing z2m docs that explain that they implemented a software trick to support this.

peregrinus, 2020-03-17:
thanks for the quick reply David. ZHA is excellent and I appreciate the effort to date. hoping this becomes the standard as it's very stable with my purely Xiaomi environment post decommissioning my Xiaomi hubs.  triple click replaced my long click but would be nice to see ZHA offer the same down the track if possible.

closed 2021-08-16 by github-actions[bot]