zigpy/zha-device-handlers #267: Add support for Develco Products A/S EMIZB-132 Norwegian HAN adapter

jorgen666, 2020-02-08:
I have such adapter installed and it shows up in HA v0.105.2. But it have clearly no suitable device handler. 
This is an adapter to read data from the norwegian energy meters.

how to contribute on making a handler for this device ?

This is the tecnical manual for this adapter.
[emizb-132-technical-manual-emi-norwegian-han.pdf](https://github.com/dmulcahey/zha-device-handlers/files/4173623/emizb-132-technical-manual-emi-norwegian-han.pdf)

TheDiverM, 2020-02-20:
I also have the same adapter. And also wonder what can be done to make this work

prilly-dev, 2020-12-27:
I realy hope we can get support for this device, seems alot of Norwegian users want it. 

Develco has very good documentation on their site for this device, it should be trivial to make a quirk for it.

prilly-dev, 2021-01-19:
Deconz have now support for this device, changing from zha to deconz is the recommend way.

MattWestb, 2021-01-20:
Then the devs dont have the device they dont have so easy to doing one implanting of it the ZHA then its one little unusual device and need much information and handon for implanting it.

Its the same with all not so standard devices like the "LIDL christmas tree light" that was coming before christmas and we was not able getting one to the devs for getting it implanted so we was not getting any Zigbee christmas tree this year :-((

Digging / harvesting information of the device can helping and also if  you is knowing some with knowledge of python (or other programing skills) that can doing little hand on can helping getting all pieces in place.

Its the same with [deCONZ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3824) have not implanting one device for the scandinavian countries (KNYCKLAN Open/Close remote (E1841)) but one user have putting all possible information in one request on ther github and its pretty standard device with minimal cluster changes (compared to the "normal" E1743).

And good luck with your recommendation to XdeCONZ users.

prilly-dev, 2021-01-20:
Looking up this device with its producer you will find very good and complete documentation to implement it: https://www.develcoproducts.com/products/

There is no need for a dev to have this device to construct a device handler for it. Iam shure there are many users willing to test a finished device handler. By the way its not possible to use this device or test against it without a propper norwegian HAN enabled electricity meter.

This device is fairly complicated, writing device handler for it is not that straight forword. But the deconz guys did it with the provided documentation available on the producers webpage

Hedda, 2021-05-19:
> This is the tecnical manual for this adapter.
> [emizb-132-technical-manual-emi-norwegian-han.pdf](https://github.com/dmulcahey/zha-device-handlers/files/4173623/emizb-132-technical-manual-emi-norwegian-han.pdf)

According to manual, it uses Zigbee SE Metering Cluster id 0x0702 so might be issue with same ZHA limitations as in ->  https://github.com/zigpy/zha-device-handlers/issues/605

Also, not quite sure zigpy which ZHA depends on support DLMS/COSEM and Zigbee SEP (Zigbee Smart Energy Profile) specification?

https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf

https://community.smartthings.com/t/dlms-and-zigbee-smart-energy-profile-sep-1-1/143411

PS: Develco Products also have another product with SKU: SMMZB-310101 which looks to use same Zigbee specifications:

https://www.develcoproducts.com/products/meter-interfaces/prosumer-meter/

https://zigbeealliance.org/zigbee_products/prosumer-meter/

https://zigbeealliance.org/no/zigbee_Produkter/prosumer-meter/

Hedda, 2021-05-19:
Develco also has two other product with SKU: SMMZB-310101 which looks to use the same Zigbee Smart Energy specifications:

- **Develco Smart Plug Mini** (Frient SPLZB-131)

https://github.com/zigpy/zha-device-handlers/issues/857

https://www.develcoproducts.com/products/smart-plugs/smart-plug-mini/

- **Develco Prosumer Meter** (SMMZB-310101)

https://www.develcoproducts.com/products/meter-interfaces/prosumer-meter/

https://zigbeealliance.org/zigbee_products/prosumer-meter/

https://zigbeealliance.org/no/zigbee_Produkter/prosumer-meter/

MattWestb, 2021-05-19:
The problem is that in the Zigbee 3 HA profile 0x0051 "HA Smart Plugs" shall having SE Metering Cluster id 0x0702 without implanting all of Smart Energies.

Hedda, 2021-05-19:
> The problem is that in the Zigbee 3 HA profile 0x0051 "HA Smart Plugs" shall having SE Metering Cluster id 0x0702 without implanting all of Smart Energies.

Developers discussion in https://github.com/zigpy/zha-device-handlers/issues/605 and https://github.com/home-assistant/core/pull/50471 also points to a limitation in ZHA integration for Home Assistant which currently only support 1 sensor per cluster, and these Zigbee Smart Energy meter devices need 2 or more sensor per cluster.

Hedda, 2021-10-05:
Test with latest Home Assistant beta or the next upcoming release:

https://github.com/home-assistant/core/pull/56909

https://github.com/home-assistant/core/pull/56666

closed 2022-04-11 by github-actions[bot]