zigpy/zha-device-handlers #268: Strange XBee switch behaver

MaximumSU, 2020-02-08:
I have one XBee as coordinator, connected to Rpi with HA, and second XBee as router. They are paired without problems, i have all GPIO pins as switches in HA. But, when i disconnected the router from power, i had at first: all states got to unavailable state after some time and begining to "blinking" ~40 seconds "unavailable", ~20 seconds - last states (the states was before i disconnected the router). When i connected the router back - the same behaver - "blinking" states. Then i rebooted HA and got last states back - no unavailable states anymore. And after that, when i disconnected the router from power, i alway have last states, no more "unavailable" states. I was wait more then a hour - no effect. Router is disconnected, last states is here.

dmulcahey, 2020-03-28:
@prairiesnpr @Shulyaka any thoughts?

Adminiuga, 2020-03-28:
Should see the debug log including for homeassistant.core, + zha ,+ zhaquirks

MaximumSU, 2020-03-28:
> Should see the debug log

It is not possible now. Many changes. I close the issue because i do't have additional information.

closed 2020-03-28 by MaximumSU