zigpy/zha-device-handlers #270: IKEA 5 Button Remote Issue

causefx, 2020-02-11:
Radio: Nortek GoControl QuickStick Combo Model HUSBZB-1 (Z-Wave & Zigbee USB Adapter)
```
arch | x86_64
-- | --
dev | false
docker | true
hassio | false
os_name | Linux
python_version | 3.7.6
timezone | America/Phoenix
version | 0.105.0
virtualenv | false
```
I see that 0.105.3 is out, I can update to that.  Also I'm unsure about `Enable debug logging for the radio lib` - Not quite sure how to enable debug logs for the radio itself.  

I'm also unable to use this ikea remote...
Bellows shows the correct endpoints:
```
Device:
  NWK: 0x9462
  IEEE: ec:1b:bd:ff:fe:8f:ea:7f
  Endpoints:
    1: profile=0x104, device_type=DeviceType.NON_COLOR_CONTROLLER
      Input Clusters:
        Basic (0)
        Power Configuration (1)
        Identify (3)
        Poll Control (32)
        LightLink (4096)
        Manufacturer Specific (64636)
      Output Clusters:
        Identify (3)
        Groups (4)
        On/Off (6)
        Level control (8)
        Ota (25)
        LightLink (4096)
```
I am getting zero zha_events for this remote.

Only thing i can see is that there is a "Unsupported ZDO request:0x8021" on the setup logs:
```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x8ce4:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=26 command_id=Command.Report_Attributes>
[0x8ce4:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1328>>]]
[0x8ce4:1:0x0402] Attribute report received: measured_value=1328
Device 0x683b (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
[0x683b] Requesting 'Node Descriptor'
[0x683b] Extending timeout for 0x2f request
Device 0x9462 (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
Device ec:1b:bd:ff:fe:8f:ea:7f changed id (0x683b => 0x9462)
Canceling old initialize call
[0x9462] Requesting Node Descriptor failed: 
[0x9462] Discovering endpoints
[0x9462] Extending timeout for 0x31 request
[0x9462] Requesting 'Node Descriptor'
[0x9462] Extending timeout for 0x33 request
Device 0x9462 (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
Skip initialization for existing device ec:1b:bd:ff:fe:8f:ea:7f
[0x9462:zdo] ZDO request 0x0013: [0x9462, ec:1b:bd:ff:fe:8f:ea:7f, 128]
[0x9462] Discovered endpoints: [1]
[0x9462:1] Discovering endpoint information
[0x9462] Extending timeout for 0x35 request
[0x9462] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x9462] Discovering endpoints
[0x9462] Extending timeout for 0x37 request
[0x9462:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 4096]>
[0x9462] Extending timeout for 0x39 request
[0x9462] Discovered endpoints: [1]
[0x9462:1] Discovering endpoint information
[... 73 lines cut ...]
[0x9462:1:0x0000]: channel: 'async_initialize' stage succeeded
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>
[0x9462] Extending timeout for 0x57 request
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Read_Attributes_rsp>
[0x9462] Extending timeout for 0x59 request
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=89 command_id=Command.Read_Attributes_rsp>
[0x9462] Extending timeout for 0x5b request
[0x9462:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=91 command_id=Command.Read_Attributes_rsp>
[0x9462:1:0x0001]: initializing channel: from_cache: False
[0x9462:1:0x0001]: channel: 'async_initialize' stage succeeded
[0x9462](TRADFRI remote control): power source: Battery or Unknown
[0x9462](TRADFRI remote control): completed initialization
[0x9462] Extending timeout for 0x5d request
[0xee15:5:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=240 command_id=Command.Report_Attributes>
[0xee15:5:0x0001] ZCL request 0x000a: [[<Attribute attrid=32 value=<TypeValue type=uint8_t, value=30>>]]
[0xee15:5:0x0001] Attribute report received: battery_voltage=30
[0x05db:1:0x0b04]: async_update
[0x05db:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=95 command_id=Command.Read_Attributes_rsp>
[0x6ea7:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x6ea7:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1780>>]]
[0x6ea7:1:0x0402] Attribute report received: measured_value=1780
[0x8ce4:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=27 command_id=Command.Report_Attributes>
[0x8ce4:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1340>>]]
[0x8ce4:1:0x0402] Attribute report received: measured_value=1340
```

Adminiuga, 2020-02-11:
hrm, you do have `bellows` log enabled. I think it is HomeAssistant is filtering the log. Add debug logging for: `bellows.ezsp: debug` (or call `loggers.set_level` service to set debug log for bellows.ezsp) rejoin the device and grab the raw `home-assistant.log` file.

causefx, 2020-02-11:
[home-assistant - Copy.log](https://github.com/dmulcahey/zha-device-handlers/files/4188065/home-assistant.-.Copy.log)
Here is the full log after pairing it again

Adminiuga, 2020-02-11:
upgrade to 0.105.3 there were issues storing model info.

I need to see why this is causing async_configure error
```
2020-02-11 13:07:59 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=4096 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=167>, 248, -67, 0x79e8, 255, 255, b'\x19\x91A\x01\x00\x00']
2020-02-11 13:07:59 DEBUG (MainThread) [zigpy.zcl] [0x79e8:1:0x1000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=145 command_id=65>
2020-02-11 13:07:59 WARNING (MainThread) [homeassistant.components.zha.core.channels] [0x79e8:1:0x1000]: channel: 'async_configure' stage failed ex: list index out of range
```

dmulcahey, 2020-02-11:
It’s the custom light link cluster. I have to adjust how it works.

causefx, 2020-02-11:
@Adminiuga I had upgraded to 0.105.3 prior to sending the logs.

Adminiuga, 2020-02-11:
yeah, it the payload response doesn't make any sense:
`b'\x19\x91A\x01\x00\x00' == b'199141010000'`
0x19 -- frame control
0x91 -- TSN
0x41 -- Command ID -- Get Group identifiers response
0x01 -- Total number of Groups
0x00 -- Starting index
0x00 -- number of groups in the list

So it tells there's 1 group but return empty list
[image]

causefx, 2020-02-12:
Let me know if I can help in any way.

dmulcahey, 2020-02-19:
can you test #274 when you have a moment

causefx, 2020-02-19:
know of any good way to clone that branch from home-assistant docker image?

dmulcahey, 2020-02-19:
Negative

bwynants, 2020-02-19:
#274 fixes my pairing issue! HA automations work now

dmulcahey, 2020-02-19:
Thanks for confirming! fixed by #274

andyzickler, 2020-02-19:
@causefx You can do something like this:

```
# Run bash in the homeassistant container
docker exec -it homeassistant bash

cd /usr/src/homeassistant/
# Install dev branch of zha-device-handlers
pip3 install --upgrade git+https://github.com/dmulcahey/zha-device-handlers.git@dev
```

Then, restart Home Assistant. Note that this change will be reverted when you update Home Assistant normally.

causefx, 2020-02-26:
I tested on 0.106.0 that has zhaquirks 0.33 and it still doesn't work. 
I unpaired and restart hass and paired again.

[home-assistant - Copy.log](https://github.com/dmulcahey/zha-device-handlers/files/4258005/home-assistant.-.Copy.log)

martinlong1978, 2020-02-27:
This also doesn't work for me with zhaquirks 0.33. 

All buttons on the remote generate zha_event events, except for the two side (scene) buttons. These no longer generate any events at all. It has been broken for some time - I think it was last working in 103, that I remember.

Adminiuga, 2020-02-28:
@causefx did you bind that remote to any groups?

causefx, 2020-02-28:
@Adminiuga I have not - I was under the assumption (bad on me) that it would work like the `Adurolight_NCC` device.  So I am only able to bind ZHA groups to this remote?  I have not made any zha groups as the only one listed is `No name group 0x000`

dmulcahey, 2020-02-28:
@causefx it should work the same way. I’ll check the scene buttons tomorrow.

causefx, 2020-02-28:
sounds good - thank you

bwynants, 2020-02-28:
Mmm for me this fixed it and I have access to all 5 buttons

causefx, 2020-02-28:
@bwynants I believe there are 2 different variants of the 5 button remote.

martinlong1978, 2020-02-28:
I've just removed and readded the remote, same issue. In fact, same issue with all 3 of my 5 button remotes. Middle button, and top and bottom buttons work exactly as expected, and generate the expected zha_events. Side buttons do absolutely nothing. This was definitely working in a previous version, before all of the new group magic was added.

dmulcahey, 2020-02-28:
Can someone having this issue give me 2 things? A clean pairing log for the device and debug traffic from the button pushes that are no longer working? 

IKEA has been changing device signatures with firmware updates. The groups should have nothing to do with this.

martinlong1978, 2020-02-28:
> Can someone having this issue give me 2 things? A clean pairing log for the device and debug traffic from the button pushes that are no longer working?
> 
> IKEA has been changing device signatures with firmware updates. The groups should have nothing to do with this.

I'll get on that over the weekend. Thanks. Would HA have updated the firmware on these devices? I haven't had them connected to an IKEA hub or anything.

Adminiuga, 2020-02-28:
Log from causefx from two days ago. here's a snip from lightlink config. It does the same weird thing Ikea does lately

```
2020-02-26 13:14:54 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=4096 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=242>, 255, -67, 0x31cd, 255, 255, b'\x19wA\x01\x00\x00']
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.zcl] [0x31cd:1:0x1000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=119 command_id=65>
2020-02-26 13:14:54 WARNING (MainThread) [zhaquirks.ikea] unable to locate required group info - falling back to group 0x0000.
2020-02-26 13:14:54 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'333921d62bc09a4ca62951c9abda2e7e'
2020-02-26 13:14:54 DEBUG (MainThread) [bellows.ezsp] Send command setMulticastTableEntry: (0, <EmberMulticastTableEntry multicastId=0x0000 endpoint=1 networkIndex=0>)
2020-02-26 13:14:54 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x31cd, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=123>, 124, b'\x00{\x06\x00 \x00 \x10\x0e0*\x01')
2020-02-26 13:14:54 DEBUG (MainThread) [bellows.ezsp] Application frame 100 (setMulticastTableEntry) received: b'00'
2020-02-26 13:14:54 DEBUG (MainThread) [bellows.multicast] Set MulticastTableEntry #0 for 0x0000 multicast id: [<EmberStatus.SUCCESS: 0>]
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.group] Adding group: 0, None
2020-02-26 13:14:54 INFO (MainThread) [homeassistant.components.zha.core.group] [No name group 0x0000](0x0000): group_added
2020-02-26 13:14:54 INFO (MainThread) [homeassistant.components.zha.core.group] [No name group 0x0000](0x0000): group_member_added - endpoint: <bellows.zigbee.application.EZSPCoordinator.EZSPEndpoint object at 0x7f9c56b7e350>
```

Quirk:
```
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.endpoint] [0x31cd:1] Manufacturer: IKEA of Sweden
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.endpoint] [0x31cd:1] Model: TRADFRI remote control
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for IKEA of Sweden TRADFRI remote control (ec:1b:bd:ff:fe:8f:ea:7f)
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremote.IkeaTradfriRemote'>
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.quirks.registry] Fail because profile_id mismatch on at least one endpoint
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote'>
2020-02-26 13:14:54 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for ec:1b:bd:ff:fe:8f:ea:7f: <class 'zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote'>
```

So NCP was successfully added to group `0x0000` which is confirmed by receiving event for the other two buttons. 

Full log.

```
2020-02-26 13:14:52 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [5, <Bool.false: 0>, 0xb7b1, ec:1b:bd:ff:fe:8f:ea:7f, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2020-02-26 13:14:52 DEBUG (MainThread) [bellows.ezsp] Application frame 35 (childJoinHandler) received: b'0501cd317fea8ffeffbd1bec04'
2020-02-26 13:14:52 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [5, <Bool.true: 1>, 0x31cd, ec:1b:bd:ff:fe:8f:ea:7f, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2020-02-26 13:14:53 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5526b18c991b6a58d66ab598b1b993499c4e797a7e'
2020-02-26 13:14:53 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2020-02-26 13:14:53 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received: b'cd317fea8ffeffbd1bec01000000'
2020-02-26 13:14:53 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0x31cd, ec:1b:bd:ff:fe:8f:ea:7f, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2020-02-26 13:14:53 INFO (MainThread) [zigpy.application] Device 0x31cd (ec:1b:bd:ff:fe:8f:ea:7f) joined the network
2020-02-26 13:14:53 DEBUG (MainThread) [zigpy.application] Device ec:1b:bd:ff:fe:8f:ea:7f changed id (0xb7b1 => 0x31cd)
2020-02-26 13:14:53 DEBUG (MainThread) [zigpy.device] Canceling old initialize call
2020-02-26 13:14:53 WARNING (MainThread) [zigpy.device] [0x31cd] Requesting Node Descriptor failed: 
2020-02-26 13:14:53 INFO (MainThread) [zigpy.device] [0x31cd] Discovering endpoints
2020-02-26 13:14:53 DEBUG (MainThread) [zigpy.device] [0x31cd] Extending timeout for 0x65 request
2020-02-26 13:14:53 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for ec:1b:bd:ff:fe:8f:ea:7f/0x31cd
2020-02-26 13:14:53 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (ec:1b:bd:ff:fe:8f:ea:7f, True)
2020-02-26 13:14:53 INFO (MainThread) [zigpy.device] [0x31cd] Requesting 'Node Descriptor'
2020-02-26 13:14:53 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'562721d62bc09a4ca62951c9abca9e7e'
2020-02-26 13:14:53 DEBUG (MainThread) [zigpy.device] [0x31cd] Extending timeout for 0x67 request
[... cut ...]
2020-02-26 13:15:05 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x31cd:1:0x0001]: initializing channel: from_cache: False
2020-02-26 13:15:05 DEBUG (MainThread) [homeassistant.components.zha.core.channels] [0x31cd:1:0x0001]: channel: 'async_initialize' stage succeeded
2020-02-26 13:15:05 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x31cd](TRADFRI remote control): power source: Battery or Unknown
2020-02-26 13:15:05 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x31cd](TRADFRI remote control): completed initialization
```

dmulcahey, 2020-03-05:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2485#issuecomment-590126057

we have to account for this...

martinlong1978, 2020-03-05:
Do you know if it's possible, in the meantime, to roll back the firmware
upgrade? Does HASS automatically apply the firmware upgrades provided by
Ikea?

>

Adminiuga, 2020-03-05:
You have to manually enable Ikea ota updates in zigpy. By default it is not enabled. Currently it is not possible to rollback. To rollback you need to get older version of it's image and make some changes to the code.

martinlong1978, 2020-03-05:
Hmm... so if I have to manually turn on firmware updates, then I'm still a
bit confused as to why this stopped working. It should be using the
original firmware, right?

>

Adminiuga, 2020-03-05:
If you did not manually enabled IKEA OTA updates, then yes, it still should be working.
Unless you tried to bind this remote to some light groups. 

Does it still work with the old HA versions? 
Logs?

martinlong1978, 2020-03-05:
So, i just installed 0.104.0 and it is definitely working in that version.

I've just tried the side buttons and got these events.

{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:81:64:8e:1:0x0005",
        "device_ieee": "14:b4:57:ff:fe:81:64:8e",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "press",
        "args": [
            257,
            13,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-05T17:46:47.119848+00:00",
    "context": {
        "id": "715a43d6f9ab4a37af82e4a054f6ef73",
        "parent_id": null,
        "user_id": null
    }
}
Event 1 fired 5:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "unique_id": "14:b4:57:ff:fe:81:64:8e:1:0x0005",
        "device_ieee": "14:b4:57:ff:fe:81:64:8e",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "press",
        "args": [
            256,
            13,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-05T17:46:41.281349+00:00",
    "context": {
        "id": "107144af59564048a90a9764f1686b3f",
        "parent_id": null,
        "user_id": null
    }
}

I'll see what logs I can pull out. Any particular logging config which
you're interested in? (I know that full logging generates a lot of noise).

>

martinlong1978, 2020-03-05:
So if I set the debugging to:
homeassistant.core: debug
homeassistant.components.zha: debug

I get absolutely nothing logged when I press the side buttons.

>

martinlong1978, 2020-03-05:
If I run a "reconfigure" on one of the remotes, I get this in the logs:

2020-03-05 18:35:23 DEBUG (MainThread)
[homeassistant.components.zha.api] Reconfiguring node with
ieee_address: 14:b4:57:ff:fe:81:64:8e
2020-03-05 18:35:23 DEBUG (MainThread)
[homeassistant.components.zha.core.device] [0x3e3b](TRADFRI remote
control): started configuration
2020-03-05 18:35:23 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:ZDO](TRADFRI
remote control): 'async_configure' stage succeeded
2020-03-05 18:35:32 DEBUG (MainThread) [homeassistant.core]
Bus:Handling <Event state_changed[L]: entity_id=climate.heating,
old_state=<state climate.heating=auto; hvac_modes=['auto', 'heat',
'off'], min_temp=5, max_temp=32, preset_modes=['none', 'boost'],
current_temperature=21.8, temperature=32.0, hvac_action=heating,
preset_mode=boost, availability=Online, battery_level=100%,
friendly_name=Heating, supported_features=17 @
2020-03-05T18:14:49.578810+00:00>, new_state=<state
climate.heating=auto; hvac_modes=['auto', 'heat', 'off'], min_temp=5,
max_temp=32, preset_modes=['none', 'boost'], current_temperature=21.9,
temperature=32.0, hvac_action=heating, preset_mode=boost,
availability=Online, battery_level=100%, friendly_name=Heating,
supported_features=17 @ 2020-03-05T18:14:49.578810+00:00>>
2020-03-05 18:35:32 DEBUG (MainThread) [homeassistant.core]
Bus:Handling <Event state_changed[L]: entity_id=climate.ethan_s_room,
old_state=<state climate.ethan_s_room=auto; hvac_modes=['auto',
'heat', 'off'], min_temp=5, max_temp=32, preset_modes=['none',
'boost'], current_temperature=19.9, temperature=19.5,
hvac_action=idle, preset_mode=None, mode=SCHEDULE,
friendly_name=Ethan's Room, supported_features=17 @
2020-03-05T18:14:49.575507+00:00>, new_state=<state
climate.ethan_s_room=auto; hvac_modes=['auto', 'heat', 'off'],
min_temp=5, max_temp=32, preset_modes=['none', 'boost'],
current_temperature=20.0, temperature=19.5, hvac_action=idle,
preset_mode=None, mode=SCHEDULE, friendly_name=Ethan's Room,
supported_features=17 @ 2020-03-05T18:14:49.575507+00:00>>
2020-03-05 18:35:40 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x0b36:1:0x0008]:
received attribute: 0 update with value: 254
2020-03-05 18:35:55 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0001]:
Failed to bind 'power' cluster: TXStatus.ADDRESS_NOT_FOUND
2020-03-05 18:36:22 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0000]:
bound 'basic' cluster: Status.SUCCESS
2020-03-05 18:36:22 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0000]:
finished channel configuration
2020-03-05 18:36:24 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0001]:
reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1:
Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
2020-03-05 18:36:24 INFO (MainThread) [zigpy_xbee.zigbee.application]
handle_rx self addressed
2020-03-05 18:36:24 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x1000]:
bound 'lightlink' cluster: None
2020-03-05 18:36:24 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x1000]:
finished channel configuration
2020-03-05 18:36:25 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0000]:
initializing channel: from_cache: False
2020-03-05 18:36:26 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0008]:
bound 'level' cluster: Status.SUCCESS
2020-03-05 18:36:26 DEBUG (MainThread)
[homeassistant.components.zha.core.channels.base] [0x3e3b:1:0x0008]:
finished channel configuration
[... 355 lines cut ...]
current_temperature=20.0, temperature=7.0, hvac_action=idle,
preset_mode=None, mode=SCHEDULE, friendly_name=Office,
supported_features=17 @ 2020-03-05T18:14:49.583499+00:00>>
2020-03-05 18:38:32 DEBUG (MainThread) [homeassistant.core]
Bus:Handling <Event state_changed[L]:
entity_id=climate.master_bedroom, old_state=<state
climate.master_bedroom=auto; hvac_modes=['auto', 'heat', 'off'],
min_temp=5, max_temp=32, preset_modes=['none', 'boost'],
current_temperature=19.0, temperature=18.5, hvac_action=idle,
preset_mode=None, mode=SCHEDULE, friendly_name=Master Bedroom,
supported_features=17 @ 2020-03-05T18:14:49.590059+00:00>,
new_state=<state climate.master_bedroom=auto; hvac_modes=['auto',
'heat', 'off'], min_temp=5, max_temp=32, preset_modes=['none',
'boost'], current_temperature=19.1, temperature=18.5,
hvac_action=idle, preset_mode=None, mode=SCHEDULE,
friendly_name=Master Bedroom, supported_features=17 @
2020-03-05T18:14:49.590059+00:00>>

On Thu, 5 Mar 2020 at 18:25, Long, Martin <martin@longhome.co.uk> wrote:
>
> So if I set the debugging to:
> homeassistant.core: debug
> homeassistant.components.zha: debug
>
> I get absolutely nothing logged when I press the side buttons.

Adminiuga, 2020-03-05:
What kind of radio? HUSBZB or Deconz?

When you installed 0.104, you just installed it and device worked without you rejoining or re configuring it, right?

martinlong1978, 2020-03-05:
Yes, just installed 0.104 and it worked straight away.

I'm using a Digi Xbee radio

Adminiuga, 2020-03-05:
And with `homeassistant.core: debug` you definitely should see zha_events in the logs for homeassistant_core.

For XBee, enable debug for:
`zigpy_xbee.zigbee.application: debug`
`zigpy_xbee.api: debug`

And for information, can you read "sw_build_id" of the basic cluster for that remote? may have to press some buttons on it, to wake it up.

martinlong1978, 2020-03-05:
So, with that debug enabled I do get the following when pressing the side buttons: 

`2020-03-05 20:51:23 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-05 20:51:23 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (ff:ff:ff:ff:ff:ff:ff:ff, 0x3e3b, 230, 5, 2, b'057c114a0700010d00')
2020-03-05 20:51:25 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-05 20:51:25 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (ff:ff:ff:ff:ff:ff:ff:ff, 0x3e3b, 230, 5, 2, b'057c114b0701010d00')
2020-03-05 20:51:30 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-05 20:51:30 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (ff:ff:ff:ff:ff:ff:ff:ff, 0x3e3b, 230, 5, 2, b'057c114c0700010d00')
2`

But definitely no zha_events generated. 

The sw_build_id of the basic cluster is: 1.2.223

martinlong1978, 2020-03-05:
`2020-03-05 20:57:20 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-05 20:57:20 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (14:b4:57:ff:fe:81:64:8e, 0x3e3b, 230, 5, 2, b'057c115d0700010d00')
2020-03-05 20:57:25 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-05 20:57:25 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (14:b4:57:ff:fe:81:64:8e, 0x3e3b, 230, 5, 2, b'057c115e0701010d00')
2020-03-05 20:57:28 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-05 20:57:28 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (14:b4:57:ff:fe:81:64:8e, 0x3e3b, 230, 5, 2, b'057c115f0700010d00')`

Don't know why the first lot didn't seem to have the devices address in them. These do.

Adminiuga, 2020-03-05:
Hi Martin,
Gonna ask you more questions. We're trying to pinpoint the problem
1. can you please use Debug log level for this components (you are using debug for some of them already, just make sure all below have debug level):
- `homeassistant.core`
- `homeassistant.components.zha`
- `zigpy: debug`
- `zigpy_xbee.zigbee.application`
- `zigpy_xbee.api`

2. Do you have quirks applied to youer device? should be able to tell in zha config panel -> device 
3. it works on 104, but does not work on 106 and 105?
4. confirm model and manufacturer as reported device panel

martinlong1978, 2020-03-06:
> @dmulcahey <https://github.com/dmulcahey> i guess the model is TRADFRI
> remote control based on

Sorry. Please bear with me. I might not get a chance to sit down and get
you answers until Sunday. I'll get back to you as soon as I can.

dmulcahey, 2020-03-06:
No problem. I’m gonna pull the one I have in about an hour to see what I can see...

causefx, 2020-03-06:
After seeing all the chatter today I decided to try the remote again today.  I subscribed to the zha_event to test and lo and behold everything was working.... no clue wtf happened...  All 5 buttons work and reporting all 9 events.  

I didn't try and re-add it or reconfigure it - I just literally pushed the button and it worked.  I'm still on HA version | 0.106.0 
```
IKEA of Sweden TRADFRI remote control
TRADFRI remote control
by IKEA of Sweden
IEEE: ec:1b:bd:ff:fe:8f:ea:7f
Nwk: 0x31cd
Device Type: EndDevice
LQI: 255
RSSI: -70
Last Seen: 2020-03-05T17:03:54
Power Source: Battery or Unknown
Quirk: zhaquirks.ikea.fivebtnremotezha.IkeaTradfriRemote
```

dmulcahey, 2020-03-06:
I quit 🙃

David F. Mulcahey

Adminiuga, 2020-03-06:
@causefx for posterity, post sw_build_id of the basic cluster for your remote

Adminiuga, 2020-03-06:
I'm getting on ver

```
 address=0x0c6c>, 1, 260, 5, b'\x05|\x11\x1b\x07\x01\x01\r\x00', 0, 175, 255, 48, 218, 11, 0, -55]
2020-03-06 01:59:41 DEBUG (MainThread) [zigpy.zcl] [0x0c6c:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=27 command_id=7>
2020-03-06 01:59:41 DEBUG (MainThread) [zigpy.zcl] [0x0c6c:1:0x0005] ZCL request 0x0007: [257, 13, 0]
2020-03-06 01:59:41 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0c6c>, ep: 1, profile: 0x0104, cluster_id: 0x0005, data: b'057c111b0701010d00'
...
2020-03-06 02:00:30 DEBUG (MainThread) [zigpy.zcl] [0x0c6c:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=28 command_id=7>
2020-03-06 02:00:30 DEBUG (MainThread) [zigpy.zcl] [0x0c6c:1:0x0005] ZCL request 0x0007: [256, 13, 0]
2020-03-06 02:00:30 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0c6c>, ep: 1, profile: 0x0104, cl
```

This is on 2.3.014 sw_build_id

Adminiuga, 2020-03-06:
and i can confirm events work with 0.104.3

Adminiuga, 2020-03-06:
>Sorry. Please bear with me. I might not get a chance to sit down and get you answers until Sunday. I'll get back to you as soon as I can.

No worries. And you have tried it with 0.106 right? not the dev tree?

Adminiuga, 2020-03-06:
at least in my tests, 0.106.4 works fine with 2.3.014 remote version. NO events on dev tree.

causefx, 2020-03-06:
@Adminiuga 
[image]

Adminiuga, 2020-03-06:
Yep, it should be working in 0.106.x and my guess Martin was trying it in dev. We should have a fix for it.

causefx, 2020-03-06:
awesome. you guys are the best. thanks.

martinlong1978, 2020-03-06:
I was indeed using it in dev, as I have some of my own patches in there.

I *thought* it had stopped working when I was using 0.105.x but my memory
is hazy so I'll have to check that. I'm out teaching tonight and tomorrow,
but hopefully will get a chance when I'm back.

lkroeker, 2020-03-07:
I’m a bit of a newb at this stuff, but I’m trying to get up to speed quickly.

Got a couple new Ikea Tradfri wireless dimmer on/off buttons (E1743) and when I added them to ZHA all that’s showing up is the battery status, which is not useful. In the logs I get “No handler for cluster command 0” (also, 1, 5, 7 depending on which button, or long/short press)

Through ZHA I was able to bind the switch to another zigbee device (a plug) and it was working fine. I’m just missing a step where I can get the button presses to be sent into HA as events. 

Hopefully this is helpful? (below is the output as I added one of the button devices to ZHA)
This is running on HA 0.106.5

`[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x9d79:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=195 command_id=Command.Report_Attributes>
[0x9d79:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>]]
[0x9d79:1:0x0006] Attribute report received: on_off=Bool.false
Device 0x4195 (14:b4:57:ff:fe:7b:a5:30) joined the network
[0x4195] Requesting 'Node Descriptor'
[0x4195] Extending timeout for 0x9a request
Device 0x7a4b (14:b4:57:ff:fe:7b:a5:30) joined the network
Device 14:b4:57:ff:fe:7b:a5:30 changed id (0x4195 => 0x7a4b)
Canceling old initialize call
[0x7a4b] Requesting Node Descriptor failed:
[0x7a4b] Discovering endpoints
[0x7a4b] Extending timeout for 0x9c request
[0x7a4b] Requesting 'Node Descriptor'
[0x7a4b] Extending timeout for 0x9e request
Device 0x7a4b (14:b4:57:ff:fe:7b:a5:30) joined the network
Skip initialization for existing device 14:b4:57:ff:fe:7b:a5:30
[0x7a4b:zdo] ZDO request 0x0013: [0x7a4b, 14:b4:57:ff:fe:7b:a5:30, 128]
[0x7a4b] Discovered endpoints: [1]
[0x7a4b:1] Discovering endpoint information
[0x7a4b] Extending timeout for 0xa0 request
[0x7a4b] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x7a4b] Discovering endpoints
[0x7a4b] Extending timeout for 0xa2 request
[0x7a4b:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 9, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 258, 4096]>
[0x7a4b] Extending timeout for 0xa4 request
[0x7a4b] Discovered endpoints: [1]
[0x7a4b:1] Discovering endpoint information
[0x7a4b] Extending timeout for 0xa6 request
Ignoring message (b'18a401040000420e494b4541206f662053776564656e050000421554524144465249206f6e2f6f666620737769746368') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on (14:b4:57:ff:fe:7b:a5:30, 1)'
[0x7a4b:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2080 device_version=1 input_clusters=[0, 1, 3, 9, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 258, 4096]>
[0x7a4b] Extending timeout for 0xa8 request
[0x7a4b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=1>
[0x7a4b:1:0x0006] ZCL request 0x0001: []
[0x7a4b:1:0x0006] No handler for cluster command 1
[0x7a4b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=168 command_id=Command.Read_Attributes_rsp>
[0x7a4b:1] Manufacturer: IKEA of Sweden
[0x7a4b:1] Model: TRADFRI on/off switch
Checking quirks for IKEA of Sweden TRADFRI on/off switch (14:b4:57:ff:fe:7b:a5:30)
Considering <class 'zhaquirks.ikea.twobtnremote.IkeaTradfriRemote2Btn'>
Found custom device replacement for 14:b4:57:ff:fe:7b:a5:30: <class 'zhaquirks.ikea.twobtnremote.IkeaTradfriRemote2Btn'>
device - 0x7a4b:14:b4:57:ff:fe:7b:a5:30 entering async_device_initialized - is_new_join: True
device - 0x7a4b:14:b4:57:ff:fe:7b:a5:30 has joined the ZHA zigbee network
[0x7a4b](TRADFRI on/off switch): started configuration
[0x7a4b:ZDO](TRADFRI on/off switch): channel: 'async_configure' stage succeeded
[0x7a4b] Extending timeout for 0xaa request
[0x7a4b] Extending timeout for 0xac request
[0x7a4b] Extending timeout for 0xae request
[0x7a4b:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x7a4b:1:0x0000]: finished channel configuration
[0x7a4b] Extending timeout for 0xb0 request
[0x7a4b:1:0x1000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=174 command_id=65>
[0x7a4b:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x7a4b] Extending timeout for 0xb2 request
[0x7a4b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=0>
[0x7a4b:1:0x0006] ZCL request 0x0000: []
[0x7a4b:1:0x0006] No handler for cluster command 0
[0x7a4b:1:0x1000]: bound 'lightlink' cluster: None
[0x7a4b:1:0x1000]: finished channel configuration
[0x7a4b:1:0x1000]: channel: 'async_configure' stage succeeded
[0x7a4b] Extending timeout for 0xb4 request
[... 52 lines cut ...]
[... cut ...]
[0x7a4b:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=17 command_id=5>
[0x7a4b:1:0x0008] ZCL request 0x0005: [0, 83]
[0x7a4b:1:0x0008] No handler for cluster command 5
[0x7a4b:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=18 command_id=7>
[0x7a4b:1:0x0008] ZCL request 0x0007: []
[0x7a4b:1:0x0008] No handler for cluster command 7`

dmulcahey, 2020-03-07:
will be fixed on dev by #292 and https://github.com/home-assistant/core/pull/32546

closed 2020-03-07 by dmulcahey

martinlong1978, 2020-03-08:
Brilliant!... almost. 

So now, with this applied, I do get the events from the scene buttons, however, I am getting 2 of each (same for both buttons)

`2020-03-08 15:45:25 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\xff\xff\xff\xff\xff\xff\xff\xff>;\x01\xe6\x00\x05\x01\x04\x02\x05|\x115\x07\x01\x01\r\x00'
2020-03-08 15:45:25 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-08 15:45:25 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (ff:ff:ff:ff:ff:ff:ff:ff, 0x3e3b, 230, 5, 2, b'057c11350701010d00')
2020-03-08 15:45:25 DEBUG (MainThread) [zigpy.zcl] [0x3e3b:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=53 command_id=7>
2020-03-08 15:45:25 DEBUG (MainThread) [zigpy.zcl] [0x3e3b:1:0x0005] ZCL request 0x0007: [257, 13, 0]
2020-03-08 15:45:25 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=14:b4:57:ff:fe:81:64:8e, unique_id=14:b4:57:ff:fe:81:64:8e:1:0x0005, endpoint_id=1, cluster_id=5, command=press, args=[257, 13, 0]>
2020-03-08 15:45:25 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=14:b4:57:ff:fe:81:64:8e, unique_id=14:b4:57:ff:fe:81:64:8e:1:0x0005, endpoint_id=1, cluster_id=5, command=press, args=[257, 13, 0]>
(END)`

Please let me know if you need anything else.

dmulcahey, 2020-03-08:
You need the recent changes in dev too

martinlong1978, 2020-03-08:
Excellent. I had tried up update both components to dev, but something hadn't taken. Thanks all for your help, and great quick resolution to this.