zigpy/zha-device-handlers #271: Smartthings button connection drops

kebel87, 2020-02-16:
Hi, 

first time I'm reporting a bug and I'm fairly new to Home assistant. I recently bought smartthings button and they works for a couple of minutes, then drops for no reason.

Here is all the info I could gather. Is it enough to spot/understand what's wrong? 

Thanks

----

arch | x86_64
-- | --
dev | false
docker | true
hassio | true
os_name | Linux
python_version | 3.7.6
timezone | America/Toronto
version | 0.105.4
virtualenv | false

I'm using **Nortek Zigbee/zwave stick (husbzb-1)**

Silicon Labs EZSP
EZSP
by Silicon Labs
IEEE:
00:0d:6f:00:11:f5:e1:93
Nwk:
0x0000
Device Type:
Coordinator
LQI:
255
RSSI:
Unknown
Last Seen:
2020-02-16T12:07:04
Power Source:
Mains
Quirk:
bellows.zigbee.application.EZSPCoordinator

-----

Here's the ZHA info for smartthings button 

Samjin button
button
by Samjin
IEEE:
28:6d:97:00:01:0d:9f:0b
Nwk:
0xbc46
Device Type:
EndDevice
LQI:
255
RSSI:
-70
Last Seen:
2020-02-16T11:52:04
Power Source:
Battery or Unknown
Quirk:
zhaquirks.samjin.button.SamjinButton

----

Here's the log of when I paired it : 
[... 120 lines cut ...]
[0x8f8a:1:0x0500]: Updated alarm state: 1
[0x8f8a] Extending timeout for 0x52 request
[0x8f8a:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=82 command_id=Command.Read_Attributes_rsp>
[0x8f8a:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=5 status=Status.SUCCESS value=button>]]
`

----

Here is some log entry copied from HA log 

[0xbc46:1:0x0500]: channel: 'async_configure' stage failed ex:
11:52 AM components/zha/core/channels/__init__.py (WARNING) - message first occurred at 11:30 AM and shows up 2 times

----

Sun Feb 16 2020 08:39:09 GMT-0500 (heure normale de l’Est nord-américain)
Failed ZDO request during device initialization: [0x8a1d:0:0x0005]: Message send failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 89, in _initialize
    epr = await self.zdo.Active_EP_req(self.nwk, tries=3, delay=2)
  File "/usr/local/lib/python3.7/site-packages/zigpy/util.py", line 88, in retry
    r = await func()
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 187, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x8a1d:0:0x0005]: Message send failure

Adminiuga, 2020-02-16:
according to the log, the Samjin Button with `ieee = 28:6d:97:00:01:0d:9f:0b` and `NWK = 0x8f8a`, joined and initialized normally.

There was some other device `Device 0x1566 (00:15:8d:00:02:27:27:29) left the network` deciding to leave the network and these messages
```
[0xbc46:1:0x0500]: channel: 'async_configure' stage failed ex:
11:52 AM components/zha/core/channels/init.py (WARNING) - message first occurred at 11:30 AM 
and shows up 2 times
...
Failed ZDO request during device initialization: [0x8a1d:0:0x0005]: Message send failure
```

Are from unrelated devices.

kebel87, 2020-02-17:
0xbc46 is indeed the SmartThings button as of 11h52 this morning. 0x8f8a was earlier this morning when I first paired it. I’ve deleted and paired it again, which explains the network name change.

However I’m clueless as to why it drops from network. :/

dmulcahey, 2020-02-19:
this isn't a quirks issue. if you are still experiencing an issue I would suggest seeking help in the zigbee channel on discord or on the forums.

closed 2020-02-19 by dmulcahey