zigpy/zha-device-handlers #276: Keen Home Smart Vent Models

starcza, 2020-03-01:
I've been having problems with the Keen Home Smart Vent Quirks and realized that there are additional models that need the DoublingPowerConfigurationCluster on them. I validated that the following manufacturer/models work when added but am unable to submit the change myself.

("Keen Home Inc", "SV01-410-MP-1.1")
("Keen Home Inc", "SV01-410-MP-1.0")
("Keen Home Inc", "SV01-410-MP-1.5")
("Keen Home Inc", "SV02-410-MP-1.3")

Adminiuga, 2020-03-01:
[Zigbee ID: "SV02-410-MP-1.3"; "SV02-610-MP-1.3"](https://zigbee.blakadder.com/Keen_Home_SV02.html)

[Zigbee ID: "SV01-410-MP-1.0'"; "SV01-410-MP-1.1"; "SV01-410-MP-1.4"; "SV01-410-MP-1.5"; "SV01-412-MP-1.0"; "SV01-610-MP-1.0"; "SV01-612-MP-1.0"](https://zigbee.blakadder.com/Keen_Home_SV01.html)

dmulcahey, 2020-03-01:
Do the signatures match for all of them?

PR #278 (2020-03-01): Add additional keen vent models
Fixes #276
closed 2020-03-01 by dmulcahey