zigpy/zha-device-handlers #291: Plaid Systems Spruce Irrigation Soil Moisture Sensor - Battery Not working

tube0013, 2020-03-06:
(https://www.amazon.com/Plaid-Systems-Spruce-Irrigation-Moisture/dp/B07TYJCG1J)

This device joins fine, Temperature worked immediately, Moisture did not report until I removed and repaired it (https://support.spruceirrigation.com/knowledge-base/moisture-data-not-reporting/)

but the battery just shows 0.

[image]

Initial joining log: https://paste.ubuntu.com/p/WHRvt5v5PM/

Battery Size: CR123A

closed 2020-03-09 by dmulcahey

tube0013, 2020-03-12:
this is still not working completely

[image]

The attribute reporting looks like this:
```2020-03-11 21:50:46 DEBUG (MainThread) [zigpy.zcl] [0xe9e1:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=3049>>]]
2020-03-11 21:50:46 DEBUG (MainThread) [zigpy.zcl] [0xe9e1:1:0x0001] Attribute report received: mains_voltage=3049``

Adminiuga, 2020-03-12:
Did you connect it to mains, since it's reporting 'mains' voltage?

tube0013, 2020-03-12:
Nope It’s got a battery only. I think it doesn’t follow spec/do the right thing.

PR #294 (2020-03-09): Add Plaid Systems Spruce Irrigation soil moisture sensor
fixes: #291