zigpy/zha-device-handlers #295: Add support lumi.ctrl_neutral1, Aqara QBKG04LM

LEJOUI, 2020-03-07:
This wall switch, without neutral, does come up in ZHA, but it hasn't the proper controls at this point. It does come with the attributes as in the picture below. However sending commands to the switch doesn't work at this point.
I'd like to code it myself, but I don't have any experience with it and my first attempt testing it local also failed.
[image]

The endpoints discovered by ZHA are:
```
<SimpleDescriptor endpoint=1 profile=260 device_type=6 device_version=2 input_clusters=[0, 3, 1, 2, 25, 10] output_clusters=[0, 10, 25]> 
<SimpleDescriptor` endpoint=2 profile=260 device_type=256 device_version=2 input_clusters=[16, 6, 4, 5] output_clusters=[]>
<SimpleDescriptor endpoint=3 profile=260 device_type=256 device_version=2 input_clusters=[16, 6, 4, 5] output_clusters=[]
<SimpleDescriptor endpoint=4 profile=260 device_type=0 device_version=2 input_clusters=[18, 6] output_clusters=[]>
Discovered endpoint information: <SimpleDescriptor endpoint=5 profile=260 device_type=0 device_version=2 input_clusters=[18, 6] output_clusters=[]>
<SimpleDescriptor endpoint=5 profile=260 device_type=0 device_version=2 input_clusters=[18, 6] output_clusters=[]>
Discovered endpoint information: <SimpleDescriptor endpoint=6 profile=260 device_type=0 device_version=2 input_clusters=[18, 6] output_clusters=[]>
```

Node descriptor:
`Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=132 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=1 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>`

Support in deconz: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356

Zigbee2mqtt:
`    {
        zigbeeModel: ['lumi.ctrl_ln1.aq1', 'lumi.ctrl_ln1'],
        model: 'QBKG11LM',
        vendor: 'Xiaomi',
        description: 'Aqara single key wired wall switch',
        supports: 'on/off, power measurement',
        fromZigbee: [
            fz.QBKG04LM_QBKG11LM_state, fz.QBKG11LM_power, fz.QBKG04LM_QBKG11LM_operation_mode,
            fz.QBKG11LM_click,
            fz.ignore_multistate_report, fz.xiaomi_power,
        ],
        toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode],
        endpoint: (device) => {
            return {'system': 1};
        },`

sanyatuning, 2020-03-25:
I have the same problem but the endpoints are incorrect. (I think)
I got these: (I tried to implement the handler without success)
```
            # <SimpleDescriptor endpoint=1 profile=260 device_type=6
            # device_version=2
            # input_clusters=[0, 3, 1, 2, 25, 10]
            # output_clusters=[0, 10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    PowerConfigurationCluster.cluster_id,
                    DeviceTemperature.cluster_id,
                    Ota.cluster_id,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Basic.cluster_id, Time.cluster_id, Ota.cluster_id,],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=256
            # device_version=2
            # input_clusters=[16, 6, 4, 5]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    BinaryOutput.cluster_id,
                    OnOff.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=3 profile=260 device_type=256
            # device_version=2
            # input_clusters=[16, 6, 4, 5]
            # output_clusters=[]
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    BinaryOutput.cluster_id,
                    OnOff.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=4 profile=260 device_type=0
            # device_version=2
            # input_clusters=[18, 6]
            # output_clusters=[]>
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [MultistateInput.cluster_id, OnOff.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=5 profile=260 device_type=0
            # device_version=2
            # input_clusters=[18, 6]
            # output_clusters=[]>
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [MultistateInput.cluster_id, OnOff.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=6 profile=260 device_type=0
            # device_version=2
            # input_clusters=[18, 6]
            # output_clusters=[]>
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [MultistateInput.cluster_id, OnOff.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=8 profile=260 device_type=83
            # device_version=2
            # input_clusters=[12]
            # output_clusters=[]>
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [AnalogInput.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
```

zigbee2mqtt send this when I turn the swich on
```
# zigbee-herdsman:controller:endpoint Command 0x00158d00024be541/2 genOnOff.on({},
#   {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false})
# zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest -
#   {"dstaddr":65311,"destendpoint":2,"srcendpoint":1,"clusterid":6,"transid":17,"options":0,"radius":30,"len":3,
#       "data":{"type":"Buffer","data":[1,8,1]}}
# zigbee-herdsman:adapter:zStack:unpi:writer -->frame [254,13,36,1,31,255,2,1,6,0,17,0,30,3,1,8,1,201]
```
and send this when turn off
```
# zigbee-herdsman:controller:endpoint Command 0x00158d00024be541/2 genOnOff.off({},
#   {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false})
# zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest -
#   {"dstaddr":65311,"destendpoint":2,"srcendpoint":1,"clusterid":6,"transid":16,"options":0,"radius":30,"len":3,
#       "data":{"type":"Buffer","data":[1,7,0]}}
# zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,31,255,2,1,6,0,16,0,30,3,1,7,0,198]
```

I also have some const for events
```
# click attr 0xF000
# single click 0x3FF1F00
# double click 0xCFF1F00
```

sanyatuning, 2020-03-25:
Based on the zigbee2mqtt log I created a test, but don't know how to fix.
```
    def test_ctrl_neutral():
        sec = 8
        ieee = 0
        nwk = 1234
        data = b"\x01\x08\x01"
        # data = b'\x00\x10\x02"\xff \x12'
        cluster = 6
        src_ep = 1
        dst_ep = 2
    
        app = mock.MagicMock()
        app.get_sequence = mock.MagicMock(return_value=sec)
        rep = Device(app, ieee, nwk)
        rep.add_endpoint(1)
        rep.add_endpoint(2)
        dev = CtrlNeutral1(app, ieee, nwk, rep)
        dev.request = mock.MagicMock()
        dev[2].in_clusters[cluster].command(1)
    
>       assert dev.request.call_args == call(None, cluster, src_ep, dst_ep, sec, data, expect_reply=True)
E       AssertionError: assert call(None, 6,...ct_reply=True) == call(None, 6,...ct_reply=True)
E         At index 0 diff: (None, 6, 2, 2, 8, b'\x01\x08\x01') != ''
E         Right contains one more item: {'expect_reply': True}
E         Full diff:
E         - call(None, 6, 2, 2, 8, b'\x01\x08\x01', expect_reply=True)
E         ?                  ---
E         + call(None, 6, 1, 2, 8, b'\x01\x08\x01', expect_reply=True)
E         ?               +++

```

sanyatuning, 2020-03-30:
Hey @LEJOUI 
I made it work.
See the pull request

dmulcahey, 2020-03-30:
fixed by #310

closed 2020-03-30 by dmulcahey

LEJOUI, 2020-04-23:
@sanyatuning Do you also experience that after a few minutes, about 10, without activity the switch can’t be controlled anymore? It does report state. But sending on and off doesn’t work.