zigpy/zha-device-handlers #297: Add transition length when the color changes

kvvoff, 2020-03-10:
I already raised this problem in **zigbee-herdsman-converters**, and Koen Kanters (Koenkk) fixed it. At zigbee2mqtt, the light turns on smoothly and the colors change smoothly too. How to implement this in ZHA?

https://github.com/Koenkk/zigbee-herdsman-converters/issues/551

dmulcahey, 2020-03-10:
I'll peek at what they did when I have some free time. I have some other things on the burner atm.

dmulcahey, 2020-03-13:
This needs device specific options in ZHA in HA. I'll investigate a way to do this. It's going to be a bit though because there are some more pressing things to implement at the moment.

TheJulianJES, 2020-08-11:
Linking this here: https://github.com/zigpy/zha-device-handlers/issues/240#issuecomment-672030092

RefineryX, 2020-12-07:
Just found this issue via the comment above. Any updates on this @dmulcahey?

dmulcahey, 2020-12-07:
Try using light profiles in HA

RefineryX, 2020-12-07:
Thanks. Would this work as a system wide setting for transitioning the brightness and colour changes? I.e, if I decided to change the brightness from Lovelace or HomeKit, the transition would be 1s or whatever I decide?

If so, and sorry to ask... do you have any guidance on how I would go about enabling that?

TheJulianJES, 2020-12-07:
As far as I know, this still doesn’t apply when changing the color of a light that’s already on. (You also need to specify a color which you may not want.) One workaround would be to put the default transition time in a constant variable again, so it can be overridden by a custom component.

RefineryX, 2020-12-07:
Ah dang. I see that you commented on my original issue.... what did you end up doing to enable the transition for colour and brightness? This you use the  light profiles as suggested by David or edit the /zha/light.py file?

TheJulianJES, 2020-12-07:
For a while I used ZHA as a custom component, but this obviously breaks all the time. For now, I’ve resorted to accepting it if I can’t explicitly set the transition time.

RefineryX, 2020-12-07:
Thanks. I am going to spend this upcoming weekend transferring my lights over to ZHA. I will wait it out for a bit and see if the feature (hopefully) gets released where it allows a smooth transition. If not, I've heard Deconz allows it but not yet fully tried it.

closed 2021-08-13 by github-actions[bot]