zigpy/zha-device-handlers #298: GLEDOPTO GL-W-001Z Does not update status (on or off)

kvvoff, 2020-03-10:
I have successfully completed pairing, but it does not update its status when I physically manage. From HA it turns on and off without problems, but when I turn it on or off physically, I can’t see the status. Such a problem was in zigbee2mqtt with this switch for a very long time, but it was successfully fixed. In ZHA status is still not updated

https://www.zigbee2mqtt.io/devices/GL-W-001Z.html

dmulcahey, 2020-03-10:
will take a look at what they did when I have a bit of free time.

dmulcahey, 2020-03-13:
```
 {
        zigbeeModel: ['GL-W-001Z'],
        model: 'GL-W-001Z',
        vendor: 'Gledopto',
        description: 'Zigbee ON/OFF Wall Switch',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        onEvent: async (type, data, device) => {
            // This device doesn't support reporting.
            // Therefore we read the on/off state every 5 seconds.
            // This is the same way as the Hue bridge does it.
            if (type === 'stop') {
                clearInterval(store[device.ieeeAddr]);
            } else if (!store[device.ieeeAddr]) {
                store[device.ieeeAddr] = setInterval(async () => {
                    try {
                        await device.endpoints[0].read('genOnOff', ['onOff']);
                    } catch (error) {
                        // Do nothing
                    }
                }, 5000);
            }
        },
    },
```

so the device doesn't support reporting... we don't have a per device way to poll yet in ZHA and until we do we won't be able to support this. I'll start thinking about the best way to handle this.

kvvoff, 2020-03-13:
@dmulcahey 
Apparently yes, looking at the log, I see that zigbee2mqtt constantly polls it
[image]

dmulcahey, 2020-03-26:
@Adminiuga should we do the same thing we did for Hue here? Or should we start to explore device specific configuration like what is in the code snippet I posted above from z2m?

Adminiuga, 2020-03-26:
Is this  a remote or  a wall switch? Do we have a pairing log?

If it really doesn't support reporting, then should go the hue route probably

dmulcahey, 2020-03-26:
https://m.banggood.com/GLEDOPTO-GL-W-001Z-AC100-240V-1-Gang-1-Way-Smart-Light-Switch-Work-With-Zigee-Amazon-Echo-Philip-Hue-p-1524408.html

Hard wired switch

Adminiuga, 2020-03-26:
What does deCONZ do with it? 
I guess we could go HUE approach then.

dmulcahey, 2020-03-26:
I’ll look in a bit

closed 2021-08-13 by github-actions[bot]