zigpy/zha-device-handlers #299: Add support for Legrand Dimmer switch w/o neutral

pdecat, 2020-03-11:
Hi, I'm looking into getting support for the Legrand Dimmer switch w/o neutral ([ref 067771](https://www.legrand.fr/pro/catalogue/42556-version-celiane-with-netatmo/interrupteur-filaire-connecte-avec-option-variateur-celiane-with-netatmo-sans-neutre-5w-a-300w-compensateur-titane), same as [ref 067721](https://www.legrand.fr/pro/catalogue/42556-version-celiane-with-netatmo/interrupteur-filaire-connecte-avec-option-variateur-celiane-with-netatmo-sans-neutre-5w-a-300w-compensateur-blanc) and [ref 064891](https://www.legrand.fr/pro/catalogue/42556-version-celiane-with-netatmo/interrupteur-filaire-connecte-avec-option-variateur-celiane-with-netatmo-sans-neutre-5w-a-300w-compensateur-graphite))

I've managed to pair it, here's its SimpleDescriptor:
```
2020-03-11 08:10:04 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 32772, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0x5bd8>, <Address add
ress_mode=ADDRESS_MODE.NWK address=0x0000>, b'\xe2\x00\xd8[\x1e\x01\x04\x01\x00\x01\x01\x08\x00\x00\x03\x00\x04\x00\x08\x00\x06\x00\x05\x00\x0f\x00\x01\xfc\x03\x00\x00\x01\xfc\x19\x00']
2020-03-11 08:10:04 INFO (MainThread) [zigpy.endpoint] [0x5bd8:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=256 device_version=1 input_clusters=[0, 3, 4, 8, 6, 5, 15, 64513] output_clusters=[0, 64513, 25]>
```

Controlling its on/off state with ZHA is ok and adding it to a group to control it with a Legrand Remote switch
 ([ref 067773](https://www.legrand.fr/pro/catalogue/42559-version-celiane-with-netatmo/commande-sans-fil-celiane-with-netatmo-pour-eclairage-ou-prise-connectee-ou-micromodule-titane), same as [ref 067723](https://www.legrand.fr/pro/catalogue/42559-version-celiane-with-netatmo/commande-sans-fil-celiane-with-netatmo-pour-eclairage-ou-prise-connectee-ou-micromodule-blanc) and [ref 064893](https://www.legrand.fr/pro/catalogue/42559-version-celiane-with-netatmo/commande-sans-fil-celiane-with-netatmo-pour-eclairage-ou-prise-connectee-ou-micromodule-graphite)) works fine too.

This switch has a manufacturer specific cluster `0xfc01` (64513 decimal) that is used to enable/disable features:
- attribute id="0x0000" (type="dat16"): 0100 = Dimmer Off, 0101 = Dimmer On
- attribute id="0x0001" (type="bool"): Enable LED in dark

Would it be possible to add support for this?

Related:
- https://github.com/dresden-elektronik/deconz-rest-plugin/pull/962/files
- https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Legrand-Netatmo-corner.md#cluster-0xfc01

dmulcahey, 2020-03-13:
Absolutely. Remove the device from ZHA wait a minute or so and then add it again through the add devices screen. Take the content of the log panel after on the screen after it joins and add it to a code block in the issue please. I'll try to get to it this weekend if you provide the info.

pdecat, 2020-03-13:
Awesome!

Here you go:

```
Device 0x409e (00:04:74:00:00:8b:0e:a2) joined the network
[0x409e] Requesting 'Node Descriptor'
[0x409e] Extending timeout for 0x4a request
[0x409e:zdo] ZDO request 0x0013: [0x409e, 00:04:74:00:00:8b:0e:a2, 142]
[0x409e] Node Descriptor: <Optional byte1=17 byte2=64 mac_capability_flags=142 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0x409e] Discovering endpoints
[0x409e] Discovered endpoints: [1]
[0x409e:1] Discovering endpoint information
[0x409e:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=256 device_version=1 input_clusters=[0, 3, 4, 8, 6, 5, 15, 64513] output_clusters=[0, 64513, 25]>
[0x409e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=80 command_id=Command.Read_Attributes_rsp>
[0x409e:1] Manufacturer:  Legrand
[0x409e:1] Model:  Dimmer switch w/o neutral
Checking quirks for  Legrand  Dimmer switch w/o neutral (00:04:74:00:00:8b:0e:a2)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x409e:00:04:74:00:00:8b:0e:a2 entering async_device_initialized - is_new_join: True
device - 0x409e:00:04:74:00:00:8b:0e:a2 has joined the ZHA zigbee network
[0x409e]( Dimmer switch w/o neutral): started configuration
[0x409e:ZDO]( Dimmer switch w/o neutral): channel: 'async_configure' stage succeeded
[0x409e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=80 command_id=Command.Read_Attributes_rsp>
[0x409e:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value= Legrand>, <ReadAttributeRecord attrid=5 status=Status.SUCCESS value= Dimmer switch w/o neutral>]]
[0x409e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=80 command_id=Command.Read_Attributes_rsp>
[0x409e:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value= Legrand>, <ReadAttributeRecord attrid=5 status=Status.SUCCESS value= Dimmer switch w/o neutral>]]
[0x409e] Delivery error for seq # 0x52, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
[0x409e:1:0x0008]: Failed to bind 'level' cluster: [0x409e:0:0x0021]: Message send failure
[0x409e] Delivery error for seq # 0x54, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
[0x409e:1:0x0006]: Failed to bind 'on_off' cluster: [0x409e:0:0x0021]: Message send failure
[0x409e:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=88 command_id=Command.Configure_Reporting_rsp>
[0x409e:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=90 command_id=Command.Configure_Reporting_rsp>
[0x409e:1:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>, <ConfigureReportingResponseRecord status=Status.SUCCESS>, <ConfigureReportingResponseRecord status=Status.SUCCESS>, <ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x409e:1:0x0006]: finished channel configuration
[0x409e:1:0x0006]: channel: 'async_configure' stage succeeded
[0x409e:1:0x0000]: Failed to bind 'basic' cluster: 
[0x409e:1:0x0000]: finished channel configuration
[0x409e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=92 command_id=Command.Read_Attributes_rsp>
[... cut ...]
[0x409e] Delivery error for seq # 0x60, on endpoint id 1 cluster 0x0006: ZiGate doesn't answer to command
[0x409e:1:0x0006]: initializing channel: from_cache: False
[0x409e:1:0x0006]: channel: 'async_initialize' stage succeeded
[0x409e] Delivery error for seq # 0x62, on endpoint id 1 cluster 0x0000: ZiGate doesn't answer to command
[0x409e:1:0x0000]: initializing channel: from_cache: False
[0x409e:1:0x0000]: channel: 'async_initialize' stage succeeded
[0x409e]( Dimmer switch w/o neutral): power source: Mains
[0x409e]( Dimmer switch w/o neutral): completed initialization
```

Thanks!

pdecat, 2020-03-13:
As I also have a Legrand Micromodule wired switch (ref 064888) device currently in test, I figured I'll post the same information for it too here:

```
Device 0xfb49 (00:04:74:00:00:0c:2b:bf) joined the network
[0xfb49] Requesting 'Node Descriptor'
[0xfb49] Extending timeout for 0x42 request
[0xfb49:zdo] ZDO request 0x0013: [0xfb49, 00:04:74:00:00:0c:2b:bf, 142]
[0xfb49] Node Descriptor: <Optional byte1=17 byte2=64 mac_capability_flags=142 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0xfb49] Discovering endpoints
[0xfb49] Discovered endpoints: [1, 242]
[0xfb49:1] Discovering endpoint information
[0xfb49:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=256 device_version=1 input_clusters=[0, 3, 4, 6, 5, 15, 64513] output_clusters=[0, 64513, 25]>
[0xfb49:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=72 command_id=Command.Read_Attributes_rsp>
[0xfb49:1] Manufacturer:  Legrand
[0xfb49:1] Model:  Micromodule switch
[0xfb49:242] Discovering endpoint information
[0xfb49:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=72 command_id=Command.Read_Attributes_rsp>
[0xfb49:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value= Legrand>, <ReadAttributeRecord attrid=5 status=Status.SUCCESS value= Micromodule switch>]]
[0xfb49:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=72 command_id=Command.Read_Attributes_rsp>
[0xfb49:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value= Legrand>, <ReadAttributeRecord attrid=5 status=Status.SUCCESS value= Micromodule switch>]]
[0xfb49:242] Discovered endpoint information: <SimpleDescriptor endpoint=242 profile=41440 device_type=97 device_version=0 input_clusters=[] output_clusters=[33]>
Checking quirks for  Legrand  Micromodule switch (00:04:74:00:00:0c:2b:bf)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 242}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 242}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1, 242}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 242}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 242}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 242}
device - 0xfb49:00:04:74:00:00:0c:2b:bf entering async_device_initialized - is_new_join: True
device - 0xfb49:00:04:74:00:00:0c:2b:bf has joined the ZHA zigbee network
[0xfb49]( Micromodule switch): started configuration
[0xfb49:ZDO]( Micromodule switch): channel: 'async_configure' stage succeeded
[0xfb49] Delivery error for seq # 0x4e, on endpoint id 0 cluster 0x0021: Message send failure 166
[0xfb49:1:0x0000]: Failed to bind 'basic' cluster: [0xfb49:0:0x0021]: Message send failure
[0xfb49:1:0x0000]: finished channel configuration
[0xfb49] Delivery error for seq # 0x50, on endpoint id 1 cluster 0x0000: Message send failure 166
[0xfb49:1:0x0000]: initializing channel: from_cache: False
[0xfb49:1:0x0000]: channel: 'async_configure' stage succeeded
[0xfb49] Delivery error for seq # 0x4c, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
[0xfb49:1:0x0006]: Failed to bind 'on_off' cluster: [0xfb49:0:0x0021]: Message send failure
[0xfb49] Delivery error for seq # 0x5e, on endpoint id 1 cluster 0x0006: Message send failure 166
[0xfb49:1:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: [0xfb49:1:0x0006]: Message send failure
[0xfb49:1:0x0006]: finished channel configuration
[0xfb49:1:0x0006]: channel: 'async_configure' stage succeeded
[0xfb49]( Micromodule switch): completed configuration
[0xfb49]( Micromodule switch): stored in registry: ZhaDeviceEntry(name=' Legrand  Micromodule switch', ieee='00:04:74:00:00:0c:2b:bf', last_seen=1584034541.6285045)
[0xfb49]( Micromodule switch): started initialization
[... cut ...]
[0xfb49] Delivery error for seq # 0x60, on endpoint id 1 cluster 0x0006: ZiGate doesn't answer to command
[0xfb49:1:0x0006]: initializing channel: from_cache: False
[0xfb49:1:0x0006]: channel: 'async_initialize' stage succeeded
[0xfb49] Delivery error for seq # 0x62, on endpoint id 1 cluster 0x0000: ZiGate doesn't answer to command
[0xfb49:1:0x0000]: initializing channel: from_cache: False
[0xfb49:1:0x0000]: channel: 'async_initialize' stage succeeded
[0xfb49]( Micromodule switch): power source: Mains
[0xfb49]( Micromodule switch): completed initialization
```

pdecat, 2020-03-13:
And here's the information for the Legrand Remote switch (ref 067773, same as ref 067723 and ref 064893):

```
Device 0x2f9d (00:04:74:00:00:a9:0c:75) joined the network
[0x2f9d] Requesting 'Node Descriptor'
[0x2f9d] Extending timeout for 0xde request
[0x2f9d:zdo] ZDO request 0x0013: [0x2f9d, 00:04:74:00:00:a9:0c:75, 128]
Ignoring message (b'142110010000f0') on cluster 0: unknown endpoint or cluster id: 1
Get bindable devices: source_ieee: [00:04:74:00:00:0c:2b:bf], bindable devices: [[]]
[0x2f9d] Extending timeout for 0xec request
[0x2f9d] Node Descriptor: <Optional byte1=18 byte2=64 mac_capability_flags=128 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0x2f9d] Discovering endpoints
[0x2f9d] Extending timeout for 0xee request
[0x2f9d] Discovered endpoints: [1]
[0x2f9d:1] Discovering endpoint information
[0x2f9d] Extending timeout for 0xf0 request
[0x2f9d:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=260 device_version=1 input_clusters=[0, 3, 15, 32, 1, 64513] output_clusters=[3, 6, 8, 0, 64513, 25]>
[0x2f9d] Extending timeout for 0xf2 request
[0x2f9d:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=242 command_id=Command.Read_Attributes_rsp>
[0x2f9d:1] Manufacturer:  Legrand
[0x2f9d:1] Model:  Remote switch
Checking quirks for  Legrand  Remote switch (00:04:74:00:00:a9:0c:75)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x2f9d:00:04:74:00:00:a9:0c:75 entering async_device_initialized - is_new_join: True
device - 0x2f9d:00:04:74:00:00:a9:0c:75 has joined the ZHA zigbee network
[0x2f9d]( Remote switch): started configuration
[0x2f9d:ZDO]( Remote switch): channel: 'async_configure' stage succeeded
[0x2f9d:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=242 command_id=Command.Read_Attributes_rsp>
[0x2f9d:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value= Legrand>, <ReadAttributeRecord attrid=5 status=Status.SUCCESS value= Remote switch>]]
[0x2f9d:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=242 command_id=Command.Read_Attributes_rsp>
[0x2f9d:1:0x0000] ZCL request 0x0001: [[<ReadAttributeRecord attrid=4 status=Status.SUCCESS value= Legrand>, <ReadAttributeRecord attrid=5 status=Status.SUCCESS value= Remote switch>]]
[0x2f9d] Extending timeout for 0xf4 request
[0x2f9d] Extending timeout for 0xf6 request
[0x2f9d] Extending timeout for 0xf8 request
[0x2f9d] Delivery error for seq # 0xf4, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
[0x2f9d:1:0x0000]: Failed to bind 'basic' cluster: [0x2f9d:0:0x0021]: Message send failure
[0x2f9d:1:0x0000]: finished channel configuration
[0x2f9d] Extending timeout for 0xfa request
[0x2f9d] Delivery error for seq # 0xf6, on endpoint id 0 cluster 0x0021: ZiGate doesn't answer to command
[0x2f9d:1:0x0001]: Failed to bind 'power' cluster: [0x2f9d:0:0x0021]: Message send failure
[0x2f9d] Extending timeout for 0xfc request
[0x2f9d] Delivery error for seq # 0xfa, on endpoint id 1 cluster 0x0000: ZiGate doesn't answer to command
[0x2f9d:1:0x0000]: initializing channel: from_cache: False
[0x2f9d:1:0x0000]: channel: 'async_configure' stage succeeded
[0x2f9d] Extending timeout for 0xfe request
```

pdecat, 2020-04-01:
Hi @dmulcahey, did I miss something in the provided information?

dmulcahey, 2020-04-01:
No, just time on my end

dmulcahey, 2020-04-08:
@pdecat what are you expecting for the other 2 devices?

pdecat, 2020-04-08:
Well, not much actually, I thought maybe it would help knowing those devices features.

Thanks for the PR, I'll give an eye to learn how to implement such changes in the future and help contribute.

closed 2020-04-20 by dmulcahey

t0mg, 2020-05-28:
Hi @pdecat apologies for  bothering you here, but I'm super new to zigbee and am having a hard time getting the dimmer to work. I just got my host zigate today and tried pairing it with a brand new Legrand dimmer switch (that hasn't been paired with the Legrand base). It does pair, and the quirk does get picked up correctly. I can also turn the switch on and off. However in ZHA interface for the entity, when I select LegrandCluster, I can't seem to get or set the attribute for dimmer mode (or led level). set_zigbee_cluster_attribute service call fails. Is there anything obvious I may be missing? How am I supposed to set the value from the UI?
Thanks!

pdecat, 2020-05-30:
Hi @t0mg, apparently some Legrand dimmer switches need their firmware to be updated from stack version 64 to 66 with the Legrand Hub to have all features available, see:
- https://zigbee.blakadder.com/Legrand_067771.html
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-538806070

That may be your issue.

You can query the `stack_version` attribute (id `0x0002`) on the `Basic` cluster (endpoint id: `1`, id: `0x0000`, type: `in`)

t0mg, 2020-05-30:
Thanks for the detailed help @pdecat !

Ok so I have stack version 66 and sw version 26. I was hoping that I could avoid purchasing the Legrand hub which I don't intend to use, but was also starting to think the ota would be the problem. However according to your reply it seems the switch should already support the dimmer feature 🤔

pdecat, 2020-05-31:
What logs do you get when `set_zigbee_cluster_attribute` fails?

t0mg, 2020-06-02:
Hello @pdecat - I've updated the zigate to 3.1c but then I found another problem. My zigate is a PiZiGate (connects directly to the GPIO) and it seems that ZHA doesn't fully support that hardware: the PiZiGate needs to be set to production mode. When done manually it looks like this:
```
$ gpio mode 0 out
$ gpio mode 2 out
$ gpio write 2 1 
$ gpio write 0 0
$ gpio write 0 1
```
If not done, the ZiGate doesn't operate properly. I never noticed because I initially tried the [homeassistant-zigate](https://github.com/doudz/homeassistant-zigate) custom integration before switching to ZHA, and I didn't power down the raspberry when I switched (only a couple of reboots) so the PiZigate remained in production mode after it was set by the homeassistant-zigate plugin. But when I returned after flashing the ZiGate from a different raspberry pi with a full raspbian, HassOS booted up with ZHA directly, and the ZiGate was no longer recognized. I had to switch back to the custom integration for the GPIO to be properly set. 

From ZHA I was _still_ not able to use the LegrandCluster. However going back to the custom integration again I was able to send a raw command, and successfully enable the dimmer mode. I was also able to set the brightness through that custom integration's entity. Alas, back again in ZHA I couldn't set the brightness of the dimmer that well. It behaved as if I was sending +1/-1 increments instead of setting a value out of 255.

(Will update with logs here in a minute)

I would rather stick with ZHA integration going forward if possible, but that would imply that I figure out why the dimmer doesn't work as expected, and also that I never power down my rpi - because as explained if I do it looks like I have to reinstall the custom integration for production mode be properly set on the board.

Crocmagnon, 2020-08-30:
I have one of these devices too. I can discover it with Home Assistant but I can’t control it afterwards. Should I open another issue?

Cyrille63, 2021-04-11:
> I have one of these devices too. I can discover it with Home Assistant but I can’t control it afterwards. Should I open another issue?

Hi, me too I have the same problem, I can not control it, is there a procedure ?

PR #330 (2020-04-20): Add Legrand dimmer without neutral
Fixes: #299