zigpy/zha-device-handlers #300: Konke Multi Function Button - Battery unknown - No zha_events generated

walthowd, 2020-03-13:
This is the Konke Multi function button: https://zigbee.blakadder.com/Konke_KPKEY.html

[image]

[image]
[image]

Here is the zigpy debug log when joining the device:

https://paste.ubuntu.com/p/PFvQZyVHdd/

It appears the button presses are sent as with hex 80 (quick press), 81 (long press), 82 (double click). 

Those generate the following error:

```
[0xbcb8:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=122 command_id=Command.Report_Attributes>
Failed to parse message (b'087a0a00001080') on cluster 6, because 128 is not a valid Bool
```

Packet captures from:
http://faire-ca-soi-meme.fr/domotique/2019/07/16/test-du-kit-zigbee-konke-smart-home/

Single click:
[image]

Long click:
[image]

Double click:
[image]

walthowd, 2020-03-13:
Battery is a CR2450 3V

Adminiuga, 2020-03-13:
If it sends standard attributes with wrong data type, then it would need a quirk like xiaomi device with custom deserialization handler

walthowd, 2020-03-13:
This thread has a lot of relevant info, but I couldn't quite decipher it all:

https://github.com/Koenkk/zigbee2mqtt/issues/1689

walthowd, 2020-12-07:
Support for Konke button has been merged into dev. Thanks @tube0013 !

closed 2020-12-07 by walthowd