zigpy/zha-device-handlers #301: Xiaomi Aquara T1 Lux Sensor - Battery values

walthowd, 2020-03-13:
This is the join log for the Xiaomi Aqara T1 Lux sensor. Lux values appear to work, no battery information reported. 

[image]

Join log:
https://paste.ubuntu.com/p/23GHfsXYFt/

walthowd, 2020-03-13:
Battery is a CR 2450 3V

dmulcahey, 2020-03-13:
You on the beta?

walthowd, 2020-03-13:
No, but can upgrade.

dmulcahey, 2020-03-13:
Pretty sure this works on beta

dmulcahey, 2020-03-13:
https://github.com/dmulcahey/zha-device-handlers/pull/277/files

walthowd, 2020-03-13:
I will upgrade and try. Sorry, I forgot to look for recent commits on this one.

dmulcahey, 2020-03-13:
No reason to be sorry!!!

walthowd, 2020-03-13:
Works perfectly in beta after re-pairing! Thank you!

closed 2020-03-13 by walthowd