zigpy/zha-device-handlers #305: [Device Support Request] Smartthing Motion Sensor v5

meauxt, 2020-03-18:
**Is your feature request related to a problem? Please describe.**
support for battery % for smartthings motion sensor
the % has not been working before but iwas seen the voltage in additional attribute.. after homeassistant 0.107 i stop seeing even the voltage

**Describe the solution you'd like**
I would love if you can add the support to show battery percentage 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again `[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x0000](EZSP): started initialization
[0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[0x0000](EZSP): power source: Mains
[0x0000](EZSP): completed initialization
Device 0x7d60 (24:fd:5b:00:01:0b:f7:ae) joined the network
Device 24:fd:5b:00:01:0b:f7:ae changed id (0xec35 => 0x7d60)
[0x7d60:zdo] ZDO request 0x0013: [0x7d60, 24:fd:5b:00:01:0b:f7:ae, 128]
Checking quirks for SmartThings motionv5 (24:fd:5b:00:01:0b:f7:ae)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because input cluster mismatch on at least one endpoint
device - 32096:24:fd:5b:00:01:0b:f7:ae entering async_device_initialized - is_new_join: False
device - 32096:24:fd:5b:00:01:0b:f7:ae has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 32096:24:fd:5b:00:01:0b:f7:ae
[32096](motionv5): started configuration
[32096:ZDO](motionv5): 'async_configure' stage succeeded
[0x7d60] Extending timeout for 0x6c request
[0x7d60] Extending timeout for 0x6e request
[0x7d60] Extending timeout for 0x70 request
[32096](motionv5): started initialization
[32096:ZDO](motionv5): 'async_initialize' stage succeeded
[0x7d60:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=110 command_id=Command.Write_Attributes_rsp>
[32096:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x7d60] Extending timeout for 0x72 request
[32096:1:0x0020]: 3300.0s check-in interval set: [[<WriteAttributesStatusRecord status=Status.SUCCESS>]]
[0x7d60] Extending timeout for 0x74 request
[32096:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[32096:1:0x0000]: finished channel configuration
[0x7d60] Extending timeout for 0x76 request
[0x7d60:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=114 command_id=Command.Configure_Reporting_rsp>
[32096:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[32096:1:0x0402]: finished channel configuration
[0x7d60] Extending timeout for 0x78 request
[0x7d60:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=118 command_id=Command.Read_Attributes_rsp>
[32096:1:0x0020]: bound 'poll_control' cluster: Status.SUCCESS
[32096:1:0x0020]: finished channel configuration
[0x7d60] Extending timeout for 0x7a request
[32096:1:0x0000]: initializing channel: from_cache: False
[32096:1:0x0402]: initializing channel: from_cache: False
[0x7d60] Extending timeout for 0x7c request
[0x7d60:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=120 command_id=Command.Read_Attributes_rsp>
[... 37 lines cut ...]
[0x7d60:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=138 command_id=Command.Read_Attributes_rsp>
[0x7d60:1:0x0001] ZCL request 0x0001: [[<ReadAttributeRecord attrid=49 status=Status.UNSUPPORTED_ATTRIBUTE>, <ReadAttributeRecord attrid=33 status=Status.UNSUPPORTED_ATTRIBUTE>, <ReadAttributeRecord attrid=32 status=Status.SUCCESS value=44>, <ReadAttributeRecord attrid=51 status=Status.UNSUPPORTED_ATTRIBUTE>]]
[0x7d60:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=140 command_id=Command.Default_Response>
[32096:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0x7d60:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=142 command_id=Command.Read_Attributes_rsp>
[32096:1:0x0402]: 'async_initialize' stage succeeded
[32096:1:0x0020]: 'async_initialize' stage succeeded
[32096:1:0x0000]: 'async_initialize' stage succeeded
[32096:1:0x0500]: 'async_initialize' stage succeeded
[32096:1:0x0001]: 'async_initialize' stage succeeded
[32096](motionv5): power source: Battery or Unknown
[32096](motionv5): completed initialization
[0x7d60:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=6 command_id=1>
[0x7d60:1:0x0500] ZCL request 0x0001: [<ZoneType.Standard_CIE: 0>, 4362]
[0x7d60:1:0x0500] No handler for cluster command 1
[32096:1:0x0500]: Enroll requested
[0x7d60] Extending timeout for 0x90 request
[0x7d60:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=7 command_id=0>
[0x7d60:1:0x0500] ZCL request 0x0000: [32, 0, 0, 0]
[0x7d60:1:0x0500] No handler for cluster command 0
[32096:1:0x0500]: Updated alarm state: 0
`

**Additional context**
Add any other context or screenshots about the feature request here.

dmulcahey, 2020-03-18:
```
source: Mains [0x0000](EZSP): completed initialization Device 0x7d60 (24:fd:5b:00:01:0b:f7:ae) joined the network Device 24:fd:5b:00:01:0b:f7:ae changed id (0xec35 => 0x7d60) [0x7d60:zdo] ZDO request 0x0013: [0x7d60, 24:fd:5b:00:01:0b:f7:ae, 128] Checking quirks for SmartThings motionv5 
```

The device wasn’t actually removed from HA. You need to unpair the device and re pair it. Wait about 60 seconds in between.

meauxt, 2020-03-18:
sorry about that @dmulcahey .. i remove it and repair it :) please let me know if this looks good 
`[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0x1edf (24:fd:5b:00:01:0b:f7:ae) joined the network
[0x1edf:zdo] ZDO request 0x0013: [0x1edf, 24:fd:5b:00:01:0b:f7:ae, 128]
[0x1edf] Requesting 'Node Descriptor'
[0x1edf] Extending timeout for 0x96 request
[0x1edf] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4362 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x1edf] Discovering endpoints
[0x1edf] Extending timeout for 0x98 request
[0xb894:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=37 command_id=0>
[0xb894:1:0x0500] ZCL request 0x0000: [33, 0, 0, 0]
[0xb894:1:0x0500] No handler for cluster command 0
[0xb894:1:0x0500]: Updated alarm state: 1
[0x1edf] Discovered endpoints: [1]
[0x1edf:1] Discovering endpoint information
[0x1edf] Extending timeout for 0x9a request
[0x1edf:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 15, 32, 1026, 1280] output_clusters=[25]>
[0x1edf] Extending timeout for 0x9c request
[0x1edf:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=156 command_id=Command.Read_Attributes_rsp>
[0x1edf:1] Manufacturer: SmartThings
[0x1edf:1] Model: motionv5
Checking quirks for SmartThings motionv5 (24:fd:5b:00:01:0b:f7:ae)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because input cluster mismatch on at least one endpoint
device - 0x1edf:24:fd:5b:00:01:0b:f7:ae entering async_device_initialized - is_new_join: True
device - 0x1edf:24:fd:5b:00:01:0b:f7:ae has joined the ZHA zigbee network
[0x1edf](motionv5): started configuration
[0x1edf:ZDO](motionv5): 'async_configure' stage succeeded
[0x1edf] Extending timeout for 0x9e request
[0x1edf] Extending timeout for 0xa0 request
[0x1edf] Extending timeout for 0xa2 request
[0x1edf:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x1edf] Extending timeout for 0xa4 request
[0x1edf:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=160 command_id=Command.Write_Attributes_rsp>
[0x1edf:1:0x0020]: 3300.0s check-in interval set: [[<WriteAttributesStatusRecord status=Status.SUCCESS>]]
[0x1edf] Extending timeout for 0xa6 request
[0x1edf:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x1edf:1:0x0000]: finished channel configuration
[0x1edf] Extending timeout for 0xa8 request
[0x1edf:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=164 command_id=Command.Configure_Reporting_rsp>
[0x1edf:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x1edf:1:0x0402]: finished channel configuration
[0x1edf] Extending timeout for 0xaa request
[0x1edf:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=168 command_id=Command.Read_Attributes_rsp>
[0x1edf:1:0x0020]: bound 'poll_control' cluster: Status.SUCCESS
[0x1edf:1:0x0020]: finished channel configuration
[0x1edf] Extending timeout for 0xac request
[0x1edf:1:0x0000]: initializing channel: from_cache: False
[... 58 lines cut ...]
[0x429b:1:0x0500]: Updated alarm state: 1
[0x1edf:zdo] ZDO request 0x0000: [00:0d:6f:00:0b:5e:25:56, 0, 0]
[... cut ...]
[0xe08b:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2200>>]]
[0xe08b:1:0x0402] Attribute report received: measured_value=2200
[0x429b:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=5 command_id=0>
[0x429b:1:0x0500] ZCL request 0x0000: [32, 0, 0, 0]
[0x429b:1:0x0500] No handler for cluster command 0
[0x429b:1:0x0500]: Updated alarm state: 0
`

Adminiuga, 2020-03-19:
```
[0x1edf] Discovered endpoints: [1]
[0x1edf:1] Discovering endpoint information [0x1edf]
[0x1edf:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 15, 32, 1026, 1280] output_clusters=[25]>
[0x1edf:1] Manufacturer: SmartThings
[0x1edf:1] Model: motionv5
```

and `battery_percent_remaining` is an unsupported attribute.

meauxt, 2020-03-19:
weird.. then how smartthing show the battery level ( notice that they have it in10% increment btw)

Adminiuga, 2020-03-19:
it's same way zha does, base on `battery_voltage` data. Yep!

meauxt, 2020-03-19:
awesome because i saw also this so battery_voltage seems supported: 
[0x1edf:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'

and this what you are referring to right?
[0x1edf:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.UNSUPPORTED_ATTRIBUTE direction=0 attrid=33>]]'

do you think it will be easy change/addition?

Adminiuga, 2020-03-19:
yes. What kind of Battery?

meauxt, 2020-03-19:
'CR2477 3V'

meauxt, 2020-03-19:
I have 10 sensors they keep dying and I have no way to tell after moving from smart things. i add lovelace card with unavailable entities for meanwhile :( so Thank you in advance !

meauxt, 2020-03-23:
do you think it's possible for this to be in HA 0.108 ? or that's tough to have it for that release ? @Adminiuga

Adminiuga, 2020-03-26:
PR is submitted, so chances are high.

PR #312 (2020-03-26): SmartThings motion quirk update.
Update signature to include `motionv5` model.
Fixes #305

Upon closer examination, signature for `motionv5` sensor exactly matches signature of `motionv4` sensor, so intstead I've added new model to `motionv4` signature and renamed the quirk. Let me know if this works for you.
closed 2020-03-26 by dmulcahey