zigpy/zha-device-handlers #306: [BUG]Ikea FYRTUR blind and remote reporting wrong battery or no battery

blair287, 2020-03-21:
First 

Blind:

Never updates the battery automatically.

However if you do a zigbee get cluster attribute for battery percentage remaining it does return a correct value.

Remote:

does update battery on its own.

Both:

both values are displayed correctly in the get attribute box but the sensor for the batteries figures are always half what it shows.
[image]
[image]

dmulcahey, 2020-03-28:
please remove the device via the remove device button in the screenshot above. activate the device after clicking the button to ensure the device is awake. Then after about a minute please re-pair the device via the add devices screen and post the entire contents of that log area on that screen in a code block here.

jmoransalama, 2020-04-11:
Hi! I'm having the exact same issue described by @blair287 

This is the output of the log area:

```
[0xfc13] Delivery error for seq # 0x64, on endpoint id 0 cluster 0x0036: message send failure
Couldn't open '08:6b:d7:ff:fe:58:64:f9' for joining: [0xfc13:0:0x0036]: Message send failure
Device 0x8499 (d0:cf:5e:ff:fe:ea:35:27) joined the network
[0x8499:zdo] ZDO request 0x0013: [0x8499, d0:cf:5e:ff:fe:ea:35:27, 128]
[0x8499] Requesting 'Node Descriptor'
[0x8499] Extending timeout for 0x66 request
[0x8499] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x8499] Discovering endpoints
[0x8499] Extending timeout for 0x68 request
[0x8499] Discovered endpoints: [1]
[0x8499:1] Discovering endpoint information
[0x8499] Extending timeout for 0x6a request
[0x8499:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=514 device_version=1 input_clusters=[0, 1, 3, 4, 5, 32, 258, 4096, 64636] output_clusters=[25, 4096]>
[0x8499] Extending timeout for 0x6c request
[0x8a95:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=96 command_id=Command.Report_Attributes>
[0x8a95:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2174>>]]
[0x8a95:1:0x0402] Attribute report received: measured_value=2174
[0x8a95:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=97 command_id=Command.Report_Attributes>
[0x8a95:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=5067>>]]
[0x8a95:1:0x0405] Attribute report received: measured_value=5067
[0x8a95:1:0x0403] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=98 command_id=Command.Report_Attributes>
[0x8a95:1:0x0403] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1013>>, <Attribute attrid=20 value=<TypeValue type=int8s, value=-1>>, <Attribute attrid=16 value=<TypeValue type=int16s, value=10132>>]]
[0x8a95:1:0x0403] Attribute report received: measured_value=1013, 20=-1, 16=10132
[0x8499:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=108 command_id=Command.Read_Attributes_rsp>
[0x8499:1] Manufacturer: IKEA of Sweden
[0x8499:1] Model: FYRTUR block-out roller blind
Checking quirks for IKEA of Sweden FYRTUR block-out roller blind (d0:cf:5e:ff:fe:ea:35:27)
Considering <class 'zhaquirks.ikea.tradfriplug.TradfriPlug'>
Fail because endpoint list mismatch: {1, 2, 242} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x8499:d0:cf:5e:ff:fe:ea:35:27 entering async_device_initialized - is_new_join: True
device - 0x8499:d0:cf:5e:ff:fe:ea:35:27 has joined the ZHA zigbee network
[0x8499](FYRTUR block-out roller blind): started configuration
[0x8499:ZDO](FYRTUR block-out roller blind): 'async_configure' stage succeeded
[0x8499] Extending timeout for 0x6e request
[0x8499] Extending timeout for 0x70 request
[0x8499] Extending timeout for 0x72 request
[0x8499:1:0x0102]: bound 'window_covering' cluster: Status.SUCCESS
[0x8499] Extending timeout for 0x74 request
[0x8499:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x8499:1:0x0000]: finished channel configuration
[0x8499] Extending timeout for 0x76 request
[... 51 lines cut ...]
[0x8499:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 0
[0x8499:1:0x0102]: read current position: 0
polling current state
read pos=0
[... cut ...]
[0x9314:1:0x0400] Attribute report received: measured_value=45
[0x9314:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=123 command_id=Command.Report_Attributes>
[0x9314:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=1>>]]
[0x9314:1:0x0406] Attribute report received: occupancy=1
[0x9314:1:0x0500]: Updated alarm state: 1
[0x9f51:1:0x0500]: Updated alarm state: 0

```

Adminiuga, 2020-04-12:
Read `battery_voltage` attribute of the `power configuration` cluster from ZHA device card in ZHA configuration panel.
And at what battery level blinds stop responding?

jmoransalama, 2020-04-12:
Sorry I'm not sure exactly what info you need and English is not my main language, so I'll give you all the possible info. Sorry if some of it was not needed.

In my case the blinds haven't stopped responding, but this is what I have to do in order to get the status of the blind displayed correctly on home-assistant:
1. Call service _cover.set_cover_position_ with position attribute, in this example I set to the value to 80. The blind will correctly move to the desired position.

2. Now if I go to the status of the cover, the value doesn't get updated, it will show the last value that I updated manually.
[image]

3. For the status to update on the frontend, I go to WindowCovering cluster and click on "Get ZIGBEE ATTRIBUTE" for attribute _current_position_lift_percentage_. Now I have the correct value for the blind displayed on the frontend (Although the status shows opening when it should say open):
[image]
[image]

The same thing happens with the battery sensor, I have to get the battery_percentage_remaining from the zha page in order for the sensor to be updated. And the sensor always shows half of the given value.
[image]
[image]

If I get the _battery_voltage_ attribute, this is what I get:
[image]

Thanks for the help, let me know if I can give you more information to help fix this.

Adminiuga, 2020-04-12:
ah, it does support `battery_percent_remaining` attribute, so it shouldn't need a quirk. Per log it configured the attribute reporting successfuly
```
[0x8499:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]
```

How long have you waited for the battery sensor to update? 

Enable debug logging for zigpy and zha, in `configuration.yaml`

logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_zigate: debug
```

run it for 24 hours, then do `grep 0x8499:1:0x0001 home-assistant.log` to see if you get attribute updates reported. If battery level doesn't change, then blind should report it once every 24 hours IIRC

jmoransalama, 2020-04-14:
This is the output of the command _grep 0x8499:1:0x0001 home-assistant.log_ after two full days of running home-assistant with debug logging enabled:
```
2020-04-13 00:26:27 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8499:1:0x0001]: initializing channel: from_cache: True
2020-04-13 00:26:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8499:1:0x0001]: 'async_initialize' stage succeeded
2020-04-13 00:27:03 DEBUG (MainThread) [zigpy.zcl] [0x8499:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=41 command_id=Command.Read_Attributes_rsp>
2020-04-13 00:27:03 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8499:1:0x0001]: initializing channel: from_cache: False
2020-04-13 00:27:04 DEBUG (MainThread) [zigpy.zcl] [0x8499:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=47 command_id=Command.Read_Attributes_rsp>
2020-04-13 00:27:04 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8499:1:0x0001]: 'async_initialize' stage succeeded
```

blair287, 2020-05-10:
It's never once updated battery on its own without the attribute being read manually. 

Now after 109.4 update I'm not getting bettery from the remote either.

blair287, 2020-05-29:
The half battery life still isnt fixed any eta surely this is simple to fix just x2 the output.

dmulcahey, 2020-05-29:
There is already a custom cluster that does this. Just needs to be used

blair287, 2020-05-29:
> There is already a custom cluster that does this. Just needs to be used

How can that be done please? How come its not used by default?

blair287, 2020-05-30:
?

dmulcahey, 2020-05-30:
Because IKEA doesn’t follow the ZCL spec. We’ll have to add a quirk for it. Be patient this is done as time permits as a hobby.

Adminiuga, 2020-05-30:
In other words: PRs are welcome.

blair287, 2020-06-02:
Im sorry i thought you meant there was a setting i could change.

PR #528 (2020-10-29): Ikea blind and remote signature fixes
Should fix #306 battery reporting half value.
closed 2020-10-29 by Adminiuga