zigpy/zha-device-handlers #307: [Device Support Request] Philips Hue Outdoor Sensor SML002

SneakSnackSnake, 2020-03-22:
**Is your feature request related to a problem? Please describe.**
Support for motion detection and battery.

**Describe the solution you'd like**
Getting the motion detection to work first, after that we can look into battery, temperature, illuminance etc.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
2020-03-22 12:32:45 INFO (MainThread) [zigpy.application] Device 0xc2ce (00:17:88:01:06:47:39:b7) joined the network
2020-03-22 12:32:45 INFO (MainThread) [zigpy.device] [0xc2ce] Requesting 'Node Descriptor'
2020-03-22 12:32:46 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
2020-03-22 12:32:46 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8035
2020-03-22 12:33:15 INFO (MainThread) [zigpy.device] [0xc2ce] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4107 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=0 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
2020-03-22 12:33:15 INFO (MainThread) [zigpy.device] [0xc2ce] Discovering endpoints
2020-03-22 12:33:15 INFO (MainThread) [zigpy.device] [0xc2ce] Discovered endpoints: [2, 1]
2020-03-22 12:33:15 INFO (MainThread) [zigpy.endpoint] [0xc2ce:2] Discovering endpoint information
2020-03-22 12:33:16 INFO (MainThread) [zigpy.endpoint] [0xc2ce:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=260 device_type=263 device_version=0 input_clusters=[0, 1, 3, 1030, 1024, 1026] output_clusters=[25]>
2020-03-22 12:33:16 INFO (MainThread) [zigpy.endpoint] [0xc2ce:1] Discovering endpoint information
2020-03-22 12:33:17 INFO (MainThread) [zigpy.endpoint] [0xc2ce:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=2128 device_version=2 input_clusters=[0] output_clusters=[0, 3, 4, 6, 8, 768, 5]>
2020-03-22 12:33:17 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xc2ce:00:17:88:01:06:47:39:b7 entering async_device_initialized - is_new_join: True
2020-03-22 12:33:17 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xc2ce:00:17:88:01:06:47:39:b7 has joined the ZHA zigbee network
2020-03-22 12:33:17 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xc2ce](SML002): started configuration
2020-03-22 12:33:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:ZDO](SML002): 'async_configure' stage succeeded
2020-03-22 12:33:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0008]: bound 'level' cluster: Status.SUCCESS
2020-03-22 12:33:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0008]: finished channel configuration
2020-03-22 12:33:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0006]: received 'on_with_timed_off' command with [0, 3000, 0] args on cluster_id '6' tsn '0'
2020-03-22 12:33:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0006]: received 'on_with_timed_off' command with [0, 3000, 0] args on cluster_id '6' tsn '0'
2020-03-22 12:33:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0300]: bound 'light_color' cluster: Status.SUCCESS
2020-03-22 12:33:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0300]: finished channel configuration
2020-03-22 12:33:18 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8011
2020-03-22 12:33:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0005]: bound 'scenes' cluster: Status.SUCCESS
2020-03-22 12:33:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0005]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0006]: Failed to bind 'on_off' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0006]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0402]: Failed to bind 'temperature' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0402]: failed to set reporting for 'measured_value' attr on 'temperature' cluster: [0xc2ce:2:0x0402]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0402]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0400]: Failed to bind 'illuminance' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0400]: failed to set reporting for 'measured_value' attr on 'illuminance' cluster: [0xc2ce:2:0x0400]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0400]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0000]: Failed to bind 'basic' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0000]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0000]: initializing channel: from_cache: False
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0001]: Failed to bind 'power' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0001]: failed to set reporting for 'battery_voltage' attr on 'power' cluster: [0xc2ce:2:0x0001]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0001]: failed to set reporting for 'battery_percentage_remaining' attr on 'power' cluster: [0xc2ce:2:0x0001]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0001]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0406]: Failed to bind 'occupancy' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0406]: failed to set reporting for 'occupancy' attr on 'occupancy' cluster: [0xc2ce:2:0x0406]: Message send failure
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0406]: finished channel configuration
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0402]: 'async_configure' stage succeeded
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0400]: 'async_configure' stage succeeded
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0000]: 'async_configure' stage succeeded
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0001]: 'async_configure' stage succeeded
2020-03-22 12:33:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:2:0x0406]: 'async_configure' stage succeeded
2020-03-22 12:33:20 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
2020-03-22 12:33:20 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0000]: Failed to bind 'basic' cluster: [0xc2ce:0:0x0021]: Message send failure
2020-03-22 12:33:20 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0000]: finished channel configuration
2020-03-22 12:33:20 WARNING (MainThread) [zigpy_zigate.api] No response to command 0x0530
[... cut ...]
2020-03-22 12:33:23 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0005]: 'async_initialize' stage succeeded
2020-03-22 12:33:23 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xc2ce:1:0x0006]: 'async_initialize' stage succeeded
2020-03-22 12:33:23 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xc2ce](SML002): power source: Battery or Unknown
2020-03-22 12:33:23 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xc2ce](SML002): completed initialization

**Additional context**

Adminiuga, 2020-03-22:
Either device went into deep sleep or left the network. When pairing the device, try keeping it awake by pressit buttons on it.

SneakSnackSnake, 2020-03-22:
I removed the device and added it again while pushing the setup button multiple times, but I still get the same Failed to bind errors in the logging.

SneakSnackSnake, 2020-03-25:
@Adminiuga 
You were right, this time I pushed the setup button multiple times really fast. That seems to do the trick. Everything seems to be working now. Thank you!
By the way, do you know what on_off is for?

dmulcahey, 2020-03-26:
@SneakSnackSnake if this works as expected can this be closed?

closed 2020-03-30 by SneakSnackSnake

voskuh123, 2020-06-25:
I have a question about this motion sensor. There is also an sensitivity setting which can be used to set an integer from 0 to 4 to change the sensitivity of motion detection. This setting is on cluster 0x0406 attribute 0x0030 but it isn't shown on the cluster and attribute page of the device. Is there a way to change this setting?
See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Philips-Hue-sniffing#hue-motion-sensor for more details about the attribute.

Adminiuga, 2020-06-25:
This would need a quirk. Open a new issue and post this link, device model and zigbee information for the motion sensor in question