zigpy/zha-device-handlers #308: [BUG] XBee quirk  - alot of warnings "FOREIGN KEY constraint failed" in log

MaximumSU, 2020-03-22:
**Describe the bug**
I setup new zha network with some XBee modules. I adding, and configuring 8 modules, disable unneeded switches and sensors in device configs. All looking good except of one node. That node send the IO sampling of enabled pins every 1 minute. And every minute i see in log 6 messages
"FOREIGN KEY constraint failed":
```
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00AUJ\x1a\x03F\xe8\xe8\x00\x92\xc1\x05\x01\x01\x0c\xb8\x00\x00('
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:13:a2:00:41:55:4a:1a, 0x0346, 232, 146, 1, b'010cb8000028')
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy.zcl] [0x0346:232:0x0092] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=0>
2020-03-21 18:05:50 DEBUG (MainThread) [zigpy.zcl] [0x0346:232:0x0092] ZCL request 0x0000: [{'digital_pins': [0, 0, 0, 1, 1, 1, 0, 1, 0, 0, 1, 1, 0], 'analog_pins': [0, 0, 0, 0, 0, 0, 0, 0], 'digital_samples': [0, 0, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0], 'analog_samples': [0, 0, 0, 0, 0, 0, 0, 0]}]
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-21 18:05:50 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
```
I even can't see, what is the query was (what is the content of that "FOREIGN KEY"), because there is no such debug logger in HA (or not finded them). We started to discuss it [here](https://github.com/zigpy/zigpy/issues/326#issue-585535842) but have stull.

prairiesnpr, 2020-03-22:
How did you disable the switches and sensors?  You have six active digital pins and six attributes failed to update, was the error present before you disabled them?

MaximumSU, 2020-03-22:
This is not depend on number of enabled switches or sensors in HA. I have 5 active switches, but 6 warnings.. The number of "FOREIGN KEY constraint failed" depend on number of enabled (sampled) pins on XBee. When i not had unneded pins disabled on XBee, the number of "FOREIGN KEY constraint failed" was 13. Enabling or disabling switches and sensors in HA not affect (remark - if you try to reenable disabled switch, ZHA get broken - need to restart HA to get it back). The only one place it may be connected with, it is the table "endpoints", from there that FOREIGN key used by attributes table. So, i think, it is something wrong with cluster_id, which is the part of FOREIGN KEY. But wjere are no possibility (or i not finded) to know the exact query was.

dmulcahey, 2020-03-28:
Is this really a quirks issue? @prairiesnpr @Shulyaka you guys aren't getting these issues are you?

prairiesnpr, 2020-03-28:
I've not seen this issue before.  But I'm still on HA 0.105.2, maybe something changed?

Adminiuga, 2020-03-28:
What's the default endpoint id the xbee routers are discovered with?

MaximumSU, 2020-03-28:
may it be connected with this?:
https://github.com/dmulcahey/zha-device-handlers/blob/ea6d031a8494c95eaa95478580e68bfea9bf26eb/zhaquirks/xbee/xbee3_io.py#L141

MaximumSU, 2020-03-28:
Where i can set the output to logger all sqlite queries from ZHA?

Adminiuga, 2020-03-28:
[This](https://github.com/zigpy/zigpy/blob/219116501545eec476ddd5f094d59b106fced1d9/zigpy/appdb.py#L79) gets called on every attribute update received from the network. In latest version Zigpy enforces schema where `endpoint` should exist for the  updated attribute. 
In other words, if the "attribute_update" comes from an endpoint, that endpoint should be present in the "simple descriptor" when devices joins  zigpy.

So the problem is: the attr updates are coming from `endpoint_id == 230` but XBee did not report endpoint 230 being present in the simple descriptor.

Adminiuga, 2020-03-28:
What XBee module is that?

MaximumSU, 2020-03-28:
> What XBee module is that?

 [Digi XBee S2C](https://www.digi.com/support/productdetail?pid=4838)

MaximumSU, 2020-03-28:
I set the logger and now all clear:
```
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00AUJ\x1a\x81\xcf\xe8\xe8\x00\x92\xc1\x05\x01\x01\x0c\xb8\x00\x00\x88'
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:13:a2:00:41:55:4a:1a, 0x81cf, 232, 146, 1, b'010cb8000088')
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.zcl] [0x81cf:232:0x0092] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=0>
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.zcl] [0x81cf:232:0x0092] ZCL request 0x0000: [{'digital_pins': [0, 0, 0, 1, 1, 1, 0, 1, 0, 0, 1, 1, 0], 'analog_pins': [0, 0, 0, 0, 0, 0, 0, 0], 'digital_samples': [0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 0, 0, 0], 'analog_samples': [0, 0, 0, 0, 0, 0, 0, 0]}]
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.appdb] Updating attribute 0, ieee : 00:13:a2:00:41:55:4a:1a, edpoint_id: 211, cluster_id: 6
2020-03-28 19:07:45 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.appdb] Updating attribute 0, ieee : 00:13:a2:00:41:55:4a:1a, edpoint_id: 212, cluster_id: 6
2020-03-28 19:07:45 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.appdb] Updating attribute 0, ieee : 00:13:a2:00:41:55:4a:1a, edpoint_id: 213, cluster_id: 6
2020-03-28 19:07:45 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.appdb] Updating attribute 0, ieee : 00:13:a2:00:41:55:4a:1a, edpoint_id: 215, cluster_id: 6
2020-03-28 19:07:45 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.appdb] Updating attribute 0, ieee : 00:13:a2:00:41:55:4a:1a, edpoint_id: 218, cluster_id: 6
2020-03-28 19:07:45 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
2020-03-28 19:07:45 DEBUG (MainThread) [zigpy.appdb] Updating attribute 0, ieee : 00:13:a2:00:41:55:4a:1a, edpoint_id: 219, cluster_id: 6
2020-03-28 19:07:45 WARNING (MainThread) [zigpy.util] Error calling listener.attribute_updated: FOREIGN KEY constraint failed
```
As you can see, the endpoint_id is not 230

Adminiuga, 2020-03-29:
I need to see the debug log of the given XBee joining the network. Can you remove it form the network, wait a couple of  minutes and join it again. Post the log, i'd like to see the `SimpleDescriptor` for this xbee

MaximumSU, 2020-03-29:
> I need to see the debug log of the given XBee joining the network.

```
2020-03-29 12:42:22 INFO (MainThread) [zigpy_xbee.zigbee.application] New device joined: NWK 0x86aa, IEEE 00:13:a2:00:41:55:46:b7
2020-03-29 12:42:22 INFO (MainThread) [zigpy.application] Device 0x86aa (00:13:a2:00:41:55:46:b7) joined the network
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy.zdo] [0x86aa:zdo] ZDO request 0x0013: [0x86aa, 00:13:a2:00:41:55:46:b7, 142]
2020-03-29 12:42:22 INFO (MainThread) [zigpy.device] [0x86aa] Requesting 'Node Descriptor'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy.device] [0x86aa] Extending timeout for 0x11 request
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #17: b'11aa86'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 0, 0x0002, 0, 0, 64, b'\x11\xaa\x86')
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x19\x00\x13\xa2\x00AUF\xb7\x86\xaa\x00\x00\x00\x02\x00\x00\x00@\x11\xaa\x86'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00AUF\xb7\x86\xaa\x00\x00\x80\x02\x00\x00\x01\x11\x00\xaa\x86\x01@\x8e\x1e\x10R\xff\x00\x00\x00\xff\x00\x00'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 32770, 1, b'1100aa8601408e1e1052ff000000ff0000')
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x19\x86\xaa\x00\x00@'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0x86aa: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.EXTENDED_TIMEOUT, Frame #25
2020-03-29 12:42:22 INFO (MainThread) [zigpy.device] [0x86aa] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4126 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=0 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>
2020-03-29 12:42:22 INFO (MainThread) [zigpy.device] [0x86aa] Discovering endpoints
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #18: b'12aa86'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 0, 0x0005, 0, 0, 0, b'\x12\xaa\x86')
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x1a\x00\x13\xa2\x00AUF\xb7\x86\xaa\x00\x00\x00\x05\x00\x00\x00\x00\x12\xaa\x86'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00AUF\xb7\x86\xaa\x00\x00\x80\x05\x00\x00\x01\x12\x00\xaa\x86\x02\xe6\xe8'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 32773, 1, b'1200aa8602e6e8')
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x1a\x86\xaa\x00\x00\x00'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0x86aa: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #26
2020-03-29 12:42:22 INFO (MainThread) [zigpy.device] [0x86aa] Discovered endpoints: [230, 232]
2020-03-29 12:42:22 INFO (MainThread) [zigpy.endpoint] [0x86aa:230] Discovering endpoint information
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #19: b'13aa86e6'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 0, 0x0004, 0, 0, 0, b'\x13\xaa\x86\xe6')
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Sending: b'\x11\x1b\x00\x13\xa2\x00AUF\xb7\x86\xaa\x00\x00\x00\x04\x00\x00\x00\x00\x13\xaa\x86\xe6'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x91\x00\x13\xa2\x00AUF\xb7\x86\xaa\x00\x00\x80\x04\x00\x00\x01\x13\x00\xaa\x86\x08\xe6\x05\xc1\x01\x00\x00\x00\x00'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: explicit_rx_indicator
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] _handle_explicit_rx: (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 32772, 1, b'1300aa8608e605c10100000000')
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.uart] Frame received: b'\x8b\x1b\x86\xaa\x00\x00\x00'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Frame received: tx_status
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] tx_explicit to 0x86aa: TXStatus.SUCCESS after 0 tries. Discovery Status: DiscoveryStatus.SUCCESS, Frame #27
2020-03-29 12:42:22 INFO (MainThread) [zigpy.endpoint] [0x86aa:230] Discovered endpoint information: <SimpleDescriptor endpoint=230 profile=49413 device_type=1 device_version=0 input_clusters=[] output_clusters=[]>
2020-03-29 12:42:22 INFO (MainThread) [zigpy.endpoint] [0x86aa:232] Discovering endpoint information
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.zigbee.application] Zigbee request tsn #20: b'14aa86e8'
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy_xbee.api] Command tx_explicit (00:13:a2:00:41:55:46:b7, 0x86aa, 0, 0, 0x0004, 0, 0, 0, b'\x14\xaa\x86\xe8')
[... cut ...]
2020-03-29 12:42:22 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:13:a2:00:41:55:46:b7: <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2020-03-29 12:42:22 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x86aa:00:13:a2:00:41:55:46:b7 entering async_device_initialized - is_new_join: True
2020-03-29 12:42:22 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0x86aa:00:13:a2:00:41:55:46:b7 has joined the ZHA zigbee network
2020-03-29 12:42:22 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x86aa](unk_model): started configuration
................................

```

Adminiuga, 2020-03-31:
stop HA instance (so it releases zigbee.db) and run this script on your `zigbee.db`. Take backup prior
```
PRAGMA foreign_keys=on;
BEGIN TRANSACTION;
CREATE TABLE attr_back (ieee ieee, endpoint_id, cluster, attrid, value);
INSERT INTO attr_back SELECT * FROM attributes;
DROP TABLE attributes;
CREATE TABLE attributes (ieee ieee, endpoint_id, cluster, attrid, value, FOREIGN KEY(ieee) REFERENCES devices(ieee) ON DELETE CASCADE);
CREATE UNIQUE INDEX attribute_idx ON attributes(ieee, endpoint_id, cluster, attrid);
INSERT INTO attributes SELECT * FROM attr_back;
DROP TABLE attr_back;
COMMIT;
```

MaximumSU, 2020-03-31:
OK, you removed the Foreign KEY with cluster_id and set it for ieee only. Now it works without warnings, thanks. Is it the temporary or permanent fix? What is the reason to do it like it was before?

Adminiuga, 2020-04-01:
Permanent.
It was added to maintain db consistency as in the past we had issues
XBee modules when used like this do not really behave like "real zigbee devices", but instead most of functionality is implemented in software to mimic  a real HW device and that causes problem, cause all On_Off endpoint are virtual one and do not exist on the physical device

MaximumSU, 2020-04-01:
Ok

closed 2020-04-01 by MaximumSU

Adminiuga, 2020-04-01:
@Shulyaka @prairiesnpr maybe should plug `configure_reporting()` and `bind()` methods on Custom on_off cluster? Something like
```
async def configure_reporting(*args, **kwargs):
    return foundation.Status.SUCCESS.serialize()
```
?