zigpy/zha-device-handlers #309: [Device Support Request] Support for LeakSmart Water Sensor

earlthesquirrel, 2020-03-25:
Based on https://github.com/home-assistant/core/issues/23653 it seems that Leaksmart Water sensors need a quirk to function properly.

There are some logs on the referenced ticket.  I have units here that I can provide info (just need to know what you need...)   I have 3 units that I'd like to migrate off of a Wink hub onto Home Assistant (so I can then relay them into Homekit).

dmulcahey, 2020-03-28:
These devices have a completely proprietary implementation that is not standards based as mentioned in the HA thread you linked. These aren't going to be easy to implement w/o device in hand and they are outrageously expensive IIRC

earlthesquirrel, 2020-03-28:
I have three sensors on hand. I can generate whatever logs you need. 

Earl

dmulcahey, 2020-03-28:
Start by following the instructions that were in the template when you opened the ticket. Make sure you have all logging for zha, zigpy and your radio lib set to debug and trip the sensor several times and post those logs as well. If it looks doable I’ll consider trying.

cjslin, 2020-05-15:
I am exploring life-after-wink-hub options at the moment. I am sure many of you have heard about the latest Wink news. I am particularly drawn to HA with the flexibility and local hosting setup. It'll take me a little bit of time to get HA going on a RaspberryPi. I would be happy to help out with leakSmart integration efforts when I am fully ramped up. So far this is the remaining device on my list that is still not fully integrated with HA.

cjslin, 2020-06-11:
I just installed HomeAssistant on my RaspberryPi4. I pulled the docker image from dockerhub and the installation went pretty smoothly. I have a Combo Z-wave and Zigbee USB stick (model # HUSBZB-1 ). By default /dev/ttyUSB0 is Z-wave and /dev/ttyUSB1 is Zigbee. I was able to pair the Waxman Leaksmart sensor in HomeAssistant. The front panel shows not only the temperature, battery level, but also the leak status. Everything worked out of the box!

cjslin, 2020-06-11:
Thanks to the developer who fully integrated the Waxman LeakSmart sensor in HomeAssistant.

dmulcahey, 2020-06-29:
Seems like this was done in #132  and #143 and based on the report above from @cjslin I am closing this.

closed 2020-06-29 by dmulcahey