zigpy/zha-device-handlers #311: [BUG] Refactor readme supported device

HydrelioxGitHub, 2020-03-26:
I think that zigbee2mqtt supported devices page is very useful. ([https://www.zigbee2mqtt.io/information/supported_devices.html](link here) ). Can we copy the layout (image, model, supported features, etc... ) ? 
I can try to do PR for this if needed.

dmulcahey, 2020-03-26:
@Adminiuga thoughts?

Adminiuga, 2020-03-26:
Fine by me, but on the other side: maybe just join efforts with Blackadder?

dmulcahey, 2020-03-26:
> Fine by me, but on the other side: maybe just join efforts with Blackadder?

I agree

HydrelioxGitHub, 2020-03-26:
Blackadder ?

dmulcahey, 2020-03-26:
https://zigbee.blakadder.com

dmulcahey, 2020-03-26:
and you can find blakadder in the zigbee discord channel in the HA discord server

HydrelioxGitHub, 2020-03-26:
Wow I was not aware about this website. There is much more devices supported than on the readme ...

HydrelioxGitHub, 2020-03-26:
> https://zigbee.blakadder.com

This website is very well build. Why not just a link to the part with the HA filter ? It seems better to maintain only one place with updated info than two.

Adminiuga, 2020-03-26:
Totally agree. 
@dmulcahey  I still have a wish we could have a button in ZHA config panel, where user could submit their device signature with all clusters and low-level info, like software version and maybe device scan results.

dmulcahey, 2020-03-26:
@Adminiuga sure. Question is where do we send it?

Adminiuga, 2020-03-26:
open a window with template filled for user to send via email?  
Or should we run a lambda function in AWS?

dmulcahey, 2020-06-29:
@Adminiuga we have the button now to get the sig. What else do we wanna do here?

closed 2021-08-13 by github-actions[bot]