zigpy/zha-device-handlers #313: [Device Support Request] konke motion sensor 3AFE14010402000D

djaydev, 2020-03-27:
**Is your feature request related to a problem? Please describe.**
After adding Konke motion sensor to ZHA, it doesn't update battery or ias_zone.  Both stay the same state continuously.  

**Describe the solution you'd like**
I think there's a quirk already in place for this device but it doesn't show the quirk in zha_network_card (probably irrelevant), I don't know if this issue is from user error or a problem with ZHA. I'm running Home Assistant 0.107.7 with conbee2 USB stick. Let me know if I'm missing any information you need.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0xfcfb:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=123 command_id=Command.Report_Attributes>
[0xfcfb:1:0x0008] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint8_t, value=254>>]]
[0xfcfb:1:0x0008] Attribute report received: current_level=254
[0xfcfb:1:0x0008]: received attribute: 0 update with value: 254
Device 0xe447 (08:6b:d7:ff:fe:e9:7b:9e) joined the network
[0xe447:zdo] ZDO request 0x0013: [0xe447, 08:6b:d7:ff:fe:e9:7b:9e, 128]
[0xe447] Requesting 'Node Descriptor'
[0xe447] Extending timeout for 0xe2 request
[0x1ec5:1:0x0b04]: async_update
[0xe447] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xe447] Discovering endpoints
[0xe447] Extending timeout for 0xe4 request
Ignoring message (b'082c0a2000201a') on cluster 1: unknown endpoint or cluster id: 1
[0xe447:zdo] ZDO request 0x0001: [0x0000, 0, 0]
[0xe447] Discovered endpoints: [1]
[0xe447:1] Discovering endpoint information
[0xe447] Extending timeout for 0xe7 request
[0xf2c7:zdo] ZDO request 0x0000: [00:21:2e:ff:ff:05:01:c6, 0, 0]
[0xa353:1:0x0500]: Updated alarm state: 0
[0xe447:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 1280] output_clusters=[3]>
[0xe447] Extending timeout for 0xea request
[0xf8b4:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=59 command_id=Command.Report_Attributes>
[0xf8b4:1:0x0300] ZCL request 0x000a: [[<Attribute attrid=3 value=<TypeValue type=uint16_t, value=21626>>, <Attribute attrid=4 value=<TypeValue type=uint16_t, value=21626>>]]
[0xf8b4:1:0x0300] Attribute report received: current_x=21626, current_y=21626
[0xc9d7:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=53 command_id=Command.Report_Attributes>
[0xc9d7:1:0x0300] ZCL request 0x000a: [[<Attribute attrid=3 value=<TypeValue type=uint16_t, value=8912>>, <Attribute attrid=4 value=<TypeValue type=uint16_t, value=2621>>, <Attribute attrid=7 value=<TypeValue type=uint16_t, value=200>>]]
[0xc9d7:1:0x0300] Attribute report received: current_x=8912, current_y=2621, color_temperature=200
[0xe447:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=234 command_id=Command.Read_Attributes_rsp>
[0xe447:1] Manufacturer: Konke
[0xe447:1] Model: 3AFE14010402000D
Checking quirks for Konke 3AFE14010402000D (08:6b:d7:ff:fe:e9:7b:9e)
Considering <class 'zhaquirks.konke.motion.KonkeMotion'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because input cluster mismatch on at least one endpoint
device - 0xe447:08:6b:d7:ff:fe:e9:7b:9e entering async_device_initialized - is_new_join: True
device - 0xe447:08:6b:d7:ff:fe:e9:7b:9e has joined the ZHA zigbee network
[0xe447](3AFE14010402000D): started configuration
[0xe447:ZDO](3AFE14010402000D): 'async_configure' stage succeeded
[... 38 lines cut ...]
[0xe447] Extending timeout for 0x02 request
[0xff48:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=69 command_id=Command.Report_Attributes>
[0xff48:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2698>>]]
[0xff48:1:0x0402] Attribute report received: measured_value=2698
[0xe447:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=254 command_id=Command.Read_Attributes_rsp>
[0xe447:1:0x0000]: initializing channel: from_cache: False
[0xe447:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=0 command_id=Command.Read_Attributes_rsp>
[0xe447:1:0x0500]: initializing channel: from_cache: False
[0xe447:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=2 command_id=Command.Read_Attributes_rsp>
[0xe447:1:0x0001]: initializing channel: from_cache: False
[0xe447] Extending timeout for 0x04 request
[0xa787:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=32 command_id=Command.Report_Attributes>
[0xa787:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2510>>]]
[0xa787:1:0x0402] Attribute report received: measured_value=2510
[0xe447] Delivery error for seq # 0x04, on endpoint id 1 cluster 0x0001: message send failure
[0xe447:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: [0xe447:1:0x0001]: Message send failure
[0xe447:1:0x0000]: 'async_initialize' stage succeeded
[0xe447:1:0x0500]: 'async_initialize' stage succeeded
[0xe447:1:0x0001]: 'async_initialize' stage succeeded
[0xe447](3AFE14010402000D): power source: Battery or Unknown
[0xe447](3AFE14010402000D): completed initialization
```
**Additional context**
[image]
[image]

dmulcahey, 2020-03-28:
Notes for myself:

`<SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 1280] output_clusters=[3]>`

```
[0xe447:1] Manufacturer: Konke
[0xe447:1] Model: 3AFE14010402000D
```

dmulcahey, 2020-03-28:
Done in above linked PR

djaydev, 2020-03-28:
It looks a lot better now.  I added that motion.py file to my home-assistant, restarted it, and lastly re-added the device to ZHA.  It now shows a battery % value (was "?" before) and it now has a third entity "_occupancy".  

Unfortunately it still doesn't report any new information after being paired.  Not sure if the device is bad or not.  

"Add Devices" log and screenshots:
```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
Device 0x744f (08:6b:d7:ff:fe:e9:7b:9e) joined the network
[0x744f:zdo] ZDO request 0x0013: [0x744f, 08:6b:d7:ff:fe:e9:7b:9e, 128]
[0x744f] Requesting 'Node Descriptor'
[0x744f] Extending timeout for 0x23 request
[0x744f] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x744f] Discovering endpoints
[0x744f] Extending timeout for 0x25 request
[0x744f] Discovered endpoints: [1]
[0x744f:1] Discovering endpoint information
[0x744f] Extending timeout for 0x27 request
[0x5e00:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=74 command_id=0>
[0x5e00:1:0x0500] ZCL request 0x0000: [34, 0, 255, 0]
[0x5e00:1:0x0500] No handler for cluster command 0
[0x5e00:1:0x0500]: Updated alarm state: 2
[0x744f:zdo] ZDO request 0x0001: [0x0000, 0, 0]
[0x744f:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 1280] output_clusters=[3]>
[0x744f] Extending timeout for 0x2a request
[0x33e9:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=Command.Report_Attributes>
[0x33e9:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2371>>]]
[0x33e9:1:0x0402] Attribute report received: measured_value=2371
[0x744f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=42 command_id=Command.Read_Attributes_rsp>
[0x744f:1] Manufacturer: Konke
[0x744f:1] Model: 3AFE14010402000D
Checking quirks for Konke 3AFE14010402000D (08:6b:d7:ff:fe:e9:7b:9e)
Considering <class 'zhaquirks.konke.motion.KonkeMotion'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.konke.motion.KonkeMotionB'>
Found custom device replacement for 08:6b:d7:ff:fe:e9:7b:9e: <class 'zhaquirks.konke.motion.KonkeMotionB'>
device - 0x744f:08:6b:d7:ff:fe:e9:7b:9e entering async_device_initialized - is_new_join: True
device - 0x744f:08:6b:d7:ff:fe:e9:7b:9e has joined the ZHA zigbee network
[0x744f](3AFE14010402000D): started configuration
[0x744f:ZDO](3AFE14010402000D): 'async_configure' stage succeeded
[0x744f] Extending timeout for 0x2c request
[0x744f] Extending timeout for 0x2e request
[0x744f] Extending timeout for 0x30 request
[0x744f:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x744f:1:0x0000]: finished channel configuration
[0x744f] Extending timeout for 0x32 request
[0x744f:1:0x0406]: bound 'occupancy' cluster: Status.SUCCESS
[0x744f] Extending timeout for 0x34 request
[0x744f:1:0x0001]: bound 'power' cluster: Status.TABLE_FULL
[0x744f] Extending timeout for 0x36 request
[0xa353:1:0x0500]: Updated alarm state: 0
[0x8182:1:0x0500]: Updated alarm state: 0
[0x744f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=50 command_id=Command.Read_Attributes_rsp>
[0x744f:1:0x0000]: initializing channel: from_cache: False
[0x744f] Extending timeout for 0x38 request
[0x1ec5:1:0x0b04]: async_update
[0x744f:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=52 command_id=Command.Configure_Reporting_rsp>
[0x744f:1:0x0406]: reporting 'occupancy' attr on 'occupancy' cluster: 0/900/1: Result: '[[<ConfigureReportingResponseRecord status=Status.UNSUPPORTED_ATTRIBUTE direction=0 attrid=0>]]'
[0x744f:1:0x0406]: finished channel configuration
[0x744f:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=54 command_id=Command.Configure_Reporting_rsp>
[0x744f:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x744f] Extending timeout for 0x3a request
[0x744f:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=56 command_id=Command.Read_Attributes_rsp>
[0x744f:1:0x0500]: started IASZoneChannel configuration
[0x744f] Extending timeout for 0x3c request
[0x744f:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=58 command_id=Command.Configure [... cut ...]
[0xf2c7:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
[0xf2c7:1:0x0300] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint8_t, value=0>>, <Attribute attrid=1 value=<TypeValue type=uint8_t, value=254>>]]
[0xf2c7:1:0x0300] Attribute report received: current_hue=0, current_saturation=254
```
[image]
[image]

closed 2020-03-28 by dmulcahey

dmulcahey, 2020-03-28:
If you have discord msg me at the same handle there

PR #314 (2020-03-28): Konke motion alternate version
fixes #313