zigpy/zha-device-handlers #315: Cannot turn light on with attributes after being turned off [BUG]

BarrettLowe, 2020-03-29:
**Describe the bug**
After turning some light bulbs off (Osram A19 White) they will not turn back on when defining a brightness level or transition time. You must first turn them on with a brightness level of 0, then turn them on to whatever level brightness you want or whatever transition time desired.

**To Reproduce**
Steps to reproduce the behavior:
1. Call light.turn_off without any additional data on the entity in home assistant
2. Call light.turn_on without any additional data on the entity in home assistant
3. Repeat step 1
4. Call light.turn_on with a desired brightness level or transition time and observe the light not turn on

**Expected behavior**
The light turns on at the brightness described or at the transition time desired

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
- Device: A19 W 10 Year by LEDVANCE
- Using nortek zigbee/zwave stick
- [This](https://community.home-assistant.io/t/turn-light-on-with-brightness-flashing-with-previous-brightness-first/133744/15) seems like it may be the same problem...

dmulcahey, 2020-03-29:
This may belong in the HA repo. I'll peek at it when I have some time. Prob not for a bit as I have a bunch of other stuff I have to finish up.

BarrettLowe, 2020-03-29:
Sounds good. I couldn't figure out the best place for it so I started here. Let me know after you look. I'll move it if necessary