zigpy/zha-device-handlers #316: [Device Support Request] Trust Zigbee Remote Control ZYCT-202

TheTimeWalker, 2020-03-29:
A very quirky device and doesn't seem to follow almost anything. When pairing with ZHA, it shows up as a generic plug switch with no model or manufacturer name at all. It seems to receive messages though as pressing the on or off button on the remote actually switches it in ZHA.

[image]
[image]

Even more so, it allows the use of 5 different lights with a toggle button, or all at the same time. From the looks of it, this is done via creating 6 different groups and adding these lights into the group.

How the remote looks like:
[image]

The specific SimpleDescriptor:
`<SimpleDescriptor endpoint=1 profile=49246 device_type=2080 device_version=2 input_clusters=[0, 4, 3, 6, 8, 4096] output_clusters=[0, 4, 3, 6, 8, 4096]>`

Full log:
```
[0x0000:zdo] ZDO request 0x0036: [60, <Bool.false: 0>]
[0x5ef1] Extending timeout for 0x15 request
[0x5ef1] Extending timeout for 0x17 request
Device 0xadad (00:15:8d:00:01:31:44:1d) joined the network
Device 00:15:8d:00:01:31:44:1d changed id (0x5ef1 => 0xadad)
Canceling old initialize call
[0xadad:zdo] ZDO request 0x0013: [0xadad, 00:15:8d:00:01:31:44:1d, 128]
Failed ZDO request during device initialization
[0xdb1b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=14 command_id=Command.Report_Attributes>
[0xdb1b:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>]]
[0xdb1b:1:0x0006] Attribute report received: on_off=Bool.true
[0xadad] Delivery error for seq # 0x15, on endpoint id 0 cluster 0x0002: message send failure
[0xadad] Requesting Node Descriptor failed
[0xadad] Discovering endpoints
[0xadad] Extending timeout for 0x19 request
[0xadad] Requesting 'Node Descriptor'
[0xadad] Extending timeout for 0x1b request
[0xadad] Discovered endpoints: [1, 2]
[0xadad:1] Discovering endpoint information
[0xadad] Extending timeout for 0x1d request
[0xadad] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>
[0xadad:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=49246 device_type=2080 device_version=2 input_clusters=[0, 4, 3, 6, 8, 4096] output_clusters=[0, 4, 3, 6, 8, 4096]>
[0xadad] Extending timeout for 0x1f request
[0xadad:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=31 command_id=Command.Read_Attributes_rsp>
[0xadad:1] Manufacturer: 
[0xadad:1] Model: 
[0xadad:2] Discovering endpoint information
[0xadad] Extending timeout for 0x21 request
[0xadad:2] Discovered endpoint information: <SimpleDescriptor endpoint=2 profile=49246 device_type=1010 device_version=2 input_clusters=[4096] output_clusters=[4096]>
Checking quirks for   (00:15:8d:00:01:31:44:1d)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
device - 44461:00:15:8d:00:01:31:44:1d entering async_device_initialized - is_new_join: True
device - 44461:00:15:8d:00:01:31:44:1d has joined the ZHA zigbee network
[44461](): started configuration
[44461:ZDO](): 'async_configure' stage succeeded
[0xadad] Extending timeout for 0x23 request
[0xadad] Extending timeout for 0x25 request
[0xadad] Extending timeout for 0x27 request
Device 0xadad (00:15:8d:00:01:31:44:1d) joined the network
Skip initialization for existing device 00:15:8d:00:01:31:44:1d
[0xadad:zdo] ZDO request 0x0013: [0xadad, 00:15:8d:00:01:31:44:1d, 128]
[44461:1:0x1000]: Failed to bind 'lightlink' cluster: 
[44461:1:0x1000]: finished channel configuration
[0xadad] Extending timeout for 0x29 request
[44461:1:0x0000]: Failed to bind 'basic' cluster: 
[44461:1:0x0000]: finished channel configuration
[0xadad] Extending timeout for 0x2b request
[44461:1:0x0006]: Failed to bind 'on_off' cluster: 
[0xadad] Extending timeout for 0x2d request
[0xadad] Delivery error for seq # 0x29, on endpoint id 0 cluster 0x0021: message send failure
[44461:1:0x0008]: Failed to bind 'level' cluster: [0xadad:0:0x0021]: Message send failure
[44461:1:0x0008]: finished channel configuration
[0xadad] Extending timeout for 0x2f request
[0xadad] Delivery error for seq # 0x2b, on endpoint id 1 cluster 0x0000: message send failure
[44461:1:0x0000]: initializing channel: from_cache: False
[0xadad] Extending timeout for 0x31 request
[0xadad] Delivery error for seq # 0x2d, on endpoint id 1 cluster 0x0006: message send failure
[44461:1:0x0006]: failed to set reporting for 'on_off' attr on 'on_off' cluster: [0xadad:1:0x0006]: Message send failure
[44461:1:0x0006]: finished channel configuration
[0x6e13:11:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=110 command_id=1>
[0x6e13:11:0x0019] ZCL request 0x0001: [0, 4478, 0, 1, None]
[0x6e13:11:0x0019] OTA query_next_image handler for 'FeiBit FNB56-SKT1JXN1.0': field_control=0, manufacture_id=4478, image_type=0, current_file_version=1, hardware_version=None
[0x6e13:11:0x0019] No OTA image is available
[0xadad] Delivery error for seq # 0x2f, on endpoint id 0 cluster 0x0021: message send failure
[44461:1:0x0006]: Failed to bind 'on_off' cluster: [0xadad:0:0x0021]: Message send failure
[44461:1:0x0006]: finished channel configuration
[44461:1:0x1000]: 'async_configure' stage succeeded
[44461:1:0x0000]: 'async_configure' stage succeeded
[44461:1:0x0006]: 'async_configure' stage succeeded
[44461:1:0x0008]: 'async_configure' stage succeeded
[44461:1:0x0006]: 'async_configure' stage succeeded
[0xadad] Delivery error for seq # 0x31, on endpoint id 0 cluster 0x0021: message send failure
[44461:2:0x1000]: Failed to bind 'lightlink' cluster: [0xadad:0:0x0021]: Message send failure
[44461:2:0x1000]: finished channel configuration
[44461:2:0x1000]: 'async_configure' stage succeeded
[44461](): completed configuration
[44461](): stored in registry: ZhaDeviceEntry(name=' ', ieee='00:15:8d:00:01:31:44:1d', last_seen=1585514703.579955)
[0xadad] Extending timeout for 0x34 request
[0xadad:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=0 command_id=64>
[0xadad:1:0x0006] ZCL request 0x0040: [0, 0]
[0xadad:1:0x0006] No handler for cluster command 64
[44461:1:0x0006]: received 'off_with_effect' command with [0, 0] args on cluster_id '6' tsn '0'
[0xadad:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=52 command_id=Command.Default_Response>
[44461:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[44461](): started initialization
[44461:ZDO](): 'async_initialize' stage succeeded
[44461:1:0x1000]: initializing channel: from_cache: False
[0xadad] Extending timeout for 0x36 request
[0xadad] Extending timeout for 0x38 request
[44461:1:0x0008]: initializing channel: from_cache: False
[44461:1:0x0006]: initializing channel: from_cache: False
[44461:2:0x1000]: initializing channel: from_cache: False
[44461:1:0x0006]: attempting to update onoff state - from cache: True
[44461:2:0x1000]: 'async_initialize' stage succeeded

```

dmulcahey, 2020-03-29:
This is gonna be fun to figure out. Will do some research when I have a bit of time

TheTimeWalker, 2020-03-29:
For completion sake and as extra info, these are the debug logs when receiving an ON state when the light button has been toggled:

Light 1:
```
[zigpy_deconz.uart] Frame received: 0x0e8e000700aa20
[zigpy_deconz.api] Device state changed response: [170, 32]
[zigpy_deconz.api] Command Command.aps_data_indication (1, 4)
[zigpy_deconz.uart] Send: 0x178e000800010004
[zigpy_deconz.uart] Frame received: 0x178e0032002b00220200000104136e17dc4904008d15000b040119000c00018801007e1100000100000000af2da3000102b6
[zigpy_deconz.api] APS data indication response: [43, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e13>, 11, 260, 25, b'\x01\x88\x01\x00~\x11\x00\x00\x01\x00\x00\x00', 0, 175, 45, 163, 0, 1, 2, -74]
[zigpy.zcl] [0x6e13:11:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=136 command_id=1>
[zigpy.zcl] [0x6e13:11:0x0019] ZCL request 0x0001: [0, 4478, 0, 1, None]
[zigpy.zcl] [0x6e13:11:0x0019] OTA query_next_image handler for 'FeiBit FNB56-SKT1JXN1.0': field_control=0, manufacture_id=4478, image_type=0, current_file_version=1, hardware_version=None
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e13>, ep: 11, profile: 0x0104, cluster_id: 0x0019, data: b'018801007e11000001000000'
[zigpy.zcl] [0x6e13:11:0x0019] No OTA image is available
[zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 136 under 236 request id, data: b'19880298'
[zigpy_deconz.api] Command Command.aps_data_request (19, 236, 0, <DeconzAddressEndpoint address_mode=2 address=0x6e13 endpoint=11>, 260, 25, 1, b'\x19\x88\x02\x98', 2, 0)
[zigpy_deconz.uart] Send: 0x128f001a001300ec0002136e0b04011900010400198802980200
[zigpy_deconz.uart] Frame received: 0x128f000900020022ec
[zigpy_deconz.api] APS data request response: [2, 34, 236]
[zigpy_deconz.uart] Frame received: 0x0e90000700a620
[zigpy_deconz.api] Device state changed response: [166, 32]
[zigpy_deconz.api] Command Command.aps_data_confirm (0,)
[zigpy_deconz.uart] Send: 0x04900007000000
[zigpy_deconz.uart] Frame received: 0x04900013000c0022ec02136e0b010000000000
[zigpy_deconz.api] APS data confirm response for request with id 236: 00
[zigpy_deconz.api] Request id: 0xec 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x6e13 endpoint=11>, status: 0x00
```

Light 2:
```
[zigpy_deconz.uart] Frame received: 0x0e91000700aa20
[zigpy_deconz.api] Device state changed response: [170, 32]
[zigpy_deconz.api] Command Command.aps_data_indication (1, 4)
[zigpy_deconz.uart] Send: 0x1791000800010004
[zigpy_deconz.uart] Frame received: 0x17910029002200220191001d04adad1d443101008d150001040106000300110b0100afffa3000102ca
[zigpy_deconz.api] APS data indication response: [34, 34, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0091>, 29, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xadad>, 1, 260, 6, b'\x11\x0b\x01', 0, 175, 255, 163, 0, 1, 2, -54]
[zigpy.zcl] [0xadad:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=11 command_id=1>
[zigpy.zcl] [0xadad:1:0x0006] ZCL request 0x0001: []
[zigpy.zcl] [0xadad:1:0x0006] No handler for cluster command 1
[homeassistant.components.zha.core.channels.base] [0xadad:1:0x0006]: received 'on' command with [] args on cluster_id '6' tsn '11'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xadad>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'110b01'
[zigpy_deconz.uart] Frame received: 0x0e92000700aa20
[zigpy_deconz.api] Device state changed response: [170, 32]
[zigpy_deconz.api] Command Command.aps_data_indication (1, 4)
[zigpy_deconz.uart] Send: 0x1792000800010004
[zigpy_deconz.uart] Frame received: 0x17920029002200220192001d04adad1d443101008d150001040106000300110c0100afffa3000102ca
[zigpy_deconz.api] APS data indication response: [34, 34, <DeconzAddress address_mode=ADDRESS_MODE.GROUP address=0x0092>, 29, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xadad>, 1, 260, 6, b'\x11\x0c\x01', 0, 175, 255, 163, 0, 1, 2, -54]
[zigpy.zcl] [0xadad:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=12 command_id=1>
[zigpy.zcl] [0xadad:1:0x0006] ZCL request 0x0001: []
[zigpy.zcl] [0xadad:1:0x0006] No handler for cluster command 1
[homeassistant.components.zha.core.channels.base] [0xadad:1:0x0006]: received 'on' command with [] args on cluster_id '6' tsn '12'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xadad>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'110c01'
```

Light 3:
```
[zigpy_deconz.uart] Frame received: 0x0e93000700aa20
[zigpy_deconz.api] Device state changed response: [170, 32]
[zigpy_deconz.api] Command Command.aps_data_indication (1, 4)
[zigpy_deconz.uart] Send: 0x1793000800010004
[zigpy_deconz.uart] Frame received: 0x17930029002200220193001d04adad1d443101008d150001040106000300110d0100afffa3000104c9
[... cut ...]
[zigpy.zcl] [0xadad:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=13 command_id=1>
[zigpy.zcl] [0xadad:1:0x0006] ZCL request 0x0001: []
[zigpy.zcl] [0xadad:1:0x0006] No handler for cluster command 1
[homeassistant.components.zha.core.channels.base] [0xadad:1:0x0006]: received 'on' command with [] args on cluster_id '6' tsn '13'
[zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xadad>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'110d01'
```

timmo001, 2020-06-05:
:+1: on this. Setting up was a mess and the only thing I can get to work is the on and off button. The groups and dimmer buttons just do nothing

Vuk996, 2020-06-14:
This is the only remote control that allows you to control more than one device, so it would be great if it could be supported.
Can I donate the remote control to you so that you can play and code "better" with it?

denilsonsa, 2020-12-21:
I'm also interested in using this device. If this helps anyone working on it, here are some links with further info people have figured out:

* https://community.home-assistant.io/t/trust-zyct-202-setting-up-dimming-via-zigbee2mqtt/174206
* https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/222
* https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/blob/master/adapters/trust/zyct202.py
* https://zigbee.blakadder.com/Trust_ZYCT-202.html

Indeed it seems to be using groups/addresses from 145 to 150 (or 0x0091 to 0x0096) for each light selection from 1 to 5, while the last group (150 or 0x0096) means all lights.

christophebaraer, 2021-01-09:
I recently purchased two of these without thinking whether they were supported or not. Sadly, I have the same problems meaning that in Home Assistant, it is only "recognized" as a switch and in contrary to @TheTimeWalker, I don't get any RSSI value nor any Zigbee events when listening to zha_events in Home Assistant. 

I concur with @Vuk996 in the possibility of this remote and I was hoping to hand this over to the kids to control the lights in their rooms. 

Here's another link to add to the list shared by @denilsonsa : https://github.com/Koenkk/zigbee2mqtt/issues/635

TheTimeWalker, 2021-08-06:
Nothing yet

Ulrar, 2021-11-03:
I've also got one of these. I used to have it bound directly to some lights but it forgot it's configuration after a while, so I was hoping to just use HA automations instead.
Guess that's not going to happen, would have been nice.

TheTimeWalker, 2022-05-02:
Nothing yet

denilsonsa, 2022-05-02:
I wish [this device was fully supported in ZHA](https://zigbee.blakadder.com/Trust_ZYCT-202.html), as I already have one, and it supports 5+1 different groups. (That means (5+1) groups × 4 buttons = 24 "virtual" buttons.) But I'm losing my hope.

It doesn't help that the manufacturer/distributor website says [“This product is no longer available in the current Trust assortment.”](https://www.trust.com/en/product/71171-zigbee-remote-control-zyct-202). It also doesn't help that there are TWO very similar-looking devices, but operating in different protocols:

* [ZYCT-202](https://klikaanklikuit.nl/product/zigbee-afstandsbediening-met-muurbevestiging/), using Zigbee.
* [AYCT-202](https://klikaanklikuit.nl/product/afstandsbediening-met-muurbevestiging-ayct202/), using KlikAanKlikUit protocol (433MHz). This is NOT Zigbee, and NOT relevant to this zha-device-handlers project.

These two devices are very similar-looking, but they have one visual difference: the Zigbee version has a button with numbers "1 2 3 4 5", while on the other version the same button has a slightly different label "1 2 3 4 G" (note the letter "G" instead of the number "5").

Unfortunately, [some stores mix up pictures of both devices in the same listing](https://www.gamma.nl/assortiment/klikaanklikuit-trust-zigbee-afstandsbediening-zyct-202/p/B603327). To make matters worse, even [the official Trust website is using the wrong pictures for the Zigbee version](https://www.trust.com/en/product/71171-zigbee-remote-control-zyct-202). So, before buying, please double-check the device model, and also read the return policy of the store.

---

Talking about buying… @dmulcahey mentioned he was going to take a look into this device, two years ago. Sure, priorities change, and this is volunteer work after all (and very much appreciated). Is there anything we can do to help? Does any developer experienced with this repository need this device for debugging? I guess some people here (myself included) wouldn't mind to donate (or lend) this device for such person.

(I wish I could implement it myself, but I have no idea how to start, and I don't have time to do it nowadays.)

Ulrar, 2022-10-29:
Not stale

denilsonsa, 2023-04-29:
Well, while technically not stale yet, I don't think this device is still being manufactured or sold anywhere (other than old stock), and I realistically don't see this issue ever being fixed.

closed 2023-11-02 by github-actions[bot]