zigpy/zha-device-handlers #317: [Device Support Request] Develco WISZB-120 door/windows sensor

magnusoverli, 2020-03-30:
**Is your feature request related to a problem? Please describe.**
This sensor is currently only working for the temperature sensor onboard. The contact sensor doesn't report state changes in HA. Also, battery sensor is not reporting any values at all.

**Describe the solution you'd like**
Support for the sensor in ZHA.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
https://paste.ubuntu.com/p/ST3Y6k9fXf/

dmulcahey, 2020-03-30:
```
Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=0 manufacturer_code=4117 maximum_buffer_size=80 maximum_incoming_transfer_size=80 server_mask=0 maximum_outgoing_transfer_size=80 descriptor_capability_field=0>

<SimpleDescriptor endpoint=1 profile=49353 device_type=1 device_version=1 input_clusters=[3, 5, 6] output_clusters=[]>

<SimpleDescriptor endpoint=35 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 15, 32, 1280] output_clusters=[10, 25]>

<SimpleDescriptor endpoint=38 profile=260 device_type=770 device_version=0 input_clusters=[0, 3, 1026] output_clusters=[3]>

Manufacturer: Develco Products A/S
Model: WISZB-120
```

dmulcahey, 2020-03-30:
<https://github.com/Koenkk/zigbee2mqtt/issues/3012> device appears to have a firmware problem causing it to send an invalid payload on status change messages for the IAS Zone cluster

Adminiuga, 2020-03-31:
```
Failed to parse message (b'090d00350000') on cluster 1280, because Data is too short to contain 1 bytes
...
2020-03-30 18:15:07 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4453>, 35, 260, 1280, b'\tI\x005\x00\x00\x00\xdc\x1a', 0, 175, 151, 20, 130, 16, 0, -75]
2020-03-30 18:15:07 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4453>, ep: 35, profile: 0x0104, cluster_id: 0x0500, data: b'09490035000000dc1a'
2020-03-30 18:15:07 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [29, 34, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4453>, 35, 260, 1280, b'\tI\x005\x00\x00', 0, 175, 143, 20, 130, 16, 0, -76]
2020-03-30 18:15:07 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x4453>, ep: 35, profile: 0x0104, cluster_id: 0x0500, data: b'094900350000'
```

as a test, modify `/usr/local/lib/python3.7/site-packages/zigpy/zcl/clusters/security.py` inside your homeassistant container and around line 80 modify schema for `"status_change_notification"` command:
```
    client_commands = {
        0x0000: (
            "status_change_notification",
            (ZoneStatus, t.bitmap8, t.Optional(t.uint8_t), t.Optional(t.uint16_t)),
            False,
        ),
```

the only change is to last two `t.uint8_t` and `t.uint16_t` making those optional. Just wanna to test if a quirk would work.

masto, 2020-06-28:
FTR, that change to status_change_notification (plus a quirks handler) was exactly what I needed to get my old SmartThings multi sensors working. It's the one halfway down [this page](https://support.smartthings.com/hc/en-us/articles/214413386), called "SmartSense Multi Sensor", and it identifies itself as a SmartThings PGC313.

I'm working on the handler now and will open a PR, but I was getting stuck on this malformed packet.

Adminiuga, 2020-06-28:
@dmulcahey I'm split whether that change should be done in ZCL or in quirk. Specs clearly state it should send those two params.

dmulcahey, 2020-06-28:
Then quirk it 🤷‍♂️

PR #404 (2020-07-10): Support for Develco WISZB-120 sensor
Develco IAS cluster doesn't send all required commands. 
Fixes #317
closed 2020-07-10 by dmulcahey