zigpy/zha-device-handlers #322: [Device Support Request] Wyze Lock

tube0013, 2020-04-05:
**Is your feature request related to a problem? Please describe.**

Wyze's refusal to help

**Describe the solution you'd like**

Hopefully this logging helps decode.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`2020-04-04 23:27:40 INFO (MainThread) [zigpy.application] Device 0xe381 (cc:cc:cc:ff:fe:c1:f7:c7) joined the network
2020-04-04 23:27:40 DEBUG (MainThread) [zigpy.zdo] [0xe381:zdo] ZDO request ZDOCmd.Device_annce: [0xe381, cc:cc:cc:ff:fe:c1:f7:c7, 128]
2020-04-04 23:27:40 INFO (MainThread) [zigpy.device] [0xe381] Requesting 'Node Descriptor'
2020-04-04 23:27:40 DEBUG (MainThread) [zigpy.device] [0xe381] Extending timeout for 0xc0 request
2020-04-04 23:27:40 INFO (MainThread) [zigpy.device] [0xe381] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4698 maximum_buffer_size=82 maximum_incoming_transfer_size=512 server_mask=11264 maximum_outgoing_transfer_size=512 descriptor_capability_field=0>
2020-04-04 23:27:40 INFO (MainThread) [zigpy.device] [0xe381] Discovering endpoints
2020-04-04 23:27:40 DEBUG (MainThread) [zigpy.device] [0xe381] Extending timeout for 0xc2 request
2020-04-04 23:27:40 INFO (MainThread) [zigpy.device] [0xe381] Discovered endpoints: [1]
2020-04-04 23:27:40 INFO (MainThread) [zigpy.endpoint] [0xe381:1] Discovering endpoint information
2020-04-04 23:27:40 DEBUG (MainThread) [zigpy.device] [0xe381] Extending timeout for 0xc4 request
2020-04-04 23:27:40 INFO (MainThread) [zigpy.endpoint] [0xe381:1] Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=10 device_version=1 input_clusters=[0, 1, 3, 32, 257, 2821, 64512] output_clusters=[10, 25, 64512]>`

**Additional context**

Did  bunch of  actions/recording zigpy debug of the bytestream, hoping it can be useful for de-coding:

https://paste.ubuntu.com/p/sFcB4QFFcS/

puddly, 2020-05-28:
This is very disappointing, though not entirely surprising given Wyze's track record. Have you tried cracking open or MITMing the gateway?

tube0013, 2020-05-28:
I've sniffed the zigbee traffic between the gateway and the lock and it doesn't do anything specifically different via zigbee. the Wyze app communicates directly with the lock over BLE when it is local and that is how it does the calibration. pretty sure the gateway just just records the same data that @dmulcahey has figured out are needed during it's calibration process.

closed 2021-08-13 by github-actions[bot]