zigpy/zha-device-handlers #323: [BUG]

ocalvo, 2020-04-06:
**Describe the bug**
Spruce humidity sensor Gen1 and Gen3 correctly report humidity and temperature but do not report battery correctly. 
The reported battery is always 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a Spruce Sensor Gen1 or Gen3 with home assistant using ZHA integration.
2. Notice the battery sensor

**Expected behavior**
Battery reported should not be 0

**Additional context**
```
TBD
```

dmulcahey, 2020-04-07:
Please remove the device via the ZHA config panel. Wake the device up when you do this. Wait about a minute and then pair it again with the add devices screen. Put the entire content of the log panel in a code block here after successfully pairing the device.

ocalvo, 2020-04-07:
I just paired the device, is there a way to look at the log from a few hours ago?
Otherwise this will take me sometime to obtain. Is that OK?

dmulcahey, 2020-04-07:
The logs should be in the homeassistant logs. It’s fine if it takes a while, we just can’t act on the ticket without the data. There are several tickets that I’m behind on so you have plenty of time.

ocalvo, 2020-04-07:
I using hassio, the file /config/home-assistant.log does not have any output regarding zha internal messages. What am I missing?

tube0013, 2020-04-07:
There is a pairing log for the gen 3 model in the closed issue #291. The battery is on the mains power attribute (confirmed with spruce tech support). This is not to spec so the quirk needs to be adjusted for it.

dmulcahey, 2020-04-07:
Doh, I didn’t realize it was for this device. Yeah I’m all set on the logs. Thanks @tube0013 for pointing that out!

ocalvo, 2020-04-07:
[My version of home assistant](https://github.com/home-assistant/core/blob/0.107.7/homeassistant/components/zha/manifest.json#L8) is using version 0.0.37 of zha-quirks.

closed 2020-04-15 by dmulcahey

ocalvo, 2020-04-29:
Running version 0.0.38 and still repro.
https://github.com/home-assistant/core/blob/0.109.0/homeassistant/components/zha/manifest.json#L8

dmulcahey, 2020-04-29:
Didn’t bump the module in 109. Will be in 110

ocalvo, 2020-05-21:
This still repro on the dev branch of core home assistant:
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](EZSP): started initialization
[0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[0x0000](EZSP): power source: Mains
[0x0000](EZSP): completed initialization
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xe440 (00:0d:6f:00:0c:1c:b4:d7) joined the network
[0xe440:zdo] ZDO request ZDOCmd.Device_annce: [0xe440, 00:0d:6f:00:0c:1c:b4:d7, 128]
[0xe440] Requesting 'Node Descriptor'
Tries remaining: 2
[0xe440] Extending timeout for 0x05 request
[0xe440] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xe440] Discovering endpoints
Tries remaining: 3
[0xe440] Extending timeout for 0x07 request
[0xe440] Discovered endpoints: [1]
[0xe440:1] Discovering endpoint information
Tries remaining: 3
[0xe440] Extending timeout for 0x09 request
[0xe440:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=1029 device_version=0 input_clusters=[0, 1, 3, 1026, 1029] output_clusters=[3, 25]>
[0xe440] Extending timeout for 0x0b request
[0xe440:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=Command.Read_Attributes_rsp>
[0xe440:1] Manufacturer: PLAID SYSTEMS
[0xe440:1] Model: PS-SPRZMS-SLP3
Checking quirks for PLAID SYSTEMS PS-SPRZMS-SLP3 (00:0d:6f:00:0c:1c:b4:d7)
Considering <class 'zhaquirks.plaid.soil.SoilMoisture'>
Found custom device replacement for 00:0d:6f:00:0c:1c:b4:d7: <class 'zhaquirks.plaid.soil.SoilMoisture'>
device - 0xe440:00:0d:6f:00:0c:1c:b4:d7 entering async_device_initialized - is_new_join: True
device - 0xe440:00:0d:6f:00:0c:1c:b4:d7 has joined the ZHA zigbee network
[0xe440](PS-SPRZMS-SLP3): started configuration
[0xe440:ZDO](PS-SPRZMS-SLP3): 'async_configure' stage succeeded
[0xe440] Extending timeout for 0x0d request
[0xe440] Extending timeout for 0x0f request
[0xe440] Extending timeout for 0x11 request
[0xe440:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xe440] Extending timeout for 0x13 request
[0xe440:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0xe440] Extending timeout for 0x15 request
[0xe440:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0xe440] Extending timeout for 0x17 request
[0xe440:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe440] Extending timeout for 0x19 request
[0xe440:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe440:1:0x0402]: finished channel configuration
[0xe440] Extending timeout for 0x1b request
[0xe440:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0405]: reporting 'measured_value' attr on 'humidity' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe440:1:0x0405]: finished channel configuration
[0xe440] Extending timeout for 0x1d request
[0xe440:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.INSUFFICIENT_SPACE direction=ReportingDirection.SendReports attrid=33>]]'
[0xe440:1:0x0001]: finished channel configuration
[0xe440:1:0x0000]: bound 'basic' cluster: Status.TABLE_FULL
[0xe440:1:0x0000]: finished channel configuration
[0xe440] Extending timeout for 0x1f request
[0xe440:1:0x0019]: bound 'ota' cluster: Status.TABLE_FULL
[0xe440:1:0x0019]: finished channel configuration
[0xe440:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=31 command_id=Command.Read_Attributes_rsp>
[0xe440:1:0x0000]: initializing channel: from_cache: False
[0xe440:1:0x0001]: 'async_configure' stage succeeded
[0xe440:1:0x0402]: 'async_configure' stage succeeded
[0xe440:1:0x0405]: 'async_configure' stage succeeded
[0xe440:1:0x0000]: 'async_configure' stage succeeded
[0xe440:1:0x0019]: 'async_configure' stage succeeded
[0xe440](PS-SPRZMS-SLP3): completed configuration
[... cut ...]
[0xe440:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2218>>]]
[0xe440:1:0x0402] Attribute report received: measured_value=2218
[0xe440:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=115 command_id=Command.Report_Attributes>
[0xe440:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
[0xe440:1:0x0405] Attribute report received: measured_value=0
```

PR #329 (2020-04-15): Make Plaid Systems soil sensor use mains voltage
Fixes: #323