zigpy/zha-device-handlers #324: [Device Support Request] OSRAM Switch-LIGHTIFY

docgalaxyblock, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
This switch version is not yet supportet

**Describe the solution you'd like**
Complete device support
Primary buttons and secondary the battery (seems to report the voltage but HA says unknown)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
https://pastebin.com/ZTE1nkrr

**Additional context**
zha_event log pressing all buttons
https://pastebin.com/Lm7W210A

dmulcahey, 2020-04-07:
The device wasn’t removed. You need to press the remove device button on the device card in the ZHA config panel then press a button on the remote to wake it up. Wait about a minute then re-pair the device

docgalaxyblock, 2020-04-07:
I've waited 5min between removing and adding it again...
Will setup a VM and pair it there...then it should be ok

dmulcahey, 2020-04-07:
Did you wake the remote up when removing it? It’s a battery powered device. You have to ensure it isn’t sleeping or it won’t remove

docgalaxyblock, 2020-04-07:
ehm nope but I have reset the remote

dmulcahey, 2020-04-07:
Yeah that’s not gonna remove the data from zigpy. Pair it, remove it while waking it up to make sure that it’s removed for real. Then pair it.

docgalaxyblock, 2020-04-07:
After I successfully destroyed my ZHA I think I got it right

dmulcahey, 2020-04-07:
Yep paste looks good. I’ll take a look in a bit.

docgalaxyblock, 2020-04-13:
@dmulcahey Do you know when this will be avaible in homeassistant?

dmulcahey, 2020-04-13:
109

closed 2020-04-15 by dmulcahey

docgalaxyblock, 2020-04-29:
@dmulcahey
I updated top 109 today, still No quirk.
(Removed it completely)

dmulcahey, 2020-04-29:
Yeah I didn’t push an update into 109. Sorry about that.

docgalaxyblock, 2020-04-29:
Can you bush it to next beta?

dmulcahey, 2020-04-29:
Won’t be any betas until 110

docgalaxyblock, 2020-04-29:
I mean the 110b1

docgalaxyblock, 2020-05-29:
@dmulcahey Hi I got some problems regarding this device
I have three bitron video power metering plugs that turn off if I press the down left button (button 2 )
What I have done:

- New HA VW with different CC2531 stick

- Added the integration and definied the stick

- Added the smart plug and the osram switch

- Waited until the adding process finishes

- Then stayed at the connection dialog (made no manuall changes anythere at HA)

- Press the second button at the switch (down left one) and the plug turned off

Should I activate the ZHA debug logger and send you the logs?

https://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/smart-plug-mit-verbrauchsdatenerfassung/

dmulcahey, 2020-05-29:
Did you directly bind them at any point?

docgalaxyblock, 2020-05-29:
No I never touched the binding because I could not find any documentatin about it

but in my opinion they are binded because I removed the coordinator ram my main and my testinh HA and the switch turns them off.
I could not understand why the switch turns all off. it should only act to the binded one right?

docgalaxyblock, 2020-06-14:
So I today got an conbee2 to play with and tested the Osram Switch again.
The Switch talks directly to all devices, I connected a hue lamp to my ZHA and the unconfigured switch controlls the hole lamp (on/off, brightness, hue and color temp)
So it is a switch bug, I will take a look to purchased an OSRAM Gateway to update the firmware. Zigbe2mqtt is not able to update this device more specifically the switch is not supported by Z2M

docgalaxyblock, 2020-06-14:
Update: The device has some problems...

#https://github.com/Koenkk/zigbee2mqtt/issues/962
#https://github.com/Koenkk/zigbee-herdsman-converters/pull/1112
#https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2917

unromeo, 2020-09-21:
Any updates on this issue ? The OSRAM switch mini is working without issues in zigbee2mqtt, they somehow managed to solve the problems they had with it. Only in ZHA, the broadcasting is still present. I migrated everything from zigbee2mqtt but this is the only device I am not able to use inside ZHA.

floriandotorg, 2023-01-08:
Is there any update on this? I just bought the switch and have this problem. @gnaggi, @unromeo could you find a solution? Thanks!

TheJulianJES, 2023-01-08:
@floriandotorg Please create a new issue and fill-out the issue template.

PR #328 (2020-04-15): Add alternate version of Osram Lightly 4x switch
fixes #324