zigpy/zha-device-handlers #325: [BUG] Samsung Motionv4 works, but v5 does not load the quirk

zaneclaes, 2020-04-07:
**Describe the bug**
I have two Samsung SmartThings motion devices. One is v4, one is v5. 

The v4 correctly loads the quirk (screenshot 1).  The v5 does not (screenshot 2). As a result, the battery sensor does not work (battery icon in screenshot 2).

AFAICT, this should be supported ([source code](https://github.com/zigpy/zha-device-handlers/blob/1f8659ac6210ab83ae431676db4ba9dbfb70be83/zhaquirks/smartthings/motion.py#L21-L46)).

**To Reproduce**
Just add a v5 motion device.

**Expected behavior**
SmartThings Motion v5 should also load the appropriate quirk.

**Screenshots**
[image]

[image]

dmulcahey, 2020-04-07:
Are you on the beta of HA?

zaneclaes, 2020-04-07:
@dmulcahey nope; `0.107.7` running in Docker. Should I be?
Thanks for the fast response btw.

dmulcahey, 2020-04-07:
Will be in 108

zaneclaes, 2020-04-07:
Just upgraded to `0.108.0b5` (via `homeassistant/home-assistant:rc`). The quirk is now loaded, but the battery still shows as "unknown":
[image]
[image]

dmulcahey, 2020-04-07:
It will take time to report, you can force it by selecting the powerconfiguration cluster and selecting the battery_voltage attribute in the zha config panel. Then click read attribute and trip the device so that it wakes up.

David F. Mulcahey

zaneclaes, 2020-04-07:
It seems a manual reset and re-pairing of the motion sensor was the only thing that fixed it.

What I tried first:
* Wait a half hour; confirm the RSSI was present
* `Reconfigure Device`
* Restart Home Assistant

Thanks for the fix though!

closed 2020-04-07 by zaneclaes