zigpy/zha-device-handlers #326: Device Support Request: Smartthings Presence sensor

ocalvo, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Smartthings presence sensor does not expose any entities to homeassistant.
Model: SmartThings PGC410EU

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
[0xba60:1:0x0702] Attribute report received: instantaneous_demand=145
Ignoring message (b'1d0a110701') on cluster 64517: unknown endpoint or cluster id: 'No cluster ID 0xfc05 on (d0:52:a8:1e:92:c6:00:05, 2)'
[0xee80:zdo] ZDO request 0x0000: [d0:52:a8:16:f6:d7:00:01, 0, 0]
```

dmulcahey, 2020-04-07:
That’s not the signature. Please remove the device and repair it and provide the correct data.

dmulcahey, 2020-04-16:
Bump... are you able to provide this info?

dmulcahey, 2020-04-20:
@ocalvo since you have other issues since this (that have been resolved) I'll wait until Friday for the requested info then I am going to close this.

ocalvo, 2020-04-20:
I will get this data today EOD.

ocalvo, 2020-04-20:
Done, updated issue description with the info

dmulcahey, 2020-04-23:
Thanks. This is going to be interesting... not sure how this works TBH

g-enzo, 2020-06-07:
I have this same issue, is there any further help on this?

ocalvo, 2020-09-26:
Any update on this?

I have development environment that I can use to hack something around, but I would love to have some place to start.

Is there another ZigBee presence device that works that I could use as baseline?

andyjenkinson, 2021-02-03:
To add another datapoint, I’d quite like to get this working but depends if ZHA itself supports presence as a possible entity, I don’t really want to get into modifying ZHA. I don’t need it that badly :)

As far as I can tell the device has 2 endpoints, one only has a simple basic info (0x0000) in cluster; the second has a proprietary profile but I expect this is used to play a sound on the device from the SmartThings app.
You may find the smartthings handler useful:
https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/arrival-sensor.src/arrival-sensor.groovy

My expectation of this device is that presence is purely based on whether the device is online and reporting messages or not. I don’t think smartthings ever reports the device as being offline (I think this is what registering it to the devicewatch service as “untracked” means), but listens for periodic status messages to update either battery/RSSI/LQI or the presence state (think it’s two types of message content). Whilst to the handler they look like parsing incoming messages I’m not 100% sure if they are really push events or responses to polling the info endpoint though.

I think it should be possible to work out how it works based on the smartthings handler, but aside from masking the offline state with a static value, what should the quirk expose to ZHA to support presence?

andyjenkinson, 2021-02-03:
To add: seems like this message, wherever it comes from, looks simply like this:

presence: 1
or
presence: 0

I am a little perplexed though. I was thinking perhaps they do this (ie the device keeps track of its presence rather than just poll it’s status frequently) to save power - it only sends the message when rejoining the network - but I cannot work out how it can possibly send a “I am not present” message that the hub is guaranteed to receive. That makes no sense to me so must be missing something.

ChrisHardie, 2021-08-06:
I would still find this a valuable feature to add. SmartThings presence sensors are the only thing keeping me active on that platform.

Hedda, 2022-02-08:
Does anyone know if the European / United Kingdom version of these works today with the ZHA integration in Home Assistant?

Can it be made to use of the [Device Tracker](https://www.home-assistant.io/integrations/device_tracker/) integration for [Presence Detection](https://www.home-assistant.io/integrations/#presence-detection) or?

I don't own one yet but got an offer to buy a couple of unused and wondering if worth it to track the arrival of our two cars.

Was sold as "SmartThings Presence Sensor" in the UK and rest of Europe (and "SmartThings Arrival Sensor" in North America).

ChrisHardie, 2022-08-07:
I believe there's still a lot of interest in this request for people migrating from SmartThings.

ocalvo, 2022-09-20:
Keep alive

Hedda, 2023-03-19:
Unstale

Hedda, 2023-09-15:
Still wanted

ChrisHardie, 2024-03-13:
Still quite interested.

closed 2024-09-16 by github-actions[bot]