zigpy/zha-device-handlers #327: [SUGGESTION] Rename the zha-device-handlers library and repository to "zigpy-device-handlers" for clarity and consistency

Hedda, 2020-04-08:
Please consider renaming the zha-device-handlers library and repository to "zigpy-device-handlers" (or something else with "zigpy" in it and without "zha" in it) for better clarity and more consistency.

As I understand it, even if zha-device-handlers library is today only used by the ZHA integration component for Home Assistant there is nothing really stopping other developers/projects using zigpy from also utilizing this zha-device-handlers library too even if it is not for the ZHA component for Home Assistant?

- https://github.com/zigpy

This is only a suggestion and not a must or a showstopper in any way, but IMHO 'name branding' is important even for hobby open-source organizations that are only made up by volunteers like the zigpy organization. As this project is now like the other zigpy projects part of a collection of projects which are a part of an organization shared by many members and is upstream for multiple sub-projects, using the same brand name can help everyone feel that they belong to the same team who are working for the same or similar goals.

There are now already all these libraries for zigpy where only "zha-device-handlers" and "bellows" do not follow the name-standard for the library and repository names (or 'brand' if you will):

- zigpy https://github.com/zigpy/zigpy/
  - **bellows https://github.com/zigpy/bellows**
  - zigpy-cc https://github.com/zigpy/zigpy-cc
  - zigpy-deconz https://github.com/zigpy/zigpy-deconz
  - zigpy-xbee https://github.com/zigpy/zigpy-xbee
  - zigpy-zigate https://github.com/zigpy/zigpy-zigate
  - zigpy-znp https://github.com/zha-ng/zigpy-znp
  - zigpy-deconz-parser https://github.com/zha-ng/zigpy-deconz-parser
  - **zha-device-handlers https://github.com/zigpy/zha-device-handlers**

If zha-device-handlers library and repo was renamed to "zigpy-device-handlers" then also PyPI project cand and probably should be renamed as well; 

- "zigpy-device-handlers" instead of "zha-quirks" https://pypi.org/project/zha-quirks/

I think that if the active PyPI project is named "zha-quirks" but the GitHub repo is named "zigpy-device-handlers" then that too can be confusing to new developers, especially to those who might wish to use zigpy and zigpy-device-handlers without Home Assistant as I can imagine that they could think that Home Assistant is a requirement for using these libraries.

Yes I have OCD but others might also find this could be a logical choice that makes total sense.

IMHO best for clarity and consistency was if both zha-device-handlers and bellows also follow the same naming-standard.

ZHA in Home Assistant is gaining traction and zigpy will likely get more and more related libraries over time. If the "zha-device-handlers" and "bellows" libraries were renamed "zigpy-device-handlers" all would follow the same naming-standard, and I believe that more and more developers could take notice and maybe join in the development of current and future zigpy related projects,

PS: @dmulcahey Regardless of above it is probably also a good idea if you could and would make a few more zigpy developers and zha-device-handlers contributors shared co-owner/co-author/co-maintainers of this project on GitHub as well the package project(s) on PyPI. That could help remove some of the personal pressure on yourselves as well as explicit dependence on specific individuals.

PPS: @dmulcahey I have not seen that @damarco and @AndreasBomholtz as active with zigpy related projects for quite some time, however, yourself, @Adminiuga and @puddly all seem to be very active with zigpy (considering these are all still just hobby projects), and also @roblandry and @Shulyaka look to be frequent contributors of this quirks project specifically so you might want to consider inviting them both to just this project?

Hedda, 2020-04-08:
For reference, this suggestion is related and very similar to https://github.com/zigpy/bellows/issues/215

Hedda, 2020-04-15:
For reference, see related https://github.com/zigpy/zigpy/pull/363 in which @Adminiuga changed  zigpy module name back to just "zigpy"

Hedda, 2020-04-23:
@dmulcahey sorry for bikeshedding but wondering how you feel about this renaming suggestion?

Apologies if my OCD ranting is not really constructive here but doesn't Parkinson's law of triviality clearly states that all bikesheds should have a corrugated galvanised iron roof and be black in colour ;)

Hedda, 2020-05-26:
@Adminiuga What is your opinion on rename the zha-device-handlers library and repo to zigpy_? ?

I understand if you guys feel that it is just too much work for little gain or too much bikeshedding.

Again this suggestion for rename the zha-device-handlers is mostly for consistency and to make it clear that zha-device-handlers could also be used by other projects than zha wanting to use zigpy.

Hedda, 2020-08-18:
@dmulcahey I think that this is also related to https://github.com/zigpy/zigpy/issues/452 as @pipiche38 posted there is he did not find it obvious that the zha-device-handlers library was a standalone library and not dependent or dedicated to ZHA for Home Assistant.

closed 2020-10-10 by dmulcahey

Hedda, 2020-10-11:
Closed without comment?

Shulyaka, 2020-10-11:
I guess at this point renaming would make more confusion than it would solve

pipiche38, 2020-10-11:
Which means that this is for ZHA first . That is where the confusion is.

Hedda, 2020-11-30:
> Which means that this is for ZHA first . That is where the confusion is.

FYI, Jeedom is another home automation project that is now implementing Zigbee support using zha-device-handlers and zigpy:

https://github.com/jeedom

https://github.com/jeedom/zha-device-handlers

https://github.com/jeedom/zigpy

@zoic21 & @sarakha63 Wondering what are your input/feedback on the current "zha-device-handlers" name as Jeedom developers? That is, wondering if you being new to zigpy found the "zha-device-handlers" naming confusing as related to the zigpy project? Based on that name alone and surely known about the existing "ZHA" integration in Home Assistant, what it in any way unclear that "zha-device-handlers" can be used with zigpy in any project and that it as a library has no direct dependencies on Home Assistant or its ZHA integration even though it was originally developed for use in that implementation?

Shulyaka, 2020-11-30:
Pip will handle dependencies automatically, so a normal user will not need this.
A developer will hardly guess the project based on its name, he will just lookup what the code imports.
Also, ZHA stands for Zigbee Home Automation, the name does not suggest it is for HA integration.
I understand your motivation and the confusion the name might imply, but the renaming itself could bring more confusion than solve. Just my 2 cents.

Hedda, 2021-06-07:
FYI, PR https://github.com/zigpy/zha-device-handlers/pull/842 tries to make it more clear that zha-device-handlers works without Home Assistant.