zigpy/zha-device-handlers #334: [Device Support Request] Quirky Outlink KWh

MagnaPilot, 2020-04-15:
Hello, I'd like to request a quirk for the Quirky/Wink Outlink outlets.  Presently, Home Assistant can enumerate the switch and a "power" meter entity, but the power meter device doesn't appear to be reading or parsing the correct cluster and attribute to calculate the load.

If it is possible, as I know this requires a calculation, I'd like to request that at least the correct attribute be visible to HA, more useful would for the correction to KWh and cycled refresh to monitor load usage over a short time.  I have included a link to how the SmartThings Community has adapted the same feature.

https://gist.github.com/mitchpond/4247278f35f3dc639c11
https://github.com/nicjansma/smart-things/blob/master/devicetypes/nicjansma/quirky-outlink/quirky-outlink.groovy

**Device signature - working on this now, will update.

Errors logged currently in HA that may be related:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 133, in async_set_state
    value = self.formatter(value)
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 270, in formatter
    return self._channel.formatter_function(value)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/smartenergy.py", line 159, in formatter_function
    return self._format_spec.format(value).lstrip()
ValueError: Format specifier missing precision
```

Adminiuga, 2020-04-15:
ATM the problem is with ZHA sensor platform which only supports one attribute per cluster.

MagnaPilot, 2020-04-15:
I by no means am educated enough in Zigbee to be arguing, I'm more just curious when I ask.  Are you saying that because ZHA is currently using the x0400 attribute, it cannot use the x0000 attribute because both reside under the 0x0702 cluster?   I only ask for my own curiosity.  I'm not sure what the x0400 attribute is used for for on this device, none of my Outlink's report any data from that attribute.

closed 2021-08-13 by github-actions[bot]