zigpy/zha-device-handlers #335: Device support request: QBKG03LM + cc2530 only shows status, no control.

levran, 2020-04-15:
the switch QBKG03LM is successfully paired with cc2530 + cp2102, is shown in ZHA. When physically turning on / off, the status changes in the HA (0,108,2) interface. But changing the status from the HA interface does not lead to a physical switch. In addition, after a few seconds in the interface HA, the switch returns to its previous state.

https://user-images.githubusercontent.com/3389141/79025251-dc02d600-7b8d-11ea-9660-86faef44a830.png

dmulcahey, 2020-04-15:
Can you reproduce this with another coordinator that is not part of the TI support? Deconz, ember, Xbee?

levran, 2020-04-15:
I have a deconz conbee2, with this coordinator the problem does not reproduce. The switch works well. 
The problem is only with TI2530.

Adminiuga, 2020-04-15:
If problem is only wit TI/ZNP, then enable debug logging, reproduce the problem and raise an issue in `zigpy-cc` repo