zigpy/zha-device-handlers #336: [BUG] Group binding fails for Hue Dimmer Switch

lordkev, 2020-04-16:
**Describe the bug**
Hue Dimmer Switch fails to bind properly to an existing group.

**To Reproduce**
Steps to reproduce the behavior:
1. Add Hue Dimmer Switch
2. Go to Hue Dimmer Switch device management page
3. Under "Group Binding" select a pre-existing group. In my case "Living Room Lights" which has only bulbs as members.
4. Select all clusters
5. Click "Bind Group"

**Expected behavior**
Dimmer switch should bind properly to the group and control the lights in the group (OnOff, LevelControl)

**Additional context**
Here is the device info:
```
Philips RWL020
RWL020
by Philips
IEEE: 00:17:88:01:06:70:2a:dc
Nwk: 0x1cb7
Device Type: EndDevice
LQI: 191
RSSI: -70
Last Seen: 2020-04-15T23:58:52
Power Source: Battery or Unknown
Quirk: zhaquirks.philips.rwl021.PhilipsRWL021
```

Here are logs from when I attempt to bind:
```
[0x1cb7:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[0x1cb7:1:0x0006] ZCL request 0x0001: []
[0x1cb7:1:0x0006] No handler for cluster command 1
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 0 to group: 0x0002
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 4 to group: 0x0002
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 3 to group: 0x0002
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 8 to group: 0x0002
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 6 to group: 0x0002
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 2, cluster: 25 to group: 0x0002
[0x1cb7:zdo] processing 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 5 to group: 0x0002
[0x1cb7] Extending timeout for 0x61 request
[0x1cb7] Extending timeout for 0x62 request
[0x1cb7] Extending timeout for 0x63 request
[0x1cb7] Extending timeout for 0x64 request
[0x1cb7] Extending timeout for 0x65 request
[0x1cb7] Extending timeout for 0x66 request
[0x1cb7] Extending timeout for 0x67 request
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 0 to group: 0x0002 completed: [<Status.TABLE_FULL: 140>]
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 4 to group: 0x0002 completed: [<Status.TABLE_FULL: 140>]
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 3 to group: 0x0002 completed: [<Status.TABLE_FULL: 140>]
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 8 to group: 0x0002 completed: [<Status.SUCCESS: 0>]
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 6 to group: 0x0002 failed: 
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 2, cluster: 25 to group: 0x0002 completed: [<Status.TABLE_FULL: 140>]
[0x1cb7:zdo] 0x1cb7: Bind_req 00:17:88:01:06:70:2a:dc, ep: 1, cluster: 5 to group: 0x0002 failed:
```

Here is what happens when I then try to use the buttons on the remote (in order):

```
[0x1cb7:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=15 command_id=1>
[0x1cb7:1:0x0006] ZCL request 0x0001: []
[0x1cb7:1:0x0006] No handler for cluster command 1

[0x1cb7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=14 command_id=2>
[0x1cb7:1:0x0008] ZCL request 0x0002: [0, 30, 9]
[0x1cb7:1:0x0008] No handler for cluster command 2

[0x1cb7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=16 command_id=2>
[0x1cb7:1:0x0008] ZCL request 0x0002: [1, 30, 9]
[0x1cb7:1:0x0008] No handler for cluster command 2

[0x1cb7:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=17 command_id=64>
[0x1cb7:1:0x0006] ZCL request 0x0040: [0, 0]
[0x1cb7:1:0x0006] No handler for cluster command 64
```

Happy to help troubleshoot however I can.

hpio, 2020-10-29:
@lordkev did you manage to solve the issue? I'm struggling with binding Hue Dimmer switches atm

Adminiuga, 2020-10-29:
You should only bind the following clusters:
- one-off cluster 0x0006
- level control cluster 0x0008
- color control clusters 0x0300

If binding an supported cluster they will respond with table full message

Adminiuga, 2020-10-29:
And after sending the binding request, octet it helps to wake up the device by pressing buttons on it. For each request sent

hpio, 2020-10-29:
Hi @Adminiuga thank you for your help, followed your info and got the following in logs:
```
2020-10-29 12:28:11 DEBUG (MainThread) [zigpy.zdo] [0x703b:zdo] 0x703b: Bind_req 00:17:88:01:04:09:1a:6a, ep: 1, cluster: 8 to group: 0x0003 completed: [<Status.TABLE_FULL: 140>]                                                                            2020-10-29 12:28:11 DEBUG (MainThread) [zigpy.zdo] [0x703b:zdo] 0x703b: Bind_req 00:17:88:01:04:09:1a:6a, ep: 1, cluster: 6 to group: 0x0003 completed: [<Status.TABLE_FULL: 140>]
2020-10-29 12:28:11 DEBUG (MainThread) [zigpy.zdo] [0x703b:zdo] 0x703b: Bind_req 00:17:88:01:04:09:1a:6a, ep: 1, cluster: 5 to group: 0x0003 completed: [<Status.TABLE_FULL: 140>]
```

However I'm still not able to control my lights and can see the following
```
2020-10-29 12:29:16 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [188] command id: 0 - args: [[1, 3145728, 0, 33, 0, 0]]
2020-10-29 12:29:16 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zhaquirks/philips/__init__.py", line 114, in _job
    self._callback(self._click_counter)
  File "/usr/local/lib/python3.8/site-packages/zhaquirks/philips/__init__.py", line 182, in send_press_event
    event_args[PRESS_TYPE] = press_type
UnboundLocalError: local variable 'press_type' referenced before assignment
```

tube0013, 2020-10-29:
According to the device page on zigbee2mqtt you have to unbind the hue dimmer from the coordinator before doing bindings.

https://www.zigbee2mqtt.io/devices/324131092621.html

hpio, 2020-10-29:
Thank you @tube0013 , would you be able to ELI5 me that?  Does that mean I need to configure MQTT discovery in Homeassistant and to bind it to a group follow https://www.zigbee2mqtt.io/information/binding.html#binding-a-remote-to-a-group ? 

I found a very comprehensive example of binding Dimmer Switch with Hue bulbs here https://zigbee2mqtt.discourse.group/t/binding-example/980/2 . Is that what I need to do as in there's no way to bind switches to lights through HA gui?

tube0013, 2020-10-29:
@Adminiuga Is there any way to unbind the coordinator with zha? That was the step I could not figure out.

Adminiuga, 2020-10-29:
[image]

?

Adminiuga, 2020-10-29:
you may need to check all groups, usually it is the group_0x0000

hpio, 2020-10-29:
I only have 5 groups that I manually created and there's no group with id `0x0000` 

[image]

Just to be clear, we're talking about ZHA groups, correct?

tube0013, 2020-10-29:
I'm the same, I don't have a 0x0000 group listed in the frontend.

[image]

tube0013, 2020-10-29:
I removed and repaired mine (as I had it joined and for a while, here is the log for joining. not sure if it sheds any light on this.

https://paste.ubuntu.com/p/H9bNHRJmTN/

Adminiuga, 2020-10-29:
For exception issue is that quirk which added multiple press support for hue is blowing up i think

hpio, 2020-10-30:
Are there any steps I can follow to troubleshoot the issue @Adminiuga ? At the moment I'm not any closer to resolving it nor I have more understanding of what's going on here

MattWestb, 2020-10-30:
I have one Philips LWB010 bulb and one RWL021 dimmer switch.
With Tasmota Zigbee bridge its only sending unbound with 0x0000 after binding the RWL012 with the groupe with the bulb and its working.
Was not getting it working with ZHA for some weeks.
Today i was reading them in ZHA and doing on new grupe with the bulb and then first remote cluster command, group binding and marking all cluster and the coordinator groupe (0x00000) and clicking unbinding for getting it unbinded.
Then changing grupe to the new one with only the bulb and marking cluster On/Off, LevelControl and Scenes and clicking bind groupe. After after little the remote is controlling the bulb but the bulb and remote is not reporting anything to the NCP.
I have one temporary grupe with one IKEA bulb and E1744 that have getting the coordinator in the groupe (0x0000) so i deleting the IKEA bulb and adding the Philips and the deleting the Philips and adding the IKEA in the new groupe, Then rebinding the IKEA E1744 to the new group and unbinding the Philips remote from the new group with the 3 clusters a bow and binding it to the old groupe with the coordinator in (0x0000).
Then I can controlling the Philips bulb with the remote and the remote is reporting to the coordinator (0x0000) but the bulb is not reporting / getting status being pulled. Have sending reconfigure device to the bulb and restarting HA but HA is not getting any status from the bulb ;-((

One tricky thing is how ZHA is adding or not adding the coordinator to Zigbee groupes and also if the coordinator groupe is being created of ZHA or not. The coordinator is not awable as device to being added in the grope management and all is doing in the magic background but i dont understanding how the logick is working.

For the moment ZHA have making one default groupe (0x0000) with the coordinator as member (0x0000), but its also happening that ZHA is not doing it and also that by mistake deleting the coordinator from the groupe its not working for the user as i have have done above .

2 steps forward  and one back wards, but for the moment i can control the bulb with bonded remote and that is great and dont need deCONZ for using it (IKEAs GW is not liking the crazy hybride remote (ala Philips ZLL/ZHA version mixture)).

MattWestb, 2020-10-30:
One PM:
I have suggested that the coordinator should not being possible being deleted in the default coordinator groupe (0x0000) and also possible manual adding the coordinator (0x0000) as device to zigbee groupes so user can adding and deleting it in in there zigbee groups for getting reporting to working for users but its not being implanted (issue posted in HA git with also default name of groups and renaming groups so not need patching the DB for renaming groups).

Adminiuga, 2020-10-30:
@hpio check the docs on how to install the dev version of zha handlers. You are getting the exception means you receive events from the remote

hpio, 2020-10-30:
@Adminiuga i added the repo and installed `Custom deps deployment` add-on, however how cna I find dev version of quirks to install? Releases page seems to list only stable versions. Thanks

hpio, 2020-11-02:
Hello @Adminiuga , thanks for your help so far! I have installed `zhaquirks==0.45.0` and re-started my HA to make sure it's updated, after that I removed and then re-added my Hue dimmer switch. Finally I bound it to 3 cluster as shown on the photo. However, I'm still seeing the following in logs
```
2020-11-02 08:19:22 DEBUG (MainThread) [zhaquirks.philips] PhilipsRemoteCluster - handle_cluster_request tsn: [56] command id: 0 - args: [[1, 3145728, 0, 33, 0, 0]]
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc01b>, ep: 2, profile: 0x0104, cluster_id: 0xfc00, data: b'1d0b1038000100003000210000'
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc01b>, 1, 260, 6, b'\x019\x01', 0, 175, 255, 216, 135, 144, 1, -22]
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy.zcl] [0xc01b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=57 command_id=1>
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy.zcl] [0xc01b:1:0x0006] ZCL request 0x0001: []
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy.zcl] [0xc01b:1:0x0006] No handler for cluster command 1
2020-11-02 08:19:22 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:17:88:01:04:09:1a:6a, unique_id=00:17:88:01:04:09:1a:6a:1:0x0006, endpoint_id=1, cluster_id=6, command=on, args=[]>
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc01b>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'013901'
2020-11-02 08:19:22 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
```

I can see `zha_event` registered in HA when pressing buttons but nothing happens with lights. What am I missing? Below are 3 events registered by HA when pressing ON button on a switch
```

Event 2 fired 08:29:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:04:09:1a:6a",
        "unique_id": "00:17:88:01:04:09:1a:6a:2:0xfc00",
        "endpoint_id": 2,
        "cluster_id": 64512,
        "command": "on_press",
        "args": {
            "button": "on",
            "press_type": "press",
            "command_id": 0,
            "args": [
                1,
                3145728,
                0,
                33,
                0,
                0
            ]
        }
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-02T08:29:22.306550+00:00",
    "context": {
        "id": "2619dfeaad24e811d5611cb80c3fd4fb",
        "parent_id": null,
        "user_id": null
    }
}

Event 1 fired 08:29:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:04:09:1a:6a",
        "unique_id": "00:17:88:01:04:09:1a:6a:2:0xfc00",
        "endpoint_id": 2,
        "cluster_id": 64512,
        "command": "on_short_release",
        "args": {
            "button": "on",
            "press_type": "short_release",
            "command_id": 0,
            "args": [
                1,
                3145728,
                2,
                33,
                0,
                0
            ]
        }
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-02T08:29:22.015663+00:00",
    "context": {
        "id": "c2fc33edcb37420c64e2f53e95fdd0f6",
        "parent_id": null,
        "user_id": null
    }
}

Event 0 fired 08:29:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:17:88:01:04:09:1a:6a",
        "unique_id": "00:17:88:01:04:09:1a:6a:1:0x0006",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-11-02T08:29:21.891877+00:00",
    "context": {
        "id": "f55b2c78ca2dbbcb54df93d2fb72db98",
        "parent_id": null,
        "user_id": null
    }
}

```

MattWestb, 2020-11-02:
The Philips HUE dimmer switch need being unbounded from the coordinator before it start sending commands to one new bounded group (i was first binding it to the new group and then undinding it from the coordinator and then it was start working).
If you having the default group (0x0000) with the coordinator in it (can being that ZHA is not doing it) then do the same as you have doing a bow (the same cluster) but with the group 0x0000 and clicking on unbind and you have done the unbinding from the coordinator.
If all OK its start sending commands to you groupe with bulbs.

hpio, 2020-11-02:
Thanks for your reply @MattWestb , however, as stated earlier I do not have the default group 0x0000 (see screenshot I posted few replies earlier). I also do not see a device/entity called coordinator

Adminiuga, 2020-11-02:
Try factory-resetting the remote, to remove all the groups from it.

Adminiuga, 2020-11-02:
1. Remove it from ZHA
2. Factory reset it
3. restart HA and make sure device is no longer there
4. re-pair it and collect the logs
5. bind to group and collect the logs

hpio, 2020-11-03:
Hi @Adminiuga , followed all the listed steps and here's the output - hope it helps to get to the bottom of the issue

logs when I pair the device with HA
```
2020-11-03 09:32:10 DEBUG (MainThread) [zigpy_deconz.api] Command Command.write_parameter (2, <NetworkParameter.permit_join: 33>, b'<')
2020-11-03 09:32:10 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x0bef0009000200213c
2020-11-03 09:32:10 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0bef000800010021
2020-11-03 09:32:10 DEBUG (MainThread) [zigpy_deconz.api] Write parameter permit_join: SUCCESS
2020-11-03 09:32:11 DEBUG (MainThread) [homeassistant.components.zha.entity] light.nursery_spotlight_4: polling current state - from cache: False
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 235 under 236 request id, data: b'00eb000000'
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 236, 0, <DeconzAddressEndpoint address_mode=2 address=0xCCA2 endpoint=11>, 260, 6, 1, b'\x00\xeb\x00\x00\x00', 2, 0)
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12f0001b001400ec0002a2cc0b0401060001050000eb0000000200
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12f0000900020022ec
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 236]
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ef1000700a600
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04f10007000000
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04f10013000c0022ec02a2cc0b010000000000
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 236: 00
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xec 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xcca2 endpoint=11>, status: 0x00
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0ef2000700aa00
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17f2000800010001
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17f20026001f00220200000102a2cc0b04010600080018eb01000000100100afff237f9101ca
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xcca2>, 11, 260, 6, b'\x18\xeb\x01\x00\x00\x00\x10\x01', 0, 175, 255, 35, 127, 145, 1, -54]
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy.zcl] [0xcca2:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=235 command_id=Command.Read_Attributes_rsp>
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xcca2>, ep: 11, profile: 0x0104, cluster_id: 0x0006, data: b'18eb010000001001'
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 237 under 238 request id, data: b'00ed000000'
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 238, 0, <DeconzAddressEndpoint address_mode=2 address=0xCCA2 endpoint=11>, 260, 8, 1, b'\x00\xed\x00\x00\x00', 2, 0)
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12f3001b001400ee0002a2cc0b0401080001050000ed0000000200
2020-11-03 09:32:11 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.nursery_spotlight_4, old_state=<state light.nursery_spotlight_4=off; min_mireds=153, max_mireds=454, off_brightness=None, friendly_name=Nursery spotlight 4, supported_features=43 @ 2020-11-03T09:28:26.635087+00:00>, new_state=<state light.nursery_spotlight_4=on; min_mireds=153, max_mireds=454, brightness=0, off_brightness=None, friendly_name=Nursery spotlight 4, supported_features=43 @ 2020-11-03T09:32:11.685665+00:00>>
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12f3000900020022ee
2020-11-03 09:32:11 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 238]
2020-11-03 09:32:11 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.nursery_lights_zha_group, old_state=<state light.nursery_lights_zha_group=on; min_mireds=153, max_mireds=454, brightness=254, color_temp=335, off_brightness=None, friendly_name=Nursery lights zha group, supported_features=43 @ 2020-11-03T09:31:33.722840+00:00>, new_state=<state light.nursery_lights_zha_group=on; min_m [... cut ...]  0, 175, 255, 245, 127, 145, 1, -41]
2020-11-03 09:37:54 DEBUG (MainThread) [zigpy.zcl] [0x9ac3:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=11 command_id=1>
2020-11-03 09:37:54 DEBUG (MainThread) [zigpy.zcl] [0x9ac3:1:0x0006] ZCL request 0x0001: []
2020-11-03 09:37:54 DEBUG (MainThread) [zigpy.zcl] [0x9ac3:1:0x0006] No handler for cluster command 1
```

Adminiuga, 2020-11-04:
@dmulcahey 
I played with RWL020 and apparently the binding table size on these devices is only 1, so it either could be bound to coordinator or to a group. Should we just allow bind/unbind on coordiator device in "manage clusters"?

Adminiuga, 2020-11-04:
@hpio wanna try something?
1. copy https://github.com/Adminiuga/zha_custom as `zha_custom` to `custom_components` folder in your configuration folder.
2. add `zha_custom:` to your `configuration.yaml`
3. restart HA
4. check that you have `zha_custom.execute` command
5. enable debug for `custom_components.zha_custom` component
6. call `zha_custom.execute` command with the following service data:

```yaml
ieee: ieee_address_of_rwl021_remote
command: unbind_coordinator
command_data: 6
```
7. press a key on remote to wake it up

check the logs (if you have enabled logging for `custom_components`) and it should unbind the on/off cluster.
repeat the command with `command_data: 8` and `command_data: 5`

Now try binding it to your group.

MattWestb, 2020-11-04:
@Adminiuga You have 110% right with the unbinding and i have posted one issue in HA for implanting the the coordinator in clustermanameng for open for binding / unbinding the coordinator and its working all times i have trying it.
I was thinking its belong in the HA part not in the zigpy then its one ZHA GUI future therefore the issue in HA. https://github.com/home-assistant/core/issues/41665#issue-718865602

hpio, 2020-11-04:
@Adminiuga this worked!

```
2020-11-04 09:11:06 INFO (MainThread) [custom_components.zha_custom] Running custom service: <ServiceCall zha_custom.execute (c:79a1e093268430c2d83dd5cd7ccfc4ac): ieee=00:17:88:01:04:09:1a:6a, command=unbind_coordinator, command_data=6>
2020-11-04 09:11:06 DEBUG (MainThread) [custom_components.zha_custom] module is <module 'custom_components.zha_custom' from '/config/custom_components/zha_custom/__init__.py'>
2020-11-04 09:11:06 DEBUG (MainThread) [custom_components.zha_custom.binds] running 'unbind coordinator' command: <ServiceCall zha_custom.execute (c:79a1e093268430c2d83dd5cd7ccfc4ac): ieee=00:17:88:01:04:09:1a:6a, command=unbind_coordinator, command_data=6>
2020-11-04 09:11:06 DEBUG (MainThread) [custom_components.zha_custom.binds] 0x09cf: unbinding ep: 1, cluster: 6
2020-11-04 09:11:07 DEBUG (MainThread) [custom_components.zha_custom.binds] 0x09cf: unbinding 0x0006: [<Status.SUCCESS: 0>]
2020-11-04 09:16:31 INFO (MainThread) [custom_components.zha_custom] Running custom service: <ServiceCall zha_custom.execute (c:10fa3dec8ceaa302ee4d093f315263fe): ieee=00:17:88:01:04:09:1a:6a, command=unbind_coordinator, command_data=8>                 2020-11-04 09:16:31 DEBUG (MainThread) [custom_components.zha_custom] module is <module 'custom_components.zha_custom' from '/config/custom_components/zha_custom/__init__.py'>
2020-11-04 09:16:31 DEBUG (MainThread) [custom_components.zha_custom.binds] running 'unbind coordinator' command: <ServiceCall zha_custom.execute (c:10fa3dec8ceaa302ee4d093f315263fe): ieee=00:17:88:01:04:09:1a:6a, command=unbind_coordinator, command_dat
a=8>
2020-11-04 09:16:31 DEBUG (MainThread) [custom_components.zha_custom.binds] 0x09cf: unbinding ep: 1, cluster: 8
2020-11-04 09:16:32 DEBUG (MainThread) [custom_components.zha_custom.binds] 0x09cf: unbinding 0x0008: [<Status.SUCCESS: 0>]
2020-11-04 09:16:41 INFO (MainThread) [custom_components.zha_custom] Running custom service: <ServiceCall zha_custom.execute (c:c77477bba5ed5da04022874e0ec33f69): ieee=00:17:88:01:04:09:1a:6a, command=unbind_coordinator, command_data=5>                 2020-11-04 09:16:41 DEBUG (MainThread) [custom_components.zha_custom] module is <module 'custom_components.zha_custom' from '/config/custom_components/zha_custom/__init__.py'>
2020-11-04 09:16:41 DEBUG (MainThread) [custom_components.zha_custom.binds] running 'unbind coordinator' command: <ServiceCall zha_custom.execute (c:c77477bba5ed5da04022874e0ec33f69): ieee=00:17:88:01:04:09:1a:6a, command=unbind_coordinator, command_dat
a=5>
2020-11-04 09:16:41 DEBUG (MainThread) [custom_components.zha_custom.binds] 0x09cf: unbinding ep: 1, cluster: 5                                                                                                                                              2020-11-04 09:16:41 DEBUG (MainThread) [custom_components.zha_custom.binds] 0x09cf: unbinding 0x0005: [<Status.SUCCESS: 0>]
```
After that I did bind the switch to a group and can now control them directly!

Thank you for your help. Would be great to be able to do it directly from UI as you suggested. Also, an entry in the documentation could definitely help, happy to submit a PR if pointed in the right direction

MattWestb, 2020-11-04:
@Adminiuga  Menny wikis is saying that all IKEA controlling devices must being unbonded from the coordinator before binding them 2 one (light)groupe (can only being bounded to one groupe). My experience that is working in ZHA to only bind them to the group and its working. Perhaps its because its was not bounded to the coordinator or its not necessarily. I have not testing binding one group and later trying binding it to one other group.

I finding it good adding the possibility binding and unbinding the coordinator in the ZHAs GUI so the user can doing it easy and also making it easier to explaining then user having problems.

Thanks for the updated zha_custom !!!

hpio, 2020-11-04:
I have just noticed one more issue, not sure if it's related to zha-device-handlers or any other component. I have 4 zha groups and 4 RWL021 switches. After unbinding remotes from coordinator I bound each RWL021 switch to a different zha group. Problem is, remotes control more zha groups than they should as in when i turn 1 group on/off more than 1 group responds.

I thought this was my error, so I removed the switch from zha, re-paired it, deleted and recreated zha group and eventually bound the switch to the group but problem is still there.

Any idea how I can debug it?

Adminiuga, 2020-11-04:
don't know. Stop home assistant to make sure it is not some automation triggering it. 
Could be lights in more than one group.
Could be scenes from remote?
For this one a Zigbee Sniffer would be useful, to see what remotes actually send over the air.

Adminiuga, 2020-11-04:
But I'm pretty sure a remote can control only one group, as there's room only for one binding. 
My take, the bulbs are members of multiple groups. Very often could happen when removing/adding groups.

1. Remove all groups from ZHA
2. in configuration -> Devices go to each ZHA light, -> Manage clusters -> Group cluster -> call "remove_all" command on the group cluster. This would remove all groups on that light
3. repeat for each light
4. power cycle the lights
5. recreate the group and add lights to it
6. remotes will keep the binding to the groups. so just need to create the groups and it will reuse the same group IDs starting from 2 (or 1, i don't remember)

Adminiuga, 2020-11-04:
@dmulcahey this hack allows Coordinator bind/unbind 🤷 I'm open to better ideas
https://github.com/Adminiuga/home-assistant/commit/6a507e111385b02d5c3226ef81f9bc6dbd766a05

dmulcahey, 2020-11-04:
I’ll see if I can peek at it later tonight

hpio, 2020-11-05:
> But I'm pretty sure a remote can control only one group, as there's room only for one binding.
> My take, the bulbs are members of multiple groups. Very often could happen when removing/adding [...]

That was it, it's all working nicely. Thank you!

closed 2020-11-06 by Adminiuga