zigpy/zha-device-handlers #340: [Device Support Request] Aqara motion sensor illuminance updates regardless of detected motion

Duncank, 2020-04-19:
**Is your feature request related to a problem? Please describe.**
Aqara motion sensors on the Aqara Hub report an updated illuminance level at least once every hour, regardless of motion detection. On ZHA they only report illuminance updates when motion is detected, making them not as useful as before.

**Describe the solution you'd like**
More regular illuminance updates, without the need for motion before an update is sent.

**Device signature**
Is that needed for this issue? If so, I will add it!

**Additional context**
[image]  
This line shows at least 8 hours without an update, since I didn't walk by the location off the sensor. The weather / sun brightness today was very different per hour; bright sun mixed with darker clouds, so a lot of brightness changes that usually would have been reported way more often.

I've found this[ similar issue from zigbee2mqtt ](https://github.com/Koenkk/zigbee2mqtt/issues/827)that talks about the same thing, with a solution that solved this in their case.

It is also possbile that this is no timed-update, but a change in lux that exceeds a certain tolerance (for example, updating every 30lx)

Adminiuga, 2020-04-19:
Need to parse sub attribute 11 from attribute 65281 report.

Adminiuga, 2020-04-19:
It won't be possible to get reports for every 30lx. We are limited by what device sends and otherwise device is in deep sleep and is not responding to polls. And on the other side xiaomi is not following specs

chmielowiec, 2020-04-19:
I created a fix for this and will test it over night

chmielowiec, 2020-04-20:
Illuminance was reported correctly after #341

closed 2020-04-20 by dmulcahey

deezid, 2024-08-09:
Still a problem it seems...

PR #341 (2020-04-20): Add lumi.sensor_motion.aq2 illuminance reports
Fixes #340