zigpy/zha-device-handlers #344: [BUG] CC2530 / CC2531 Router not detected by ZHA

fluffymadness, 2020-04-26:
**Describe the bug**
I tried to add a cc2530 and a cc2531 router via zha on homeassistant 
using the following firmwares (i use a conbee2 stick for zha)
https://ptvo.info/cc2531-based-router-firmware-136/
https://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
These firmwares are also used for zigbee2mqtt

I tried basically everything. Reflashing the sticks and trying the repair procedures by holding the sw2 button on cc2531 and by un/repluggin 3 times on the cc2530. 
Nothing gets detected by zha. All other zigbee devices I have, are working fine.
As far as I've understood. The firmwares listed above emulate a xiaomi lightbulb device.
I've used the cc2531 stick with deconz before were it also worked fine.

**To Reproduce**
- Plug in CC2531/CC2530
- Zha integration add new device
- Starting pairing procedure for CC2531/CC2530

**Expected behavior**
- The sticks should pair and register themselves as bulbs

Adminiuga, 2020-04-26:
Not sure this is the right repo for the issue. Do you have any other "router" devices on the network? As I had problems joining certain devices to ConBee II with only ConBee on the network.

Enable debug log for:
- zigpy_deconz
- zigpy
- homeassistant.components.zha
- homeassistant.core

Reset and try re-pair the device. Send logs. And make sure you are using an USB extension cable for ConBee

dmulcahey, 2020-04-29:
Use the router firmware from zigbee2mqtt to flash the sticks. I know that works because I have several of them

Adminiuga, 2020-04-29:
+1, Try firmwares from https://github.com/Koenkk/Z-Stack-firmware/tree/master/router

fluffymadness, 2020-04-30:
the firmwares should be the same as the ones I linked
https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531
Under the credits of the repo:
Credits for this firmware go to ptvoinfo

I tested the 2530 again, which I could pair without any problems.
So this one works.
But with 2531 no avail.
Does anyone have CC2531 working as router with zha?

Adminiuga, 2020-04-30:
Yes, CC2531 do work.

dmulcahey, 2020-06-28:
These work, I have several paired as routers

closed 2020-06-28 by dmulcahey

danieledwardgeorgehitchcock, 2020-08-11:
Also experiencing this issue.. Any pointers other than the above?

Adminiuga, 2020-08-11:
Std one

atus42, 2021-03-08:
I also have this problem. Have two cc2531.
One of them flashed as coordinator. This one works as expected, I can connect/pair devices through this device.
The other was flashed with the latest router firmware (CC2531_router_2020_09_29.zip). Plugging it in resulted fast red led blinks.
If I try to pair it via HA add device GUI, nothing happens.

dmulcahey, 2021-03-08:
I have 5 of them as routers (will be replacing them though) they definitely work. Make sure you have the right firmware on them.

atus42, 2021-03-08:
I thought I put the right firmware on it:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_Home_1.2/bin/CC2531_router_2020_09_29.zip

what firmware did You used?

Adminiuga, 2021-03-08:
Older one, but it shouldn't matter. Zha doesn't do anything special in treating cc2531 as routers.

Adminiuga, 2021-03-08:
Enable debug logging and check the logs, just in case ui is not showing anything. Make sure the router DOES try to join.

atus42, 2021-03-08:
Well, it DOES tries to connect, but it fails.

On the GUI, I see this:
[0x95fc:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=16 command_id=36>
Failed to parse message (b'19102400af') on cluster 61184, because Data is too short to contain 1 bytes

I attached the debug log.
[zha_router.txt](https://github.com/zigpy/zha-device-handlers/files/6102521/zha_router.txt)

Adminiuga, 2021-03-09:
There's no traffic from the device. Just permit broadcasts looped back. Are you sure the router tries to join?

atus42, 2021-03-10:
I think so.
When I plug it in an usb, it starts to flash the red led every second. To force it reconnect I pressed the S2 button (the one which didn't turn on/off the grean led) as long as the rd led blink stops a little. Then I release the button, and the red led blinking starts again.

I also tried plugging the device in and pull it out for three times (leaving it 2 seconds plugged). It didn't changed the result.

Today I will try to flash it with the usb debug version, and see "the other side" messages. It may help pinpoint what happens.

atus42, 2021-03-10:
With the debug version, I was able to add the router, and turn on/off the green led remotely.
On the other hand, I was not able to pair a new device through the newly paired router. (add device through this device)
I will paste logs here soon.

mortenmoulder, 2021-05-26:
@atus42 I have this exact issue you're having. Exact same logs in ZHA as well.

Did you ever get yours to work?

atus42, 2021-05-26:
It's been a long time since then.
So far 
- I ordered a new CC2531 with external antenna. This solved my issues (frequent unavailable endpoints).
- I upgraded my HomeAssistant to 2021.05.x

If I open the visualization of my zigbee network I can see that two endpoints communicates the gateway through this router.
So I guess they (the devices themselves) solved the issue over time, but without me taking any measure.

mortenmoulder, 2021-05-26:
@atus42 Good to hear you managed to solve it.

The CC2531 I have has an external antenna as well. However, mine won't pair just like yours wouldn't.

I am supposed to just go into ZHA and add a new device, and then it should appear.. right? Nothing else?

ruimarinho, 2021-06-12:
I am able to pair a CC2531 but I can't add devices going through it - I always get this "No such device" message:

```
2021-06-12 21:55:20 DEBUG (MainThread) [zigpy_zigate.zigbee.application] No such device 0x85dd
2021-06-12 21:55:20 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001f05000104040301010285dd020000180d0a000029db03140028ff100029932663
2021-06-12 21:55:20 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'000104040301010285dd020000180d0a000029db03140028ff1000299326' LQI:99
2021-06-12 21:55:20 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 1027, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0x85dd>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18\r\n\x00\x00)\xdb\x03\x14\x00(\xff\x10\x00)\x93&']
```