zigpy/zha-device-handlers #345: Centralite 3157100 with Home Assistant using a Husbzb-1

okan74, 2020-04-26:
I paired Centralite 3157100 with Home Assistant using a Husbzb-1 controller. ZHA discovery worked but it only discovers it as fan and show battery level. Isn't there supposed to be a temperature, heat, cold options? I'm using RPi4 with supervisor version 219 and HassOS 3.13. Thanks.

Adminiuga, 2020-04-26:
What is Centralite 3157100?

okan74, 2020-04-26:
It's a thermostat a/c controller. It's shown as supported on "[Code](https://github.com/zigpy/zha-device-handlers)" page.

Adminiuga, 2020-04-26:
There's no official climate support yet for zha. You could try semi official https://github.com/Adminiuga/home-assistant/tree/ac/new-zha-climate-0.108.x branch

jbrechtel, 2020-04-28:
@Adminiuga Does that fork have support for the Zen Thermostat? I noticed you committed some zha quirks for it recently. If it does then I might switch to it.

Do you have an outline on what's left to get it merged into master? I might be able to contribute.

Adminiuga, 2020-04-28:
I actually tested/developed it with a Zen thermostat,  albeit on a test stand and not on a real equipment,  but only till the heat wave comes. 

Needs time, tests and some clean up, Still haven't added time update for the Sinope thermostat.

Probably easiest way to test  is to copy ZHA folder from those branches (0.109.x branch to come soon) into your `/config/custom_components/zha`  folder and run it as a custom component.

Hedda, 2020-05-18:
FYI, Adminiuga has now submitted a pull request for "ZHA Climate" (beta) with thermostats support.

Adminiuga, 2020-06-28:
This should be working in 0.112 beta

closed 2020-06-28 by Adminiuga