zigpy/zha-device-handlers #346: [Device Support Request] Support for Xiaomi Aqara Smart LED bulb (ZNLDP12LM)

pcmester, 2020-04-29:
Currently, this bulb is not supported by this project, although, it's supported by the zigbee2mqtt project. I think according to that, it would be easy to support it by ZHA.
https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html

albalaing, 2020-05-12:
With this custom quirks I have got it to work. But it comes out as a colored light bulb and not as a single dimmable one
```python
import logging

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, MultistateInput, OnOff, Identify,Groups,Scenes,LevelControl 
from zigpy.zcl.clusters.lighting import Color

from .. import LUMI, BasicCluster, PowerConfigurationCluster, XiaomiCustomDevice
from ... import CustomCluster
from ...const import (
	DEVICE_TYPE,
	ENDPOINTS,
	INPUT_CLUSTERS,
	MODELS_INFO,
	OUTPUT_CLUSTERS,
	PROFILE_ID,
	VALUE,
)

_LOGGER = logging.getLogger(__name__)

class LightAqcn02(XiaomiCustomDevice):

	class ColorCluster(CustomCluster, Color):

		cluster_id = Color.cluster_id

		def __init__(self, *args, **kwargs):
			"""Init."""
			super().__init__(*args, **kwargs)

		def _update_attribute(self, attrid, value):
			_LOGGER.info("Color: %X = %X",attrid, value )
			if attrid == 0x400A:
				super()._update_attribute(attrid, 0x0017)
			else:
				super()._update_attribute(attrid, value)

	signature = {
		#endpoint=1 profile=260 device_type=258 device_version=1 input_clusters=[0, 4, 3, 5, 10, 258, 13, 19, 6, 1, 1030, 8, 768, 1027, 1029, 1026] output_clusters=[25, 10, 13, 258, 19, 6, 1, 1030, 8, 768]>

		MODELS_INFO: [(LUMI, "lumi.light.aqcn02")],
		ENDPOINTS: {
			1: {
				PROFILE_ID: zha.PROFILE_ID,
				DEVICE_TYPE: 258,
				INPUT_CLUSTERS: [0, 4, 3, 5, 10, 258, 13, 19, 6, 1, 1030, 8, 768, 1027, 1029, 1026],
				OUTPUT_CLUSTERS: [25, 10, 13, 258, 19, 6, 1, 1030, 8, 768],
			}
		},
	}
	replacement = {
		ENDPOINTS: {
			1: {
				DEVICE_TYPE: zha.DeviceType.COLOR_DIMMABLE_LIGHT,
				INPUT_CLUSTERS: [
					BasicCluster, #0
					Groups.cluster_id, #4
					Identify.cluster_id, #3
					Scenes.cluster_id, #5
					# Time 10
					# WindowCovering 258
					# AnalogOutput 13
					# Multiestateoutput 19
					OnOff.cluster_id, # 6
					PowerConfigurationCluster, #1
					# OccupancySensing 1030
					LevelControl.cluster_id, # 8
					ColorCluster, #Color.cluster_id, # Color 768
					# PressureMeasurement 1027
					# RelativeHumidity 1029
					# TemperatureMeasurement 1026
		  
				],
				OUTPUT_CLUSTERS: [],
			}
		}
	}

```

Adminiuga, 2020-05-15:
@albalaing IIRC color channel cluster is reading the "color_capabilities"  attribute and it may need to be "plugged" with color temperature capabilities, but not with color control. 

Intercepting `_update_attribute()`  may not  work, as color  channel  processes results of `read_attributes()`.
Although, upon restart, it should get the value overriden by  the `_update_attribute()` :shrug:

albalaing, 2020-05-23:
With this custom quirks it works well for me but as Adminiuga says you have to restart to get it right.
I don't know how you can make it work when you pair
```python 
import logging

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, MultistateInput, OnOff, Identify,Groups,Scenes,LevelControl 
from zigpy.zcl.clusters.lighting import Color

from .. import LUMI, BasicCluster, PowerConfigurationCluster, XiaomiCustomDevice
from ... import CustomCluster
from ...const import (
	DEVICE_TYPE,
	ENDPOINTS,
	INPUT_CLUSTERS,
	MODELS_INFO,
	OUTPUT_CLUSTERS,
	PROFILE_ID,
	VALUE,
)

_LOGGER = logging.getLogger(__name__)

class LightAqcn02(XiaomiCustomDevice):

	class ColorCluster(CustomCluster, Color):

		cluster_id = Color.cluster_id

		def __init__(self, *args, **kwargs):
			"""Init."""
			super()._update_attribute(0x400A, 0x0010)
			super()._update_attribute(0x400B, 153)
			super()._update_attribute(0x400C, 370)
			super().__init__(*args, **kwargs)

	signature = {
		#endpoint=1 profile=260 device_type=258 device_version=1 input_clusters=[0, 4, 3, 5, 10, 258, 13, 19, 6, 1, 1030, 8, 768, 1027, 1029, 1026] output_clusters=[25, 10, 13, 258, 19, 6, 1, 1030, 8, 768]>

		MODELS_INFO: [(LUMI, "lumi.light.aqcn02")],
		ENDPOINTS: {
			1: {
				PROFILE_ID: zha.PROFILE_ID,
				DEVICE_TYPE: 258,
				INPUT_CLUSTERS: [0, 4, 3, 5, 10, 258, 13, 19, 6, 1, 1030, 8, 768, 1027, 1029, 1026],
				OUTPUT_CLUSTERS: [25, 10, 13, 258, 19, 6, 1, 1030, 8, 768],
			}
		},
	}
	replacement = {
		ENDPOINTS: {
			1: {
				DEVICE_TYPE: zha.DeviceType.COLOR_DIMMABLE_LIGHT,
				INPUT_CLUSTERS: [
					BasicCluster, #0
					Groups.cluster_id, #4
					Identify.cluster_id, #3
					Scenes.cluster_id, #5
					# Time 10
					# WindowCovering 258
					# AnalogOutput 13
					# Multiestateoutput 19
					OnOff.cluster_id, # 6
					PowerConfigurationCluster, #1
					# OccupancySensing 1030
					LevelControl.cluster_id, # 8
					ColorCluster, #Color.cluster_id, # Color 768
					# PressureMeasurement 1027
					# RelativeHumidity 1029
					# TemperatureMeasurement 1026
		  
				],
				OUTPUT_CLUSTERS: [],
			}
		}
	}
```

AdrinkBeer, 2020-05-26:
@albalaing I tried to test your quirk but I got the following error on restart :

`ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /zigbeestick for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 217, in async_setup
    hass, self
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 141, in async_initialize
    app_config, auto_form=True, start_radio=True
  File "/usr/local/lib/python3.7/site-packages/zigpy/application.py", line 61, in new
    await app._load_db()
  File "/usr/local/lib/python3.7/site-packages/zigpy/application.py", line 53, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.7/site-packages/zigpy/appdb.py", line 351, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.7/site-packages/zigpy/quirks/__init__.py", line 17, in get_device
    return registry.get_device(device)
  File "/usr/local/lib/python3.7/site-packages/zigpy/quirks/registry.py", line 163, in get_device
    device = candidate(device._application, device.ieee, device.nwk, device)
  File "/usr/local/lib/python3.7/site-packages/zhaquirks/xiaomi/__init__.py", line 84, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/zigpy/quirks/__init__.py", line 48, in __init__
    self.add_endpoint(endpoint_id, replace_device=replaces)
  File "/usr/local/lib/python3.7/site-packages/zigpy/quirks/__init__.py", line 63, in add_endpoint
    ep = custom_ep_type(self, endpoint_id, replacement_data, replace_device)
  File "/usr/local/lib/python3.7/site-packages/zigpy/quirks/__init__.py", line 87, in __init__
    cluster = c(self, is_server=True)
  File "/usr/local/lib/python3.7/site-packages/zhaquirks/xiaomi/aqara/**light_aqcn02.py**", line 32, in __init__
    super()._update_attribute(0x400A, 0x0010)
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/__init__.py", line 487, in _update_attribute
    self._attr_cache[attrid] = value
  File "/usr/local/lib/python3.7/site-packages/zigpy/zcl/__init__.py", line 507, in __getattr__
    raise AttributeError("No such command name: %s" % (name,))
AttributeError: No such command name: _attr_cache`

I'm doing this for the first time so maybe I did something wrong.

I'm using HA (0.110.2) in Docker, and I added light_aqcn02.py with the other quirks.

Do you have any ideas ? Thanks !

albalaing, 2020-05-26:
Sorry but change the order of the lines without realizing that it could cause problems.
The correct order is:
```python
		def __init__(self, *args, **kwargs):
			super().__init__(*args, **kwargs)
			super()._update_attribute(0x400A, 0x0010)
			super()._update_attribute(0x400B, 153)
			super()._update_attribute(0x400C, 370)
```

AdrinkBeer, 2020-05-26:
Perfect, thank you !

AdrinkBeer, 2020-05-27:
It works, the only thing I can't do is changing the color temperature (I tested on 2 different bulbs)

albalaing, 2020-05-27:
Color temperature works well for me. I'm using HA (0.110.2)

[image]

These two lines record the max and min values 
```
super()._update_attribute(0x400B, 153)
super()._update_attribute(0x400C, 370)
```
My light bulb
2700k to 6500k (370mired to 153mired)
Mired = 1000000 / K

AdrinkBeer, 2020-06-18:
Hi, 

Thanks for your help. Do you think that a pull request could be made in order to integrate this quirk ? Even if a restart is still required.

Adminiuga, 2020-06-18:
You could replace
```
		def __init__(self, *args, **kwargs):
			super().__init__(*args, **kwargs)     # Corrected
			super()._update_attribute(0x400A, 0x0010)
			super()._update_attribute(0x400B, 153)
			super()._update_attribute(0x400C, 370)
```
with
```
    _CONSTANT_ATTRIBUTES = {
         0x400A: 0x0010,
         0x400B: 153,
         0x400C: 370
}
```

Adminiuga, 2020-06-18:
this way whenever entity reads those attrs, it would get the correct values

albalaing, 2020-06-18:
As for doing the pull request, I haven't done it because I'm new to github and I don't know how to do it. But if someone did, it would be great.

AdrinkBeer, 2020-06-18:
@albalaing I will try to make a PR. Could you please send me the last version of your quirk ? With my version, I still see the bulb as a colored bulb and I could not make the color temp work. Thank you !

albalaing, 2020-06-18:
I just tested this code and it works for me
```python
import logging

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, MultistateInput, OnOff, Identify,Groups,Scenes,LevelControl 
from zigpy.zcl.clusters.lighting import Color

from .. import LUMI, BasicCluster, PowerConfigurationCluster, XiaomiCustomDevice
from ... import CustomCluster
from ...const import (
	DEVICE_TYPE,
	ENDPOINTS,
	INPUT_CLUSTERS,
	MODELS_INFO,
	OUTPUT_CLUSTERS,
	PROFILE_ID,
	VALUE,
)

_LOGGER = logging.getLogger(__name__)

class LightAqcn02(XiaomiCustomDevice):

	class ColorCluster(CustomCluster, Color):

		_CONSTANT_ATTRIBUTES = {
         	0x400A: 0x0010,
         	0x400B: 153,
         	0x400C: 370
		}

	signature = {
		#endpoint=1 profile=260 device_type=258 device_version=1 input_clusters=[0, 4, 3, 5, 10, 258, 13, 19, 6, 1, 1030, 8, 768, 1027, 1029, 1026] output_clusters=[25, 10, 13, 258, 19, 6, 1, 1030, 8, 768]>
		MODELS_INFO: [(LUMI, "lumi.light.aqcn02")],
		ENDPOINTS: {
			1: {
				PROFILE_ID: zha.PROFILE_ID,
				DEVICE_TYPE: 258,
				INPUT_CLUSTERS: [0, 4, 3, 5, 10, 258, 13, 19, 6, 1, 1030, 8, 768, 1027, 1029, 1026],
				OUTPUT_CLUSTERS: [25, 10, 13, 258, 19, 6, 1, 1030, 8, 768],
			}
		},
	}
	replacement = {
		ENDPOINTS: {
			1: {
				DEVICE_TYPE: zha.DeviceType.COLOR_DIMMABLE_LIGHT,
				INPUT_CLUSTERS: [
					BasicCluster, #0
					Groups.cluster_id, #4
					Identify.cluster_id, #3
					Scenes.cluster_id, #5
					OnOff.cluster_id, # 6
					PowerConfigurationCluster, #1
					LevelControl.cluster_id, # 8
					ColorCluster,  # 768
				],
				OUTPUT_CLUSTERS: [],
			}
		}
	}
```

closed 2020-06-18 by dmulcahey

Siggenv, 2020-07-11:
I'm a beginner
I can`t make it work. gets this error message.

Org.codehaus.groovy.control.MultipleCompilationErrorsException: startup failed: script_dth_metadata_0493b88a_811e_4a05_b221_4bcf17bb454a: 3: expecting EOF, found 'import' @ line 3, column 21. from zigpy.profiles import zha ^ 1 error

any suggestions?

Adminiuga, 2020-07-11:
This code is not for groovy or Smart Things device handlers