zigpy/zha-device-handlers #347: [BUG] Aqara vibration sensor no zha_event just state_change

GH2user, 2020-05-02:
**Describe the bug**
related to #124 I guess.
The Aqara vibration sensor is not reporting any zha_events but the binary_sensor does get updated with just 'vibration' instead of the tilt (incl angle), drop as available in the zha_event.

**To Reproduce**
Steps to reproduce the behavior:
4. See error

**Expected behavior**
I would expect to see a zha_event  but it's a state-change I guess, not allowed to listen for these.

**Screenshots**
device details
[image]

binary events log
[image]

zha_event log same time period, nothing received after first two events (wakeup and motion)
[image]

zigbee.db attributes table for this device
[image]

dmulcahey, 2020-05-02:
How are you testing this?

Adminiuga, 2020-05-02:
Aqara vibration sensor is an entity, not a remote so it doesn't do zha_events?

dmulcahey, 2020-05-02:
It does events for tilt entity for vibration

GH2user, 2020-05-02:
I'm listening for zha_events in HA - developer tools - listening for events.
Thinking there would always be a zha_event to get to the sensor_state....
But not receiving those at all, just the state change which is only 'vibration'

dmulcahey, 2020-06-29:
`<Event zha_event[L]: device_ieee=00:15:8d:00:02:af:97:27, unique_id=00:15:8d:00:02:af:97:27:1:0x0500, endpoint_id=1, cluster_id=1280, command=Tilt, args=({'degrees': 79},)>` I get these... which is what is expected.

closed 2020-06-29 by dmulcahey