zigpy/zha-device-handlers #348: Add Profalux support (Covers)

Tobby-NZ, 2020-05-03:
Hello

In HA, Profalux covers appear as a switch and not as a cover.
I use Pi-Zigate, documentation and compatibility are describe here (but in french):

*  Zigate : https://zigate.fr/
*  Motor : https://zigate.fr/produits-profalux-compatibles-zigate/compatible/moteurvoletprofalux/
*  Remote : https://zigate.fr/produits-profalux-compatibles-zigate/compatible/telecommandesimpleprofalux/

How can I change type of this entity or anything else to manage level of covers ?

Some informations are missing :

    Basic (Endpoint id: 1, Id: 0x0000, Type: in)
        Attributes :
            manufacturer (id: 0x0004) : None
            model (id: 0x0005) : None
            stack_version (id: 0x0002) : None
            zcl_version (id: 0x0000) : 1
    ManufacturerSpecificCluster (Endpoint id: 1, Id: 0xfc20, Type: in)
        No attribute
        No commands
    ManufacturerSpecificCluster (Endpoint id: 1, Id: 0xfc21, Type: in)
        No attribute
        No commands
    ManufacturerSpecificCluster (Endpoint id: 1, Id: 0xfc20, Type: out)
        No attribute
        No commands

But LevelControl and other commands to manage cover are presents :

    LevelControl (Endpoint id: 1, Id: 0x0008, Type: in)
        Attributes :
            current_level (id: 0x0000)
            default_move_rate (id: 0x0014)
            off_transition_time (id: 0x0013)
            on_level (id: 0x0011)
            on_off_transition_time (id: 0x0010)
            on_transition_time (id: 0x0012)
            remaining_time (id: 0x0001)
        Commands :
            move (id: 0x0001)
            move_to_level (id: 0x0000)
            move_to_level_with_on_off (id: 0x0004)
            move_with_on_off (id: 0x0005)
            step (id: 0x0002)
            step_with_on_off (id: 0x0006)
            stop (id: 0x0003)
            stop (id: 0x0007)

**Device signature**
#### Cover
```
# <SimpleDescriptor
#     endpoint=1 profile=260 device_type=512 device_version=0
#     input_clusters=[0, 3, 4, 5, 6, 8, 10, 21, 256, 64544, 64545]
#     output_clusters=[3, 64544]
# >
```

#### Remote
```
# <SimpleDescriptor
#     endpoint=1 profile=260 device_type=513 device_version=0
#     input_clusters=[0, 3, 21]
#     output_clusters=[3, 4, 5, 6, 8, 256, 64544, 64545]
# >
```

**Additional context**
[... 201 lines cut ...]
                    OnOff.cluster_id, # 6
                    LevelControl.cluster_id, # 8
                    Shade.cluster_id, # 256
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_1, # 64544
                    MANUFACTURER_SPECIFIC_CLUSTER_ID_2, # 64545
                ],
            },
        }
    }
```

#### Const
```python
"""Module for Profalux quirks implementations."""
# TODO : Add in zhaquirks.const
DEVICE_VERSION = "device_version"
DEVICE_MANUFACTURER = "manufacturer"
DEVICE_MODEL = "model"

MANUFACTURER_SPECIFIC_CLUSTER_ID_1 = 0xFC20  # decimal = 64544
MANUFACTURER_SPECIFIC_CLUSTER_ID_2 = 0xFC21  # decimal = 64545
PROFALUX_MANUFACTURER = "PROFALUX"
PROFALUX_MODEL_COVER = "Cover / Volet roulant"
PROFALUX_MODEL_REMOTE = "Remote / Télécommande"
```

europrimus, 2020-06-03:
Hello,
I have profalux cover and there register as switch (off = closed, on = opened).
I want to try your quirk, but i don't find where to put the files. With ssh , i can't find any ```zhaquirks``` directory. Could you help me to test?

I'm running home assistant on a raspberyPi 3 with zigate usb/ttl
```
arch | armv7l
dev | false
docker | true
hassio | true
host_os | HassOS 3.13
installation_type | Home Assistant
os_name | Linux
os_version | 4.19.114-v7
python_version | 3.7.7
supervisor | 226
version | 0.110.4
virtualenv | false
```

tube0013, 2020-06-03:
try HA beta shortly, this may be fixed there see the release notes re zha
covers - https://rc.home-assistant.io/blog/2020/06/01/release-111/

Tobby-NZ, 2020-06-03:
Hi

Try bêta is better, my quircks  doesn't work =(
For the moment, I use many automation with HA called by button. Each automation call ZHA service zha.issue_zigbee_cluster_command in action menu. For close cover, you can try this data :
cluster_id: 6
cluster_type: in
command: 0
command_type: server
endpoint_id: 1
ieee: '20:91:8a:00:00:xx:xx:xx'
In zigbee panel settings, you can explore all cluster and command from your Zigbee sévices.

I think use Automation in HA is most powerfull. You can do everything with your home before any developper start himself idea.

If you want, I can help you with this tomorrow or this week-end in French.
I will upgrade HA for use new ZHA cover on next release because my Home is not a lab ! =)

europrimus, 2020-06-04:
Thanks for this information.
I'll wait for the new release too.

europrimus, 2020-06-29:
Now my profalux covers works. But they still shown as **cover.unk_manufacturer_unk_model_56d60a00_level_on_off_shade**
Is  the manufacturer and model are ID (like for USB) ? so there is a file for converting to *human readable name*?

Tobby-NZ, 2020-06-29:
You can add my code in ZHA Quirck directory for add name, manufacturer and other informations.

glamirand, 2020-09-19:
Hi!
Glad to find this post, I'm actually having troubles trying to control my Profalux covers with a Zigate by HA.
I managed to pair a first cover with the Zigate and control it using "ZigBee Gateway UI" provided as a test SW for the Zigate.
My cover appears in HA as "unk_manufacturer unk_model", I have a lovelace card with up/down/stop buttons.
I see things happening when I click these buttons as the USB adapter activity led shows up but with no effect on the shutter.
Can anyone help me with ZHA quirks for those Profalux shutters as I think this is the best way to solve my problem?
PS : I can speak french ;)
Thanks!

pipiche38, 2020-09-19:
For more insight https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Profalux-corner.md

Tobby-NZ, 2020-09-21:
Hi @glamirand 

When you try to get attribut LevelControl of Profalux in Cluster Management, what's do you obtain ?
[image]

You can try to set Toggle in command's Cluster (UP to DOWN or DOWN to UP)
[image]

If you doesn't get any activity on Zigbee log, the integration doesn't work and probably you need to add the WA join on top of thread.

pipiche38, 2020-09-23:
As mentioned in the wiki page I have put above, I do suggest to not provide a mapping to the OnOff as recommended by Profalux themselves

Tobby-NZ, 2020-09-24:
Integration of Cover in HA doesn't work with LevelControl for use Up/Stop/Down. 
Without LevelControl, you can only use Slider for select the level of Cover. So in Lovelace you can'nt select action without extend Automation.

[image]

closed 2021-08-13 by github-actions[bot]