zigpy/zha-device-handlers #350: [Device Support Request] GLEDOPTO GL-B-008ZS

msslava, 2020-05-07:
**Is your feature request related to a problem? Please describe.**
https://www.zigbee2mqtt.io/devices/GL-B-008Z.html
This device can be added to the home assistant, but does not save its color state after turning it off.
1) I adjust the color
2) Turn off the device (via the wall switch)
3) Turn on the device (via the wall switch)
4) I see a different color, not the same as I set

Thanks

tube0013, 2020-05-07:
It is up to the device to determine its power-on behavior.  Some device will use their last setting, others just go back to a default state.

msslava, 2020-05-07:
If I use addon zigbee2mqtt, the settings are saved. And recover after shutdown. 
Can this be fixed in ZHA?

Adminiuga, 2020-05-07:
Correct. On power cycle, smart lights reset to the default settings. Some manufacturers have implemented  manufacturer specific commands to change the power on  defaults. If you know it exist for gledopto and know the specifics of the command, then it could be implemented via a quirk

msslava, 2020-05-07:
Also, if you combine these lamps in a group, then nothing works. I can not turn it on or off, choose a color and control the brightness

msslava, 2020-05-07:
Strange, but sometimes after turn off via wall switch device keep the set color. I can’t understand the logic...

closed 2021-08-13 by github-actions[bot]