zigpy/zha-device-handlers #351: [BUG] Centralite 3200-sgb

blair287, 2020-05-10:
My CentraLite 3200-Sgb works fine for on off and power reporting but have a couple of issues.

1) it's discovered as a light not a switch resulting in it being on the light domain causing issues with Google home etc

2) power reporting after a while like 3 weeks decides to 10x the figure so what was 100w is now showing as 1000w not sure if ZigBee or HA fault.

**Expected behavior**
Should be added as a switch and in the switch. Domain.

Power reporting to work like it does when first added for ever.

**Screenshots**
[image]
[image]

Adminiuga, 2020-05-10:
post zigbee inforrmation for  the device

blair287, 2020-05-12:
> post zigbee inforrmation for the device

Sorry I'm a big noob when it comes to this how can I get you the info you need?

Adminiuga, 2020-05-12:
The device reports itself as `"device_type": "0x0100"` which corresponds to ON_OFF_LIGHT in Zigbee specs.
To make it a switch, you would need to add the following in your `configuration.yaml`
```yaml
zha:
  device_config:
    00:0d:6f:00:0f:3b:20:2d-1:
      type: "switch"
```
this would override the light as a switch.

Adminiuga, 2020-05-12:
as for the 2nd issue
> power reporting after a while like 3 weeks decides to 10x the figure so what was 100w is now showing as 1000w not sure if ZigBee or HA fault. - removing the plug and then factory reset and readding it back fixes the issues till it happens again

going to need logs and isolating the steps which trigger/reproduce this effect. See https://www.home-assistant.io/integrations/zha/#debug-logging on how to enable logging. 
I see to possibilities:
1. device just starts reporting wrong values
2. ZHA couldn't get corrrect multiplier and/or divisor for some reason, although once it gets it, it should keep the value cached. 

When it happens next time, can you pinppoint when exactly it happened and if happened after a restart.
Read the following attributes of the "Electrical Measurement" cluster
- "ac_power_divisor",
- "power_divisor",
- "ac_power_multiplier",
- "power_multiplier",

Actually, read those attributes when device is reporting correct and incorrect measurements.

When it start reporting wrong values:
1. Does HA restart fix it
2. Does Config panel -> ZHA -> Pick the plug -> Reconfigure Device fix it? just make sure plug is available and not off the network

blair287, 2020-05-12:
Right so the first issues I can just list them all and force it to think it's a switch? Will I need to reads them or anything or will it switch to a switch after a restart?

For your second post issues I'll get the info next time it happens but I can say now that a HA restart doesn't correct the issue neither does reconfiguring it.

Only way I know to fix it is remove device factory reset device and add back then it reports values correctly.

blair287, 2020-05-12:
> The device reports itself as `"device_type": "0x0100"` which corresponds to ON_OFF_LIGHT in Zigbee specs.
> To make it a switch, you would need to add the following in your `configuration.yaml`
> 
> ```yaml
> zha:
>   device_config:
>     00:0d:6f:00:0f:3b:20:2d-1:
>       type: "switch"
> ```
> 
> this would override the light as a switch. You could use the same to change switches to light. the "key" in device config is the "ieee address" of the device, "dash" and "endpoint number". In some more specific cases you may have to ad "dash" + "`cluster_id`", eg `00:0d:6f:00:0f:3b:20:2d-1-6`

Thanks this at least gets around the issue. The new smartthings plugs 2019 correctly show as switches from the get go at least. although the energy monitoring doesn't work gets detected as a sensor but just shows unavailable, power monitoring works though. Would it be possible to make it work or is energy monitoring not supported? would need some kind of UI to reset it etc.

blair287, 2020-05-26:
> as for the 2nd issue
>
>
> going to need logs and isolating the steps which trigger/reproduce this effect. See https://www.home-assistant.io/integrations/zha/#debug-logging on how to enable logging.
> I [...]

Right it happened again it seems that it is happening after a HA restart but the values for those clusters dont change between when its working and when its getting the wrong reading.

It also seems like it does sometimes fix it when HA restarts.

It seems to happen quite frequently anyway to fix it?

McFarTech, 2022-02-07:
> Thanks this at least gets around the issue. The new smartthings plugs 2019 correctly show as switches from the get go at least. although the energy monitoring doesn't work gets detected as a sensor [...]

Hi. 

I am also having this issue with the ST plug showing as a Light and not a switch. I have put the code in my yaml file but its still showing as a light. Have i missed something? The device details of the plug are below. 

Device info
3200-Sgb
by CentraLite
Connected via Electrolama Zigbee Coordinator
Firmware: 0x30065310
Zigbee info
IEEE: 00:0d:6f:00:0f:3b:1d:32
Nwk: 0xb77e
Device Type: Router
LQI: Unknown
RSSI: Unknown
Last Seen: 2022-02-07T20:52:05
Power Source: Mains

McFarTech, 2022-02-07:
> Hi.
>
> I am also having this issue with the ST plug showing as a Light and not a switch. I have put the code in my yaml file but its still showing as a light. Have i missed something? The device [...]

Sorry this is working. I had missed the -1: off the end of the address. Thanks for providing a solution